tzapu/WiFiManager #1201: ESP32 - OTA not work. With ESP8266 is OK !!

robert4you, 2021-02-02:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32:  ESP32

Hardware:		NODE ESP-32

Core Version:      	2.4.0,

### Description

I am using the new example for the OTA update test (https://github.com/tzapu/WiFiManager/tree/master/examples/DEV/OnDemandConfigPortal ). To test this function I am trying to flash the same sketch from exampel.
If I use Node ESP8266 everything work ok, but when I try to use ESP32 I get the following error:

11:13:05.006 -> *WM: [3] -> 192.168.4.1 
11:13:05.006 -> Update: TestOTA.bin
11:13:05.006 -> Bad Size Given
11:13:05.006 -> *WM: [1] [OTA] Update ERROR Not enough space
11:13:05.006 -> *WM: [3] -> 192.168.4.1 
11:13:05.006 -> .Bad Size Given
11:13:05.048 -> *WM: [3] -> 192.168.4.1 
11:13:05.048 -> .Bad Size Given
11:13:05.048 -> *WM: [3] -> 192.168.4.1 
11:13:05.048 -> .Bad Size Given
11:13:05.103 -> *WM: [3] -> 192.168.4.1 
11:13:05.103 -> .Bad Size Given

The website displays a message:  OTA Error: 5

### Settings in IDE

Module:  		

Additional libraries: 

### Sketch:   Exampel: https://github.com/tzapu/WiFiManager/tree/master/examples/DEV/OnDemandConfigPortal - only.

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tablatronix, 2021-02-02:
I had this problem also, I think this means there is no space for ota, or maybe mem for your board is set wrong..

robert4you, 2021-02-02:
I tried different memory settings on the board and several board models. Same result.

tablatronix, 2021-02-04:
Try it now

closed 2021-02-04 by tablatronix

Bighoneypot, 2023-02-21:
@robert4you  fix link https://github.com/tzapu/WiFiManager/tree/master/examples/DEV/OnDemandConfigPortal  in this page