tzapu/WiFiManager #1203: WiFi Password field asterisks

OldGreyCells, 2021-02-03:
### WiFi password field 'placeholder' causes confusion

#### Hardware
WiFimanager Branch/Release:  Development (I have tested this on master as dev was merged into master yesterday: https://github.com/tzapu/WiFiManager/issues/500)

Esp8266/Esp32: Esp8266

Hardware:			ESP-12e, D1 Mini

Core Version:      	2.4.0, staging

### Description

If _showPassword is false the S_passph is displayed in the WiFI password input, even if a password is not set. This causes a level of confusion because it looks like the field has already been completed. ie user saves the WiFi settings but the connection fails even though it looks like there is a password entered.

I humbly suggest the S_passph should only be used if WiFi_psk() is not empty?

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: None

tablatronix, 2021-02-03:
true but you also do not have to resave wifi everytime anymore if its already saved. We can reuse same password if empty and ssid is not, or we can find another way to indicate this is saved

tablatronix, 2021-02-03:
Maybe disable the inputs and you have to click change or something ?

tablatronix, 2021-02-03:
There seems to be 3 bugs here
working on them now

tablatronix, 2021-02-03:
hmm issue comment triggers are not working

e933a20d20b6bc636053f811810bdf924d3c2b73

tablatronix, 2021-02-03:
I fixed the **** when no psk saved
I also fixed the ssid showing blank if connect failed, also the status showing no ap set even if there is one

closed 2021-02-03 by tablatronix

OldGreyCells, 2021-02-05:
Apologies for the tardy response time - been 'under the weather'...
I'll test as soon as I can get platformio to pull the latest version - or failing that, I'll just copy the .cpp across.
Many thanks - much appreciated.

tablatronix, 2021-02-05:
np, thanks so much for creating an issue and finding that

OldGreyCells, 2021-02-05:
OK, tested and works great. There is a little usability issue when re-entering the portal if already holding a valid WiFi connection as the SSID and password are blank - should perhaps be filled? Not part of this issue and quite a trivial thing but happy to raise another issue if you'd like.
Many thanks for your work (and patience!)

tablatronix, 2021-02-05:
hmm, do you mean the placeholders re still blank ?

Are you on esp8266 or esp32?

OldGreyCells, 2021-02-05:
I'm on an esp8266 - when initiating the portal on an erased flash, the SSID and password are blank as expected but if you set them, get connected to your regular router (or not), if you re-enter the portal, both fields are blank. This is not a big issue, (provided you don't touch the list of available APs because the stored SSID/psk aren't updated if the fields are empty) it is just not necessarily what a user would expect.

tablatronix, 2021-02-05:
ok , looks like I only tested on esp32, will fix, must be a bug on esp8266

tablatronix, 2021-02-05:
Nope works fine for me same as screenshot above..

tablatronix, 2021-02-05:
Works fine here
[image]

tablatronix, 2021-02-05:
Maybe make sure you pulled latest commits in

OldGreyCells, 2021-02-05:
Hmmm... I'll check again.

OldGreyCells, 2021-02-05:
My apologies for taking up your time - my test sketch (the OnDemandWebPortal) works now I've manually copied the latest pull across. I'm still getting used to platformio using separate libraries for each sketch and not sure how to force it update to the the latest master - I have to create a new project with lib_deps = https://github.com/tzapu/WiFiManager.git#master and copy it across to the original sketch.
All the best

tablatronix, 2021-02-05:
ahh I see
lib_deps

https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps

Personally I use extra_lib_dir and git pull my own stuff for experimental or cutting edge libs

OldGreyCells, 2021-02-05:
Thanks for the tip - it's a bit of a learning curve - made more so as C++ is not my first (second, or even third) language!

tablatronix, 2021-02-05:
Yeah I tend to not use pio as projects, and I use it more like arduino and have all libraries in the same place

OldGreyCells, 2021-02-16:
Apologies for the absence - I tested positive for Covid, so am not entirely with it! (not that I ever am)

tablatronix, 2021-02-16:
oof sorry to hear, Best wishes with that

commit e933a20d (2021-02-03): Fixes #1203

Psk placeholder should not show if no psk is saved