tzapu/WiFiManager #1205: WebPortal param save doesn't seem to save to eeprom

cpainchaud, 2021-02-04:
I am using lastest code (using OnDemandWebPortal example) and when I spawn config portal manually to edit some of my saved parameters, it seems to work but after I restart my ESP all values are back to what they were before.

Interestingly if I refresh the params page before i reboot, it properly displays the new values. I suspect they are never written to EEPROM

cpainchaud, 2021-02-04:
somewhere in my setup:

`WiFiManagerParameter mqtt_s_param("mqtt_s", "hostname or ip",  MQTT_SERVER.c_str(), 40);
WiFiManagerParameter mqtt_p_param("mqtt_p", "port",  MQTT_PORT.c_str(), 6);
WiFiManagerParameter mqtt_id_param("mqtt_id", "id", MQTT_ID.c_str(), 20);
WiFiManagerParameter mqtt_u_param("mqtt_u", "user", MQTT_USER.c_str(), 20);
WiFiManagerParameter mqtt_sec_param("mqtt_sec", "mqtt password", MQTT_PSWD.c_str(), 20);

void setup_WifiManager(){

  const char* menu[] = {"wifi","param","info","close","sep","erase","restart","exit"};
  
  wifiManager.addParameter(&mqtt_s_param);
  wifiManager.addParameter(&mqtt_p_param);
  wifiManager.addParameter(&mqtt_id_param);
  wifiManager.addParameter(&mqtt_u_param);
  wifiManager.addParameter(&mqtt_sec_param);

  wifiManager.setMenu(menu, sizeof(menu));`

later i call ondemand portal!

`wifiManager.setConfigPortalBlocking(false);
wifiManager.startWebPortal();`

tablatronix, 2021-02-04:
WM does not write to eeprom or save your params for you

cpainchaud, 2021-02-04:
Ok, this is confusing : where does it store wifi credentials then? These
are properly saved and remembered....

tablatronix, 2021-02-04:
The esp library stores them automatically

cpainchaud, 2021-02-04:
Oh sorry! It made sense to me that WifiManager would store all configs, to
avoid the various other pieces  ans librairies to "fight" for eeprom space
and step on each other toes. That would be a wonderful addition

closed 2021-02-04 by cpainchaud

tablatronix, 2021-02-04:
Most of the examples use spiffs not eeprom, wm does not tie itself to any specific storage and leaves it to the user. I would love to see some child classes though to do this automatically, just not in core