tzapu/WiFiManager #1207: Show extra page after connection successful

juliovedovatto, 2021-02-05:
I'm trying to figure out if there is a way to show a extra configuration page after connection successful.

I'm developing a ESP application that user will have to provide credentials in order to use the service. So I was imagining showing an extra page after the connected credentials, without closing the configuration portal.

Is this possible or do I need to rely on custom parameters?

tablatronix, 2021-02-05:
This has been addressed in other issues of the same request, at this time no, for more info try to find those related issues.

closed 2021-02-05 by tablatronix

juliovedovatto, 2021-02-05:
@tablatronix thanks for the quickly reply

I navigated through the opened issues and didn't find anything related to my question.

 Could you point a issue/direction/keywords so I can check?  Thanks

tablatronix, 2021-02-05:
ill try to write it up one day, but there are a few...
ummm

#1024

tablatronix, 2021-02-05:
This will be on the roadmap after this release

litamin, 2021-02-15:
@tablatronix you mean there will eventually be a way to display the assigned IP (in browser) after successful connection? If so: YAAAS! ;)