tzapu/WiFiManager #1209: AP Does Not Respect Bandwidth Directive

lbussy, 2021-02-10:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master (I don't see a "development" branch in the list)

Esp32: 

Hardware:			ESP32-WROOM-32

Core Version:      	v3.2.3-14-gd3e562907

### Description

When AP mode starts, it starts with 40MHz bandwidth and multiple channels.  When I set the bandwidth and channel the code does not seem to respect the setting as determined by a spectrum analyzer:

[image]
[image]

### Settings in IDE

Module:  			Lolin D32

Additional libraries: N/A

### Sketch

```cpp
#include <Arduino.h>
#include <WiFiManager.h>

void setup()
{
    Serial.begin(BAUD);
    Serial.setDebugOutput(true);
    Serial.flush();
    Serial.println();

    WiFiManager wm;
    wm.setDebugOutput(true);
    wm.debugPlatformInfo();
    wm.resetSettings();

    wm.setCountry("US");
    wm.setWiFiAPChannel(1);

    esp_wifi_set_bandwidth(WIFI_IF_AP, WIFI_BW_HT20);

    if (!wm.autoConnect("AutoConnectAP"))
    {
        Serial.println("Failed to connect.");
    }
    else
    {
        Serial.println("Connected.");
    }
}

void loop() {}
```

### Debug Messages

```
*WM: [1] Free heap:        317176
*WM: [1] ESP SDK version:  v3.2.3-14-gd3e562907
*WM: [1] resetSettings
*WM: [1] SETTINGS ERASED
*WM: [1] AutoConnect
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal
*WM: [2] Disabling STA
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID:  AutoConnectAP
*WM: [2] Starting AP on channel: 1
*WM: [2] AP has anonymous access! 
*WM: [1] AP IP address: 192.168.4.1
*WM: [2] WiFiSetCountry to US
*WM: [2] [OK] esp_wifi_set_country:  US
*WM: [1] Starting Web Portal
*WM: [2] HTTP server started
*WM: [2] Config Portal Running, blocking, waiting for clients... 
```

tablatronix, 2021-02-10:
Same, I looked into this before not sure if it was an esp bug or hardware bug.

tablatronix, 2021-02-10:
ok figured it out, let me see if i can work something ( i thought setcountry was able to set this )

tablatronix, 2021-02-10:
You have to use apcallback, needs to be set after starting ap

```c++
  wm.setAPCallback(configModeCallback);

//gets called when WiFiManager enters configuration mode
void configModeCallback (WiFiManager *myWiFiManager) {
  esp_wifi_set_bandwidth(WIFI_IF_AP, WIFI_BW_HT20);
}
```

closed 2021-02-10 by tablatronix

lbussy, 2021-02-10:
Hah!  You had a fix for it all along. :)

Thank you, sir.

tablatronix, 2021-02-10:
I was playing with this a few weeks ago, and forgot all about it, I had to test again to see if it needed to be before or after, if you set it before, it doesn't work, unlike channels. Shrug

lbussy, 2021-02-12:
You know this - but that ended up working.  Thank you for your help.