tzapu/WiFiManager #1211: How to set SSID and Password programmatically ?

sn00zerman, 2021-02-10:
#### Hardware
WiFimanager Branch/Release:  Development
Esp32: 
Hardware:			Lilygo TTGO T8
Core Version:      	2.4.0, staging

### Description:
I'm looking for a way to "pre-fill" the SSID and Password field,
when the captive portal shows up.
(Like one would do when presenting extra parameters, you also have a way to fill it in from the software-side ...)

### Additional libraries: 
#include "WiFi.h"
#include "HTTPClient.h"
#include "WiFiManager.h" 
#include "WiFiClient.h"
#include "WebServer.h"
#include "ESPmDNS.h"
#include "Update.h"

sn00zerman, 2021-02-12:
I managed to solve my own problem :-) (after a good night-sleep, I could think more clear again)

- I added 2 parameters, of type "hidden": hidden_s and hidden_p  (hidden_s contains programmatically SSID, and hidden_p the password)
- I added a function in the HTTP_SCRIPT[] in strings_en.h, just before the </script> tag:
  "function myHidden()\n" 
  "{\n"
  "  var txt_s = document.getElementById('s');\n"
  "  var txt_hidden_s = document.getElementById('hidden_s');\n"
  "  var txt_p = document.getElementById('p');\n"
  "  var txt_hidden_p = document.getElementById('hidden_p');\n"
  "  txt_s.value = txt_hidden_s.value;\n"
  "  txt_p.value = txt_hidden_p.value;\n"
  "}\n"
  - in the HTTP_HEAD_END[], I added onload='myHidden()' , right after the <body 
  
  Everytime, the page loads, it passes the data from the hidden field, into the s and p field :-)

tablatronix, 2021-02-12:
It will use the creds from the ESP, so just save them to the esp. Are you obtaining them from somewhere else?

sn00zerman, 2021-02-12:
I needed a way to programmatically set them in the captive portal, giving the user the possibility to "adjust" these settings and save them. I obtain them trough an algorithm, AP's are taken offline, "re-calculated" and taken back online, in a 24hour cycle ...

But my problem is solved now, with the hidden_s and hidden_p :-)

closed 2021-02-12 by sn00zerman

tablatronix, 2021-02-12:
I mean why not just do WiFi.begin(SSID,PSK);

tablatronix, 2021-02-12:
fyi I added this a few days ago

    // to preload autoconnect for test fixtures or other uses that skip esp sta config
    bool          preloadWiFi(String ssid, String pass);