tzapu/WiFiManager #1212: Get SSID AND PASSWORD FROM CONFIG PORTAL

shakirhussain78652, 2021-02-11:
Hello, I am new here and don't have much knowledge about programming.
I want to use the auto connect non-blocking code for my project and for that I will be needing the SSID and PASSWORD entered by the user from config portal into my remotexy project.

#define REMOTEXY_WIFI_SSID "Redmi" // these should be fetch from the config portal
#define REMOTEXY_WIFI_PASSWORD "123456789" // these should be fetch from the config portal
#define REMOTEXY_CLOUD_SERVER "cloud.remotexy.com"

below is the autocoonect non blocking code that i will be using
[image]
[image]

#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
WiFiManager wm;

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP    
    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //reset settings - wipe credentials for testing
    //wm.resetSettings();

    wm.setConfigPortalBlocking(false);

    //automatically connect using saved credentials if they exist
    //If connection fails it starts an access point with the specified name
    if(wm.autoConnect("BLUE1", "12345678")){
        Serial.println("connected...yeey :)");
    }
    else {
        Serial.println("Configportal running");
    }
}

void loop() {
    wm.process();
    // put your main code here, to run repeatedly:
    
}

sn00zerman, 2021-02-11:
If you want to read the SSID and password, after you made a succesfull connection in the Captive Portal,
you simple need to read out:
WiFi.SSID()    //this contains the SSID
WiFi.psk()       //this contains the password

shakirhussain78652, 2021-02-13:
Thank you for your help. But as I told you earlier I am a beginner and I don't know where to put this code and how to print it. Can you please write the complete code.
Many thanks.

#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
WiFiManager wm;

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP    
    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //reset settings - wipe credentials for testing
    //wm.resetSettings();

    wm.setConfigPortalBlocking(false);

    //automatically connect using saved credentials if they exist
    //If connection fails it starts an access point with the specified name
    if(wm.autoConnect("BLUE1", "12345678")){
        Serial.println("connected...yeey :)");
    }
    else {
        Serial.println("Configportal running");
    }
}

void loop() {
    wm.process();
    
    // put your main code here, to run repeatedly:

shakirhussain78652, 2021-02-13:
Thank you so much for your help. That saved my day.

shakirhussain78652, 2021-02-13:
Also please help me to enter these  
WiFi.SSID() //this contains the SSID
WiFi.psk() //this contains the password
 into this code

#define REMOTEXY_WIFI_SSID ""
#define REMOTEXY_WIFI_PASSWORD "" 
#define REMOTEXY_CLOUD_SERVER "cloud.remotexy.com"
#define REMOTEXY_CLOUD_PORT 6376
#define REMOTEXY_CLOUD_TOKEN "479a49b065595afa5cdd3e7a3f8a66f0"