tzapu/WiFiManager #1213: Can't reconnect with waitForConnectResult after disconnect

Primus007, 2021-02-15:
### Basic Infos
Arduino IDE 1.8.13 

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ x] Development

**Esp8266/Esp32:**
- [ x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [x] v2.7.4
- [ ] staging (master/dev)

### Description

I use your WifiManager to connect to my home Network. It works fine. But after i get some infos from the internet i disconnect the wifi in my Code. it should reconnect if i press a button. but if i do that, always the AP Config mode fires up and i get the error: [ERROR] wifi begin failed

### Settings in IDE

Module:  			Wemos D1 & R2 mini

Additional libraries: 

### Sketch

```#include <ESP8266WiFi.h>
#include <WiFiManager.h>

WiFiManager wm;

void setup() {
  Serial.begin(115200);
  delay(100);
  Serial.println("");
  WiFi.mode(WIFI_STA); //  Station Mode 
  Serial.print("try to connect to wifi ");
  Serial.println("SSID: " + WiFi.SSID());
  startWiFi(); // Connect
  delay(2000);
  wm.disconnect();
  Serial.println("Wifi Disconnected");
  Serial.println("");
  delay(2000);
  Serial.print("Try to RECONNECT to wifi ");
  Serial.println("SSID: " + WiFi.SSID());
  startWiFi(); // Connect
}

void loop() {
//
}

// -------------------------------------------------------------------------
//                               startWiFi
// -------------------------------------------------------------------------
   //for reference:
    //typedef enum {
    //    WL_NO_SHIELD = 255,   // for compatibility with WiFi Shield library
    //    WL_IDLE_STATUS = 0,
    //    WL_NO_SSID_AVAIL = 1,
    //    WL_SCAN_COMPLETED = 2,
    //    WL_CONNECTED = 3,
    //    WL_CONNECT_FAILED = 4,
    //    WL_CONNECTION_LOST = 5,
    //    WL_DISCONNECTED = 6
    //} wl_status_t;
void startWiFi() {
    Serial.print("Is WiFi Configured and saved: ");
    Serial.println((String)wm.getWiFiIsSaved() ? "YES" : "NO");

    if (!WiFi.isConnected()) { 
        Serial.println("SSID: " + WiFi.SSID());
        int conn_result = WiFi.waitForConnectResult(10000); // connect try it for 10 seconds
        Serial.print("conn_result: ");
        Serial.println(conn_result, DEC);
    }
    if (WiFi.isConnected()) {
        Serial.println("Wifi is connected!");
    }
    else {
        Serial.println("Wifi is NOT connected!");
    }
    
    Serial.println("");
}

```

The serial result:
First connect:
conn_result: 3
Wifi is connected!

Wifi Disconnected

Try to RECONNECT to wifi SSID: InProg1
Is WiFi Configured and saved: YES
SSID: InProg1
conn_result: 0
Wifi is NOT connected!

I use the WiFi.waitForConnectResult(10000);  method because i want limit the waiting time to connect (10 Seconds).

Why can i not connect again with the same method after a disconnect?
Is it impossible? Is there another way to reconnect to the same wifi with a limit time method for connection?

Thank's for Help.

tablatronix, 2021-02-15:
Try a non persistent disconnect first, stuff gets stuck in esp and I suspect there are some lib bugs or sdk bugs atm.

wm.disconnect();

Or try connecting twice, another esp bug

Primus007, 2021-02-15:
A second try to connect again does not solve the problem.

How do i an non persistent disconnect?

tablatronix, 2021-02-15:
using the wm func i pasted above

Primus007, 2021-02-15:
I use wm.disconnect(); for disconnect . You can see it in the code in my first post.

At first i connect with: WiFi.waitForConnectResult(10000);
Then a delay
Then wm.disconnect();
and then again
WiFi.waitForConnectResult(10000);

The first connect works fine, the second not. Why?

What are the alternatives for a reconnect to the saved wifi with a connection timeout eg waitForConnectResult(10000)?

tablatronix, 2021-02-16:
Ah sorry did not see that not sure whats going on

tablatronix, 2021-02-16:
Ill see what mine does, maybe try a diff version for esp?

Primus007, 2021-02-16:
I have tryed Core v2.7.0 - .4 always the same problem. Arduino sdk or Visual Studio always the same problem.

tablatronix, 2021-02-16:
Where are you actually starting wifi ?

Primus007, 2021-02-16:
I have startet the wifimanager config portal and setup my wifi one times.

In the sketch from my first post is no start wifi. it is all what you see in my first post.
I use this line
`int conn_result = WiFi.waitForConnectResult(10000); // connect try it for 10 seconds`
to connect to the bevore saved wifi.

How should i start wifi?

tablatronix, 2021-02-16:
WiFi.begin() or wm autoconnect

That just waits for the result to change.

Primus007, 2021-02-16:
Many thanks you for your help. Now it's working.
I didn't know that it was just waiting for the result. I was confused that it connected at the first try.