tzapu/WiFiManager #1214: On demand portal: Updating (just) password does not save and then does not shut down portal

OldGreyCells, 2021-02-16:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32: 

Hardware:			ESP-12e
Core Version:      	2.4.0, staging

### Description
When re-entering the AP portal on demand:
1. Go to [Configure WiFi]
2. Enter a new password (do not select the network or enter the SSID because it is already displayed)
3. Hit [Save]

#### Expected results

New password is saved and the WiFI [re]connected using new value.

#### Actual Result:

Updated password  is not saved, the client is dropped (from the AP) and WiFi is disconnected - messages from wm: 
```
*WM: [2] <- HTTP WiFi save  
*WM: [2] processing save 
*WM: [2] No ssid, skipping wifi save 
*WM: [2] Disabling STA 
````

The AP portal then keeps running, ignoring the ```wm.setConfigPortalTimeout(140);```

(I suspect I have to stop the portal programmatically, but would expect the portal timeout to be honoured.)

### Settings in IDE

Module:  			Wemos D1

Additional libraries: None

```cpp

#include <Arduino.h>
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// select which pin will trigger the configuration portal when set to LOW
int ONDDEMANDPIN = 0; // gpio for button
long lastState = 0;
long stateInterval = 10000;

WiFiManager wm;

void setup() {
  Serial.begin(9600);
  delay(1000);
  Serial.println("\n Starting");
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
  wm.setConfigPortalTimeout(140);
  wm.setParamsPage(false);

  if (!wm.autoConnect("AP_Portal"))
  {
    Serial.println("failed to connect and hit timeout");
  }
  else
  {
    Serial.println("connected...yeey :)");
  };
  // put your setup code here, to run once
  pinMode(ONDDEMANDPIN, INPUT_PULLUP);
}

void startApPortal() {
  Serial.println("startApPportal()...");
  if (wm.getConfigPortalActive()) {
    Serial.println("AP portal is already running");
    return;
  }
  //this->wm.setConfigPortalBlocking(false); //Doesn't work if this is set
  wm.setConfigPortalTimeout(140);
  wm.setParamsPage(false);
  wm.setConfigPortalBlocking(false); //Don't forget wm.process() in loop()
  wm.startConfigPortal("AP_Portal");
};

void loop() {
  long now = millis();
  if ((now - lastState) > stateInterval) {
    lastState = now;
    Serial.println("Looping...");
    Serial.printf("WiFi %s connected: ", (WiFi.isConnected() ? "is" : "not"));
    Serial.println(WiFi.localIP());
    Serial.printf("AP portal %s running.\n", (wm.getConfigPortalActive() ? "is" : "not"));
    Serial.printf("WiFi password %s\n", wm.getWiFiPass().c_str());
    Serial.printf("Wifi is saved %s\n", (wm.getWiFiIsSaved()?"true":"false"));
  }

  // is configuration portal requested?
  if (digitalRead(ONDDEMANDPIN) == LOW)  {
    delay(100);
    if (digitalRead(ONDDEMANDPIN) == LOW) {
      Serial.println("BUTTON PRESSED");
      startApPortal();
    }
  }
  if (wm.getConfigPortalActive() ) {
    wm.process();
  }
};
```

### Debug Messages

```
 Starting
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: <MY NETWORK>
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.72
connected...yeey :)
Looping...
WiFi is connected: 192.168.1.72
AP portal not running.
WiFi password <1234>
Wifi is saved true
BUTTON PRESSED
startApPportal()...
*WM: [2] Starting Config Portal 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  AP_Portal
*WM: [2] AP has anonymous access! 
*WM: [1] AP IP address: 192.168.4.1
*WM: [1] Starting Web Portal 
*WM: [2] HTTP server started 
*WM: [2] Config Portal Running, non blocking/processing 
Looping...
WiFi is connected: 192.168.1.72
AP portal is running.
WiFi password <1234>
Wifi is saved true
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
Looping...
WiFi is connected: 192.168.1.72
AP portal is running.
WiFi password <1234>
Wifi is saved true
*WM: [2] <- HTTP Root 
*WM: [2] <- HTTP Wifi 
*WM: [2] WiFi Scan SYNC started 
*WM: [2] WiFi Scan completed in 1584 ms
*WM: [1] 2 networks found
*WM: [2] DUP AP: <MY NETWORK>
*WM: [2] AP:  -47 <MY NETWORK>
Looping...
WiFi is connected: 192.168.1.72
AP portal is running.
WiFi password <1234>
Wifi is saved true
*WM: [2] <- HTTP WiFi save  
*WM: [2] processing save 
*WM: [2] No ssid, skipping wifi save 
*WM: [2] Disabling STA 
Looping...
WiFi not connected: (IP unset)
AP portal is running.
WiFi password <1234>   <---- Original password, not the one entered in the portal
Wifi is saved true
...
Looping... 
WiFi not connected: (IP unset)
AP portal is running. <---- Does not honour the portal time out
WiFi password <1234>
Wifi is saved true
```

tablatronix, 2021-02-16:
These are 2 conditions that I think are undefined

* change password, no password equality eval check
* save persistent even on sta fail

I will take a look, pretty sure I do not have issues for these yet, thanks

OldGreyCells, 2021-02-16:
I think this is similar to #1203 in that it's partly a UI issue - the SSID *looks* like it is filled in but at save time it is not set (as in ``` *WM: [2] No ssid, skipping wifi save ```.
Everything works as expected if you select or fill in the SSID on the portal (apart from remaining connected to the existing network if you deliberately set the wrong password - this could be very confusing because the user will *think* they've entered the right password until the device is reset... - perhaps a separate issue to raise?

joannela, 2021-08-24:
hi there! i am having this exact same issue. did it get resolved? i am working with v 0.14.