tzapu/WiFiManager #1215: (Question) How to achieve high reliabilty that credentials are never lost across reboots

bwjohns4, 2021-02-18:
I'm using WiFiManager in a project where I need high reliability and never want to have to redo entering in SSID/password. Sometimes when I do an OTA update, the credentials get cleared and the device stays offline if no one is around to redo the configuration page. I'm trying to understand how to save the credentials is wifimanager.autoconnect("SSID NAME") fails to try them again. Is that unnecessary? Should I store them in EEPROM after successful connection and then try to directly connect with them first before calling WiFimanager, or does the library have a better way of doing this? Right now I'm doing this, but it does not always work...

```
wifiManager.setTimeout(600);

String previousSuccessfulSSID = WiFi.SSID();
  String previousSuccessfulPWD = WiFi.psk();
  BWJ_DEBUG_PRINT_FLASH("Previous SSID: ");
  BWJ_DEBUG_PRINTLN(previousSuccessfulSSID);
  BWJ_DEBUG_PRINT_FLASH("Previous PWD: ");
  BWJ_DEBUG_PRINTLN(previousSuccessfulPWD);

  while (!wifiManager.autoConnect("STARTUP")){
    BWJ_DEBUG_PRINTLN_FLASH("failed to connect and hit timeout, now trying last successful");
    delay(3000);
    if (WiFi.status() == WL_CONNECTED) {
      BWJ_DEBUG_PRINTLN_FLASH("Connected after WM Timeout()");
      break;
    } 
    BWJ_DEBUG_PRINTLN_FLASH("BWJ Status:");
    BWJ_DEBUG_PRINTLN(WiFi.status());
    wl_status_t res;
    //check if we have ssid and pass and force those, if not, try with last saved values
    if (previousSuccessfulSSID != "") {
      WiFi.mode(WIFI_STA);
      //trying to fix connection in progress hanging
      #ifdef ESP8266
      ETS_UART_INTR_DISABLE();
      wifi_station_disconnect();
      ETS_UART_INTR_ENABLE();
      #else
      esp_wifi_disconnect();
      #endif
      res = WiFi.begin(previousSuccessfulSSID.c_str(), previousSuccessfulPWD.c_str(),0,NULL,true);
      if(res != WL_CONNECTED){
        BWJ_DEBUG_PRINTLN_FLASH("[ERROR] WiFi.begin res:");
        BWJ_DEBUG_PRINTLN(res);
      }
      int connRes = WiFi.waitForConnectResult();
      BWJ_DEBUG_PRINTLN_FLASH("Connection result: ");
      BWJ_DEBUG_PRINTLN( connRes );
      if (connRes == WL_CONNECTED) break;
    } else BWJ_DEBUG_PRINTLN_FLASH("No Previous Connection, retrying WifiManager");
  }
```

Djmartel, 2021-02-19:
I didn't know this was a potential issue. If it is, I need the solution also. Are you sure that it cleared the credentials? Could it have failed to connect to the WiFi network with the correct credentials? In my code if it fails to connect after a previous successful attempt I deep sleep for a bit and then try again.

tablatronix, 2021-02-19:
This is not a thing.

bwjohns4, 2021-02-19:
How does WiFiManager keep track of the credentials? If it fails to connect to the last known network, does it not clear the known credentials in the act of establishing the config portal? I'm guessing not. So then does it try the captive portal and if that fails, then it just had never cleared the previous credentials and then just tries those again?

tablatronix, 2021-02-19:
Credentials are never cleared, they are stored by the esp library itself. If you have issues with wiped credentials , there is a bug in esp lib, you have corrupt flash (do full erase) , or you are explicitly clearing them eg. WiFi.disconnect()

bwjohns4, 2021-02-19:
So should I just do  
`while (!wifiManager.autoConnect("STARTUP"))`

And leave it at that? Allowing periodic option to do portal, but if no connection just keep falling back and retrying previous credentials?

tablatronix, 2021-02-19:
Well no I would use a configportal timeout, or never start configportal unless necessary button press or no connect for some time

Wm now has the capability to do retries and connect timers, there are also some bugs posted that reconnects are failing first time and some other stuff

closed 2021-03-02 by bwjohns4