tzapu/WiFiManager #1217: help request

dyrvigk, 2021-02-26:
Hi
i have been using this great libary for some time but now i have a application where i need ethernet conetivity and i have made my own esp32 board with w5500 and Poe for power but i would like to be able to use wifimanager to setup my  mqtt . so basicaly i would like to use wifimanager in ap mode only and never connect to network  ethernet takes care of that . would that be possibel to keep ap and captive portal on always and stil being able to save my mqtt credentials and other setups 

kenn

EgHubs, 2021-04-01:
Can you try to set the esp to AP mode after the library autoConnect line of code? I think it might work as a workaround.

```
 if (!wifiManager.autoConnect(SSID, PASS)) {
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }
WiFi.softAP("ESPsoftAP_01", "pass-to-soft-AP");
```

and if you want you want to configure your AP, [here](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/soft-access-point-class.html#softapconfig)

tablatronix, 2021-04-01:
Thanks @EgHubs

tablatronix, 2021-04-01:
Yes use ondemand configportal exmaple start whatever you want