tzapu/WiFiManager #1220: (Question) Sometimes AP randomly comes back on

bwjohns4, 2021-03-02:
I've made a lot of customizations to Wifimanager and everything appears stable, however I see that sometimes the AP randomly starts broadcasting it's SSID well after WiFimanager has gone out of scope? Is this a known issue that sometimes happens, or am I doing something wrong. Is there a way to prevent that, or delete the AP configuration from the ESP8266 after Wifimanager runs? On the devices I've observed this on, it had a previous sketch on it from months ago, then I OTA updated it to newer firmware. Could this be an ESP bug that would go away with erase_flash? Any insight would be appreciated. Is there a way to load a new sketch and check if AP mode is already active or something like that?

tablatronix, 2021-03-02:
Set WiFi.mode in your sketch, the esp automatically starts aps

bwjohns4, 2021-03-02:
Wifimanager usually closes that right? And sets things back to STA mode before returning? Does that sometimes fail and get stuck? I call these two in setup() and in general the SSID goes away after wifimanager.autoConnect is complete. It just sometimes comes back randomly after days though. Why is it sporadic and not always that I see the SSID being broadcasted after startup? 
wifiManager.setTimeout(600);
wifiManager.autoConnect("MySSID");

I can just add WiFi.mode(WiFi_STA); after autoConnect() if that won't mess anything up. Just curious why this only sometimes happens...

tablatronix, 2021-03-02:
Add it before anything, it gets saved to flash
WM no longer clobbers persistent settings.

bwjohns4, 2021-03-02:
I still want WiFiManager to go into it's normal config portal mode when needed and broadcast it's SSID when needed. Will adding this before calling WM not prevent that from happening?

tablatronix, 2021-03-02:
Nope, thats just ESP default stuff

tablatronix, 2021-03-02:
I tried adding a way to force this for consistency with old versions, I guess its not working though

bwjohns4, 2021-03-02:
How does that work? ESP saves the mode in flash and WM starts up an AP without changing the mode and somehow doesn't tamper with the persistent settings in flash? Also, your way of forcing for consistency may still work, I have made lots of tweaks to my copy of WM although I intended to leave everything alone related to the WIFI management portion since I'm not too savvy on the underlying details of that.

tablatronix, 2021-03-02:
Are you using esp8266 or esp32 ?

bwjohns4, 2021-03-02:
esp8266

tablatronix, 2021-03-02:
Remind me what version you are using again?

bwjohns4, 2021-03-02:
It's off the master branch but I forked it about a month ago and started putting my own tweaks into it

tablatronix, 2021-03-02:
Oh ok so before dev was merged into master?

bwjohns4, 2021-03-03:
It was after ESP32 portion was merged into master, so I'm thinking that is what you mean about dev being merged into master. I made these modifications:

- Added handler for a logo image
- Added handler for a javascript file
- Added an input_type to the parameters class so I could add different types of html pieces such as <select> elements
- A few parameter constructor overloads to take my new parameters
- Then since my code size was pushing upper limits for a 1M chip, I commented out a bunch of the new features such as OTA and other stuff I wasn't using.

I left alone everything related to wifi connection management aspect.

tablatronix, 2021-03-03:
Ok I was looking into defining out some stuff, help strings, ota, info page stuff also , let me know how you managed those, I am looking into making html inputs better and some stuff you mentioned as discussed in other issues and I have some stashed branches I am playing with

bwjohns4, 2021-03-03:
Why did you recommend calling WiFi.mode(WiFi_STA) before WM.autoConnect()? Is there any reason not to call it after? I'd prefer the call after WM, just a way of ensuring and future proofing that after WM we always go back to WIFI_STA. Would it hurt to call it before AND after?

tablatronix, 2021-03-03:
technically you are only doing it once, once its saved it never changes, subsequent calls should be ignored by esp lib

hebertreis, 2021-03-14:
Using the version 2.x this did happen with me, after connected show a open network with SSID starting with `ESP-` plus 6 last digits from the mac address. 

Adding `WiFi.mode(WiFi_STA)` didn't work, Arduino IDE says that `WiFi_STA` variable is not declared. Try to look other ways to fix on STA mode but didn't work anyway. 

Doing the downgrade to 0.16 version this behavior stop and solved for me.

tablatronix, 2021-03-14:
Sorry its
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP