tzapu/WiFiManager #1222: Recover connectivity after disconnection

rhalaly, 2021-03-06:
I'm using the basic example code. But I've noticed that when the WiFi disconnected (router restart, out of range, etc.) the device doesn't try to reconnect to the network. And in order to reconnect I need to reboot the device manually.

Is there any way to implement retry mechanism? That will reconnect to the WiFi after disconnection or open AP if it fails?

invidia191, 2021-03-10:
I have tried in loop to add:

WiFiManager wifiManager;
wifiManager.autoConnect();

This connects as AP, how are the saved values stored ?

tablatronix, 2021-03-10:
esp32 ?

rhalaly, 2021-03-10:
My original question is about ESP8266. I'm not sure if `wifiManager.autoConnect();` should be placed in the `loop`... I just want a retry mechanism if any disconnection occurred (without opening AP again)

tablatronix, 2021-03-10:
esp8266 should auto reconnect by default

rhalaly, 2021-03-10:
The basic example does not reconnect after I shut down my router for couple of minutes... (With esp8266)

The device is not reachable / detectable after the disconnection and there is no AP. I have to reboot the device and then it connects again...

tablatronix, 2021-03-10:
hmm works for me, how do you know you are not connected?
are you checking status ?

rhalaly, 2021-03-10:
Actually, I already put the chip in a way that I don't have direct access to it. But I cannot see it on my router and also the AP is not working. I also added an http server and it also not reachable (static IP)

I don't have access to it, but I can control the power supply (😈). So after a reboot with power down / up, it works again.

The firmware is the basic example with http server. Nothing more (I don't have anything on the `loop`)

invidia191, 2021-03-10:
In my case, I have a ubidots connection in loop. I can not use the reconnection function from ubidots library because  I need the saved wifi values.

askarkurymbayev, 2021-03-11:
good day! I also ran into this problem, after turning off / turning on the electricity, the ESP8266 starts first, and the router after a couple of minutes after charging, in this regard, the esp8266 does not find the network and goes into AP mode, you can delay the search for your network in a couple of minutes so that the router has time turns on and esp8266 is connected?

tablatronix, 2021-03-11:
well the ap only starts when you call it, you dont want it starting all the time

tarontop, 2021-03-16:
> good day! I also ran into this problem, after turning off / turning on the electricity, the ESP8266 starts first, and the router after a couple of minutes after charging, in this regard, the esp8266 [...]

The problem I encountered is the same as you. I used non-blocking. How can I reconnect to Wi-Fi without restarting ESP8266?

chrisncc1701, 2021-03-16:
my problem is my router port keeps closing after hours or days making the esp8266 unavailable externally, but i can log on locally port 80 and this reopens the router port, otherwise i have to reboot the esp, is this the same issue

tablatronix, 2021-03-16:
hmm sounds like something else is not restarting, sounds like a different issue or a routing issue.

tarontop, 2021-03-17:
After the power is off (the ESP8266 and the router are both powered off), the router takes about 2 minutes to start up. The ESP8266 is first started and the router has not been started. The ESP8266 judges that there is no Wi-Fi available and automatically enters the AP mode. I have to restart the ESP8266 Will connect to the router. How can I connect to the router without restarting the ESP8266?

tarontop, 2021-03-17:
[image]
I added wifiManager.setConnectTimeout(120); it seems to have no effect, in fact it is only about 54 seconds, how can I make him wait for 120 seconds

tablatronix, 2021-03-17:
Let me check

tablatronix, 2021-03-17:
Shrug it works for me

```
*WM: Using last saved values, should be faster
*WM: Waiting for connection result with time out
*WM: 71 start ms
*WM: 120000
*WM: Connection timed out
*WM: 120062 duration
*WM: Connection result: 
*WM: 1
*WM: 
```

tablatronix, 2021-03-17:
Fixed in master

closed 2021-03-17 by tablatronix