tzapu/WiFiManager #1223: On Client Connect Callback?

OldGreyCells, 2021-03-06:
### Basic Infos

I notice the WiFiManager knows when a client connects to the Access Point:
```
*WM: [2] <- Request redirected to captive portal
```
Would it be possible to have an ```onAPClientConnectCallback```?

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266: 

Hardware:			D1 Mini

### Description

Problem description

### Settings in IDE

Module:  			Wemos D1

Additional libraries:

tablatronix, 2021-03-06:
possibly, you could also do loop check

uint8_t WiFiManager::WiFi_softap_num_stations(){
  #ifdef ESP8266
    return wifi_softap_get_station_num();
  #elif defined(ESP32)
    return WiFi.softAPgetStationNum();
  #endif
}

tablatronix, 2021-03-06:
Also look at wifievents

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiEvents/WiFiEvents.ino

OldGreyCells, 2021-03-06:
Thank you, I  will have a look at those - I'm trying to get the trigger to change the end user prompt from "Please connect to this WiFi Access Point" to "Please use this URL if the captive portal hasn't popped up on you phone" (cos I've only seen the elusive portal automatically once!) so I'll have to see where those WiFi 'Station Connected' events pop up while in AP Mode. Will report back!

tablatronix, 2021-03-06:
ahh, so you want to actually detect the captive portal.. hmm

OldGreyCells, 2021-03-06:
Thank you - it's all about the user experience with me! My first programming 'customer' broke *everything*, so I still think like he does. 
I thought the lack of captive portal failure may be because it's a 'saved' network?

tablatronix, 2021-03-07:
Sometimes that is also a reason, mostly on IOS you MUST forget ap to get it to come back up typically.

tablatronix, 2021-03-07:
You know I have been trying to identify captive portals also for ota, as upload forms do not work. RIght now it just says goto ip on the page..

OldGreyCells, 2021-03-07:
It would appear the callback I require is ```WiFi.onSoftAPModeStationConnected()``` - this is fired when a client connects to a bare (ie ```WiFiManager``` not installed, flash erased) ESP8266. It would seem an AP is created automatically by the core WiFi lib but if I go to 192.168.4.1 I get 'Connection Refused' rather than the expected 'Address unreachable' which implies something is running on the ESP... 
The ```WiFi.onSoftAPModeStationConnected()``` event has two members:
.aid - appears to be the number of the the client (ie 1 for the first, 2 for the second).
.mac - I haven't yet matched to the client mac but is different for each one...
Still digging.

OldGreyCells, 2021-03-07:
> let me think on it, a callback might be the best way

On the basis it looks like the ```WiFi.onSoftAPModeStationConnected()``` callback meets the requirement, feel free to close this without creating a callback (following the *nix 'do one thing well' principle').