tzapu/WiFiManager #1225: Combining with ubidots

invidia191, 2021-03-07:
### Basic Infos

#### Hardware

Esp8266/Esp32: esp8266 

Hardware: wemos d1 mini

Core Version:      	2.4.0, staging

### Description

Hello, i'm trying to combine with ubidots. Problem is all dots in serial monitor after ap config. 
I need to define the wifi settings, but there is the problem i think.

This is the problem i think (very noobish in scripting):

#define WIFINAME "WiFi.SSID()"   // Your SSID
#define WIFIPASS "WiFi.psk()"  // Your Wifi Pass

### Settings in IDE

Module: Wemos D1

Additional libraries: UbidotsESPMQTT

### Sketch

#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <UbidotsESPMQTT.h>

//define your default values here, if there are different values in config.json, they are overwritten.
char mqtt_server[40];
char mqtt_port[6] = "8080";
char blynk_token[34] = "YOUR_BLYNK_TOKEN";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

/****************************************
 * Define Ubidots Constants
 ****************************************/

#define TOKEN "***********************"     // Your Ubidots TOKEN
#define WIFINAME "WiFi.SSID()"   // Your SSID
#define WIFIPASS "WiFi.psk()"  // Your Wifi Pass

Ubidots client(TOKEN);

/****************************************
 * Auxiliar Ubidots Functions
 ****************************************/

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();
}

/****************************************
 * Main Functions
 ****************************************/

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);

#ifdef ARDUINOJSON_VERSION_MAJOR >= 6
        DynamicJsonDocument json(1024);
        auto deserializeError = deserializeJson(json, buf.get());
        serializeJson(json, Serial);
        if ( ! deserializeError ) {
#else
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
#endif
          Serial.println("\nparsed json");
          strcpy(mqtt_server, json["mqtt_server"]);
          strcpy(mqtt_port, json["mqtt_port"]);
          strcpy(blynk_token, json["blynk_token"]);
        } else {
          Serial.println("failed to load json config");
        }
        configFile.close();
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  //end read

  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 6);
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //set static ip
  //wifiManager.setSTAStaticIPConfig(IPAddress(10, 0, 1, 99), IPAddress(10, 0, 1, 1), IPAddress(255, 255, 255, 0));

  //add all your parameters here
  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_blynk_token);

  //reset settings - for testing
  //wifiManager.resetSettings();

  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  //wifiManager.setMinimumSignalQuality();

  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  //wifiManager.setTimeout(120);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("AutoConnectAP", "password")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //read updated parameters
  strcpy(mqtt_server, custom_mqtt_server.getValue());
  strcpy(mqtt_port, custom_mqtt_port.getValue());
  strcpy(blynk_token, custom_blynk_token.getValue());
  Serial.println("The values in the file are: ");
  Serial.println("\tmqtt_server : " + String(mqtt_server));
  Serial.println("\tmqtt_port : " + String(mqtt_port));
  Serial.println("\tblynk_token : " + String(blynk_token));

  //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");
#ifdef ARDUINOJSON_VERSION_MAJOR >= 6
    DynamicJsonDocument json(1024);
#else
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
#endif
    json["mqtt_server"] = mqtt_server;
    json["mqtt_port"] = mqtt_port;
    json["blynk_token"] = blynk_token;

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

#ifdef ARDUINOJSON_VERSION_MAJOR >= 6
    serializeJson(json, Serial);
    serializeJson(json, configFile);
#else
    json.printTo(Serial);
    json.printTo(configFile);
#endif
    configFile.close();
    //end save
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());

  client.ubidotsSetBroker("things.ubidots.com");  // Sets the broker properly for the business account
  client.setDebug(true);   
  client.wifiConnection(WIFINAME, WIFIPASS);
  client.begin(callback);

  
}

void loop() {
  // put your main code here, to run repeatedly:
 if (!client.connected()) {
    client.reconnect();
  }

  // Publish values to 2 different data sources

  client.add("stuff", 10.2);  // Insert your variable Labels and the value to be sent
  client.ubidotsPublish("source1");
  client.add("stuff", 10.2);
  client.add("more-stuff", 120.2);
  client.ubidotsPublish("source2");
  client.loop();
  delay(5000);
}

tablatronix, 2021-03-07:
Where is your logging ?

invidia191, 2021-03-07:
Is this log output  ?:

11:56:13.181 -> *WM: 1
11:56:13.228 -> *WM: 
11:56:13.228 -> *WM: Configuring access point... 
11:56:13.228 -> *WM: AutoConnectAP
11:56:13.228 -> *WM: password
11:56:13.697 -> *WM: AP IP address: 
11:56:13.697 -> *WM: 192.168.4.1
11:56:13.697 -> *WM: HTTP server started
11:56:26.519 -> *WM: Request redirected to captive portal
11:56:26.800 -> *WM: Request redirected to captive portal
11:56:36.907 -> *WM: Scan done
11:56:36.907 -> *WM: DUP AP: telenet-392
11:56:36.907 -> *WM: telenet-392-ext2
11:56:36.907 -> *WM: -42
11:56:36.907 -> *WM: telenet-392-modem
11:56:36.907 -> *WM: -58
11:56:36.907 -> *WM: TelenetWiFree
11:56:36.907 -> *WM: -59
11:56:36.907 -> *WM: telenet-392
11:56:36.907 -> *WM: -70
11:56:36.907 -> *WM: DIRECT-OdC43x Series
11:56:36.907 -> *WM: -81
11:56:36.907 -> *WM: TP-DEV
11:56:36.907 -> *WM: -88
11:56:36.907 -> *WM: Sent config page
11:56:48.238 -> *WM: WiFi save
11:56:48.238 -> *WM: Parameter
11:56:48.238 -> *WM: server
11:56:48.238 -> *WM: 
11:56:48.238 -> *WM: Parameter
11:56:48.238 -> *WM: port
11:56:48.238 -> *WM: 8080
11:56:48.238 -> *WM: Parameter
11:56:48.238 -> *WM: blynk
11:56:48.238 -> *WM: YOUR_BLYNK_TOKEN
11:56:48.238 -> *WM: Sent wifi save page
11:56:49.254 -> *WM: Connecting to new AP
11:56:49.254 -> *WM: Connecting as wifi client...
11:56:49.254 -> *WM: Status:
11:56:49.254 -> *WM: 0
11:56:49.348 -> *WM: [ERROR] WiFi.begin res:
11:56:49.348 -> *WM: 6
11:56:53.854 -> *WM: Connection result: 
11:56:53.854 -> *WM: 3
11:56:53.948 -> Should save config
11:56:53.948 -> connected...yeey :)
11:56:53.948 -> The values in the file are: 
11:56:53.948 -> 	mqtt_server : 
11:56:53.948 -> 	mqtt_port : 8080
11:56:53.948 -> 	blynk_token : YOUR_BLYNK_TOKEN
11:56:53.948 -> saving config
11:56:53.948 -> {"mqtt_server":"","mqtt_port":"8080","blynk_token":"YOUR_BLYNK_TOKEN"}local ip
11:56:53.995 -> 192.168.1.252
11:56:54.572 -> ...................................................

tablatronix, 2021-03-07:
11:56:49.254 -> *WM: Connecting as wifi client...
11:56:49.254 -> *WM: Status:
11:56:49.254 -> *WM: 0
11:56:49.348 -> *WM: [ERROR] WiFi.begin res:
11:56:49.348 -> *WM: 6
11:56:53.854 -> *WM: Connection result:
11:56:53.854 -> *WM: 3

Your wifi is not connecting

invidia191, 2021-03-07:
> 11:56:49.254 -> *WM: Connecting as wifi client...
> 11:56:49.254 -> *WM: Status:
> 11:56:49.254 -> *WM: 0
> 11:56:49.348 -> *WM: [ERROR] WiFi.begin res:
> 11:56:49.348 -> *WM: 6
> 11:56:53.854 -> *WM: Connection result:
> 11:56:53.854 -> *WM: 3
> 
> Your wifi is not connecting

But i am getting an ip address ?
When defining the wifi settings manually, no problem. But i want to use the saved settings to connect with ubidots. I think it is a problem with defining the wifi settings for ubidots.

tablatronix, 2021-03-07:
you can just skip this ?
  client.wifiConnection(WIFINAME, WIFIPASS);

invidia191, 2021-03-07:
> you can just skip this ?
> client.wifiConnection(WIFINAME, WIFIPASS);

Damn, that was simple.
Thx for the great help ! Awesome.