tzapu/WiFiManager #1226: ESP32-S2 Support

lienbacher, 2021-03-15:
Hello!

I'm trying to use WifiManager on a custom ESP32-S2 device and I am running into a compilation problem.
I am using platformio with following platformio.ini:

```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-v4.2
board = esp32-s2-saola-1
framework = arduino
board_build.mcu = esp32s2

lib_deps =
    https://github.com/tzapu/WiFiManager.git#development
```

Arduino support for ESP32-S2 is not stable yet, the issue may have to do with that (that's why i'm using a repo from platformio with arduino)

apart from a few warnings that I can fix (warnings regarding system vs arduino events, for some reason there's some ambiguity) there is one compilation error I am a bit stuck on and I don't quite understand what's happened here.

here's the compiler warning:
```
.pio/libdeps/esp32dev/WifiManager/WiFiManager.cpp: In member function 'void WiFiManager::WiFi_autoReconnect()':
.pio/libdeps/esp32dev/WifiManager/WiFiManager.cpp:2967:79: error: no matching function for call to 'WiFiClass::onEvent(std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type)'
       wm_event_id = WiFi.onEvent(std::bind(&WiFiManager::WiFiEvent,this,_1,_2));
                                                                               ^
In file included from /Users/username/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:28,
                 from /Users/wolfganglienbacher/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:32,
                 from .pio/libdeps/esp32dev/WifiManager/WiFiManager.h:63,
                 from .pio/libdeps/esp32dev/WifiManager/WiFiManager.cpp:13:
/Users/username/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:145:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
/Users/username/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:145:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type' {aka 'std::_Bind<void (WiFiManager::*(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>))(arduino_event_id_t, system_event_info_t)>'} to 'WiFiEventCb' {aka 'void (*)(arduino_event_id_t)'}
/Users/wolfganglienbacher/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:146:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventFuncCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventFuncCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
/Users/username/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:146:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type' {aka 'std::_Bind<void (WiFiManager::*(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>))(arduino_event_id_t, system_event_info_t)>'} to 'WiFiEventFuncCb' {aka 'std::function<void(arduino_event_id_t, arduino_event_info_t)>'}
/Users/wolfganglienbacher/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:147:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventSysCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
/Users/username/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:147:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type' {aka 'std::_Bind<void (WiFiManager::*(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>))(arduino_event_id_t, system_event_info_t)>'} to 'WiFiEventSysCb' {aka 'void (*)(arduino_event_t*)'}
```

The overloads for Wifi.onEvent are declared as follows in WiFiGeneric.h:
```
class WiFiGenericClass
{
  public:
    WiFiGenericClass();

    wifi_event_id_t onEvent(WiFiEventCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
    wifi_event_id_t onEvent(WiFiEventFuncCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
    wifi_event_id_t onEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
  ...
```

any pointers on whats going on and how I could fix this?

tablatronix, 2021-03-15:
Sure Ill take a look

lienbacher, 2021-03-16:
looking at the function that throws the error, what does the std::bind part actually do? I've never used it and have trouble following along

```CPP
void WiFiManager::WiFi_autoReconnect(){
  #ifdef ESP8266
    WiFi.setAutoReconnect(_wifiAutoReconnect);
  #elif defined(ESP32)
    // if(_wifiAutoReconnect){
      // @todo move to seperate method, used for event listener now
      DEBUG_WM(DEBUG_VERBOSE,"ESP32 event handler enabled");
      using namespace std::placeholders;
      wm_event_id = WiFi.onEvent(std::bind(&WiFiManager::WiFiEvent,this,_1,_2));
    // }
  #endif
}
```

lienbacher, 2021-03-16:
Regardless, I got it to compile by replacing all system_event references with arduino_event references. They seem to correlate, however I can't verify as I am still having some struggles with getting my code onto the flash memory of my S2. I'll report back once that's finished.

tablatronix, 2021-03-16:
https://github.com/tzapu/WiFiManager/tree/esp32s2

lienbacher, 2021-03-18:
I've tried including that branch in a fresh project and unfortunately it fails to compile looking for Update.h:
```PHP
In file included from .pio/libdeps/esp32dev/WiFiManager/WiFiManager.cpp:13:
.pio/libdeps/esp32dev/WiFiManager/WiFiManager.h:65:14: fatal error: Update.h: No such file or directory
```
it is not in my modified version of #develop, so I tried uncommenting `#include <Update.h>` and it then fails to find `WebServer.h`, which should be part of the arduino framework and is found fine in my original project. 

Any ideas what that could be?

tablatronix, 2021-03-18:
hmm not sure what is your exact build enviroment? pio ini ?
What does build out put say ?

It could be a bad pio ldf mode

lienbacher, 2021-03-18:
yeah it's weird, I've tripple checked everything i am aware of, all necessary paths are in `.vscode/c_cpp_properties.json` ...

pio.ini:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-v4.2

board = esp32-s2-saola-1
board_build.partitions = partition_map.csv
framework = arduino
board_build.mcu = esp32s2

monitor_speed = 115200
upload_speed = 921600

monitor_port = /dev/cu.SLAB_USBtoUART
upload_port = /dev/cu.SLAB_USBtoUART

lib_deps =
    https://github.com/tzapu/WiFiManager.git#esp32s2
```

output:
```PHP
Processing esp32dev (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-v4.2; board: esp32-s2-saola-1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s2-saola-1.html
PLATFORM: Espressif 32 (3.0.0+sha.c03dd11) > Espressif ESP32-S2-Saola-1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.beedeea 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32s2 1.80400.210211 (8.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFiManager> 2.0.4-beta+sha.a9507c9
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
Building in release mode
Archiving .pio/build/esp32dev/lib689/libWiFi.a
Indexing .pio/build/esp32dev/lib689/libWiFi.a
Compiling .pio/build/esp32dev/lib00b/WiFiManager/WiFiManager.cpp.o
Compiling .pio/build/esp32dev/FrameworkArduino/Print.cpp.o
Compiling .pio/build/esp32dev/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/esp32dev/FrameworkArduino/StreamString.cpp.o
Compiling .pio/build/esp32dev/FrameworkArduino/USB.cpp.o
In file included from .pio/libdeps/esp32dev/WiFiManager/WiFiManager.cpp:13:
.pio/libdeps/esp32dev/WiFiManager/WiFiManager.h:65:14: fatal error: Update.h: No such file or directory

****************************************************************
* Looking for Update.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Update.h"
* Web  > https://platformio.org/lib/search?query=header:Update.h
*
****************************************************************

     #include <Update.h>
              ^~~~~~~~~~
compilation terminated.
Compiling .pio/build/esp32dev/FrameworkArduino/USBCDC.cpp.o
Compiling .pio/build/esp32dev/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/esp32dev/FrameworkArduino/WString.cpp.o
Compiling .pio/build/esp32dev/FrameworkArduino/base64.cpp.o
Compiling .pio/build/esp32dev/FrameworkArduino/cbuf.cpp.o
Compiling .pio/build/esp32dev/FrameworkArduino/esp32-hal-adc.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/esp32-hal-bt.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/esp32-hal-cpu.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/esp32-hal-dac.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/esp32-hal-gpio.c.o
*** [.pio/build/esp32dev/lib00b/WiFiManager/WiFiManager.cpp.o] Error 1
========================================================================================= [FAILED] Took 2.66 seconds =========================================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

tablatronix, 2021-03-18:
This seems wrong 
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0

tablatronix, 2021-03-18:
It just stops there which is odd.

You might have to add verbose so it shows you the paths for the libs

```php
Processing esp32-s2-IDF (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-v4.2; board: esp32-s2-saola-1; framework: arduino; board_build.mcu: esp32s2; lib_extra_dirs: /Users/alverson/projects/microcontrollers/dev/libraries; upload_speed: 921600)
--------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s2-saola-1.html
PLATFORM: Espressif 32 (3.0.0+sha.c03dd11) (git+https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-v4.2) > Espressif ESP32-S2-Saola-1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.beedeea (git+https://github.com/espressif/arduino-esp32.git#idf-release/v4.2) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32s2 1.80400.210211 (8.4.0)
xtensa-esp32s2-elf-g++ -o "/Users/alverson/projects/microcontrollers/dev/libraries/WiFiManager/examples/DEV/OnDemandConfigPortal/OnDemandConfigPortal.ino.cpp" -x c++ -fpreprocessed -dD -E "/var/folders/6g/ljwft31n7r5g593rqjywgvmh0000gn/T/tmpM_daTR"
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 151 compatible libraries
Scanning dependencies...
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Dependency Graph
|-- <ArduinoOTA> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/ArduinoOTA)
|   |-- <ESPmDNS> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/ESPmDNS)
|   |   |-- <WiFi> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|   |-- <Update> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/Update)
|   |-- <WiFi> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|-- <WiFi> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|-- <WiFiManager> 2.0.4-beta (/Users/../projects/microcontrollers/dev/libraries/WiFiManager)
|   |-- <DNSServer> 1.1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/DNSServer)
|   |   |-- <WiFi> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|   |-- <WebServer> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/WebServer)
|   |   |-- <WiFi> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|   |   |-- <FS> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/FS)
|   |-- <ESPmDNS> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/ESPmDNS)
|   |   |-- <WiFi> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|   |-- <WiFi> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|   |-- <Update> 1.0 (/Users/../.platformio/packages/framework-arduinoespressif32/libraries/Update)
Building in release mode
```

lienbacher, 2021-03-18:
I just tried moving WifiManager into the libs folder and remove it from pio.ini and it compiled just fine, I don't know why 🤷‍♂️

tablatronix, 2021-03-18:
It is definitely your linker mode

try adding includes to your code and see if it changes

    #include <WiFi.h>
    #include <esp_wifi.h>

tablatronix, 2021-03-18:
try lib_ldf_mode=deep+

https://docs.platformio.org/en/latest/librarymanager/ldf.html

tablatronix, 2021-03-18:
are you including wifimanager.h properly?

lienbacher, 2021-03-18:
Oh boy now i am embarrassed. I did not include any code yet, just added the library. After including it the error went away 🤦 
sorry for the clutter.

There is one last warning left: 
```
.pio/libdeps/esp32dev/WiFiManager/WiFiManager.cpp: In member function 'void WiFiManager::WiFiEvent(arduino_event_id_t, arduino_event_info_t)':
.pio/libdeps/esp32dev/WiFiManager/WiFiManager.cpp:3498:20: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
   else if(event == SYSTEM_EVENT_SCAN_DONE){
                    ^~~~~~~~~~~~~~~~~~~~~~
```

replacing `SYSTEM_EVENT_SCAN_DONE` with `ARDUINO_EVENT_WIFI_SCAN_DONE` mutes the warning.

tablatronix, 2021-03-18:
OOPS

tablatronix, 2021-03-18:
I will merge this once I figure out how they suggest we handle these changes in 3.x and make them backward compatible..

lienbacher, 2021-03-18:
yeah i figured so, but that repo here is a great start!

I am still running into problems, just tried the basic example and this is the output: 
```PHP
*wm:[1] AutoConnect 
*wm:[1] No Credentials are Saved, skipping connect 
*wm:[2] Starting Config Portal 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  AutoConnectAP
*wm:[2] AP has anonymous access! 
[  1851][E][WiFiGeneric.cpp:983] mode(): Could not set mode! 12289
[  1857][E][WiFiAP.cpp:118] softAP(): enable AP first!
*wm:[0] [ERROR] There was a problem starting the AP 
*wm:[1] AP IP address:
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] NUM CLIENTS:  0
```

I have tried scanning networks which works fine, also manually connecting to wifi without wifimanager works just fine. If I disconnect and then run wifimanager it will reconnect to the previous AP just fine, so I guess the radio is working.

tablatronix, 2021-03-18:
maybe there is a special way to start ap on s2, interesting. It says enable AP first

lienbacher, 2021-03-18:
calling `WiFi.mode(WIFI_AP);` before starting wifi manager seems to fix this issue and for some extra fun, there's more issues popping up. Here's the output of me connecting to the AP and saving my wifi credentials.

I am having trouble reliably connecting to the ap I set, and there is 
```PHP
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x620
load:0x4004c000,len:0xa40
load:0x40050000,len:0x284c
entry 0x4004c190
*wm:[1] AutoConnect 
*wm:[1] No Credentials are Saved, skipping connect 
*wm:[2] Starting Config Portal 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  AutoConnectAP
*wm:[2] AP has anonymous access! 
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] NUM CLIENTS:  0
[ 56410][E][WebServer.cpp:630] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] NUM CLIENTS:  1
[ 61929][E][WebServer.cpp:630] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
[ 62175][E][WebServer.cpp:630] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] <- HTTP Wifi 
*wm:[2] WiFi Scan SYNC started 
*wm:[2] WiFi Scan completed in 4704 ms
*wm:[1] 1 networks found
*wm:[2] AP:  -61 my_wifi_ssid
[ 80316][E][WebServer.cpp:630] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] NUM CLIENTS:  1
*wm:[2] <- HTTP WiFi save  
*wm:[2] processing save 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] CONNECTED:
*wm:[1] Connecting to NEW AP: my_wifi_ssid
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_DISCONNECTED
*wm:[0] [ERROR] Connect to new AP Failed 
*wm:[2] Disabling STA 
[114376][E][WebServer.cpp:630] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] NUM CLIENTS:  1
```

after I pressed reset the first time it started up, had the credentials saved and connected fine to the wifi, however it is not reliably doing so after every restart. Plus those `request handler not found` messages are not supposed to be in there 🤔

lienbacher, 2021-03-18:
gotcha. makes sense. what's the preferred way of adding a retry? just restart the esp?

lienbacher, 2021-03-18:
ok I found `setConnectRetries()`, felt like i could go for 4 right away and coincidence had it, it took all 4 attempts to finally connect. Not entirely sure why 🤔 

```PHP
[ 26224][E][WebServer.cpp:630] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] NUM CLIENTS:  1
*wm:[2] <- HTTP WiFi save  
*wm:[2] processing save 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connect Wifi, ATTEMPT # 1 of 4
*wm:[1] CONNECTED:
*wm:[1] Connecting to NEW AP: my_wifi_ssid
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_DISCONNECTED
*wm:[1] Connect Wifi, ATTEMPT # 2 of 4
*wm:[1] CONNECTED:
*wm:[1] Connecting to NEW AP: my_wifi_ssid
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] Connect Wifi, ATTEMPT # 3 of 4
*wm:[1] CONNECTED: 1
*wm:[1] Connecting to NEW AP: my_wifi_ssid
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] Connect Wifi, ATTEMPT # 4 of 4
*wm:[1] CONNECTED: 1
*wm:[1] Connecting to NEW AP: my_wifi_ssid
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] Connect to new AP [SUCCESS] 
*wm:[1] Got IP Address: 
*wm:[1] 192.168.0.40 
[ 67295][E][WebServer.cpp:630] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] disconnect configportal 
*wm:[2] restoring usermode STA
*wm:[2] wifi status: WL_CONNECTED
*wm:[2] wifi mode: STA
*wm:[1] config portal exiting 
connected...yeey :)
```

I'll assemble another prototype and try another wifi to rule out a hardware fault.
thanks a lot for your support so far!

tablatronix, 2021-03-18:
No I added a setretry function

tablatronix, 2021-03-18:
Yup bug thanks!

tablatronix, 2021-03-18:
I fixed that, I think... untested, I also merged master in there were some updates

lienbacher, 2021-03-20:
Just a quick report, everything in the current ESP32-S2 repo seems to work just fine with two retries.

marcosmatilla, 2021-04-15:
Hi! I'm having the same problem as lienbacher when I build program. Any solution?

narath, 2021-04-17:
hey team, in doing a fresh development machine install, I updated to the latest espressif libraries and even with an ESP32 (i.e. not S2) I got the errors described here and was unable to build. By using the esp32s2 branch I was able to build and deploy. I'd give a +1 for merging with the main branch if possible since I think this is now an ESP32 general problem with incompatibility with the latest espressif libs.

tablatronix, 2021-04-17:
Sorry I only use platformio, I have not tried this on arduino.

What does your actual code look like?
have you tried including assets?

tablatronix, 2022-01-20:
I am pretty sure this can be closed now

closed 2022-03-02 by lienbacher