tzapu/WiFiManager #1228: how do I find out the IP Address esp8266 in MobilePhone

liwajaya, 2021-03-18:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32: 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
how do I find out the IP Address esp8266 connected to MobilePhone,
because it reads on the phone only the MAC address number.
and my client doesn't have a computer to track it down, he only has a phone connected to the ESP8266

Thanks
```

EgHubs, 2021-04-01:
So, you want to know the esp IP address from your phone!?, I guess that that has nothing to do with this library, it mainly depends on your phone, you should search that online and not post it here.

tablatronix, 2021-04-01:
I think they are asking after success connection, which is a known problem and many issues about it.