tzapu/WiFiManager #1229: WiFiManager.h: No such file or directory Platform.io

entrusc, 2021-03-19:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  0.16.0

Esp8266/Esp32: 

Hardware:			ESP32

Core Version:      	2.4.0, staging

### Description

I don't know if I'm doing something wrong, but I am unable to use the WifiManager with esp32dev board using Platformio. I am using the following platformio.ini 

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
upload_speed = 921600
lib_deps = 
	Button2@1.0.0
	bodmer/TFT_eSPI@^2.3.59
	tzapu/WiFiManager^0.16.0

```

and I get the following error message:

```
[...]
src/main.cpp:6:25: fatal error: WiFiManager.h: No such file or directory

*********************************************************************
* Looking for WiFiManager.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiManager.h"
* Web  > https://platformio.org/lib/search?query=header:WiFiManager.h
*
********************************************************************
```

I don't know if this is really a bug with this library, but I'm unable to find a solution for that. Any help would be appreciated. Thanks!

EgHubs, 2021-04-01:
I never used it but maybe [this](https://github.com/tzapu/WiFiManager#install-using-platformio) will help.

tablatronix, 2021-04-01:
I updated that page, thanks, master is now development
This should works, maybe a problem with pio lib manager, it should auto install deps

shellster, 2021-06-02:
I am also experiencing this issue.  I had to manually download and include `WiFiManager.h` and `strings_en.h` from master in my project before I could get it to compile.

tablatronix, 2021-06-02:
Even if you include wifimanager.h?

shellster, 2021-06-03:
Yeah.  I started with just:  `#include <WifiManager.h>` (besides Arduino.h) and adding this line in my platformio.ini:
`lib_deps = tzapu/WiFiManager@^0.16.0`.  When that didn't work, I went and tried to manually install the library in platformio.  It still didn't work.  I don't think it is too relevant, but I'm using an old NodeMCU with an ESP8266MOD chip.  It seems to work just fine when I included local file copies.

noweare1, 2021-06-09:
I am getting the error on PIO .pio/libdeps/esp32dev/WiFiManager/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory. 

After installing the library using platformio I compare the github wifimanager.h  file with the wifmanager.h file downloaded into my project by platformio and they are different. Have not tried just a manual install. The github wifimanager.h is the one that is correct don't know how pio is grabbing a different file. The fun never ends.

tablatronix, 2021-06-09:
Sounds like it doesn't know its esp32, Sounds like a platform issue

noweare1, 2021-06-09:
It is a platform issue. I cloned the repo to my project added #include in main and it all worked.

shellster, 2021-06-16:
My problem turned out to be a typo:
`#include <WifiManager.h> #Should be: <WiFiManager.h>`

Not sure where the typo came from, if I just fat fingered it or if it was incorrectly mentioned in some documentation somewhere.  Also uncertain why it worked when I cloned from the repo directly.  So to be explicit:

    lib_deps = 
        WiFiManager
    
And in my code:

`#include <WiFiManager.h>`

tablatronix, 2022-11-30:
I use  6.1.5, your enviroment or platform has an issue I would assume

jenschr, 2022-11-30:
This is in my ini and here you can see the problem:
[env:adafruit_feather_esp32s3_nopsram]
platform = espressif32
board = adafruit_feather_esp32s3_nopsram
framework = arduino
lib_deps = tzapu/WiFiManager@^0.16.0

For some reason, I have a two year old version as the only option to install. If I look at WiFiManager in the PIO Library browser, the 0.16.0 version is the most recent I can install. I can see from the changelog that this obviously is a very old version, but the version dropdown in the PlatformIO Libraries listing only allows me to select versions 0.15.0 and 0.16.0. That's the issue.

tablatronix, 2022-11-30:
WiFiManager@^0.16.0 does not support esp32

tablatronix, 2022-11-30:
lib_deps =
	https://github.com/tzapu/WiFiManager.git

jenschr, 2022-12-01:
That's great since it resolves my immediate issue, but it won't help anyone using the standard Library handling in PlatfromIO and it'll generate lots of unneeded issues here on Github. What would be required to make this work with the PlatformIO library manager? Is there aything I can help to make that "just work"?

jenschr, 2022-12-01:
According to the documentation (https://docs.platformio.org/en/latest/librarymanager/creating.html#publishing), one needs to use the "pio pkg publish" command in PlatformIO for every release to publish. According to https://registry.platformio.org/libraries/tzapu/WiFiManager this was last done two years ago. I think that's the root issue here.

tablatronix, 2022-12-01:
platformio library manager does not support beta realeases