tzapu/WiFiManager #1230: AutoConnect

MReed1959, 2021-03-19:
I could like to use WiFiManager instead of AutoConnect. But I can't. WiFiManager requires that you connect to the AP, then go to the browser and type in the IP address. I'm making something for an 86 YO, she can't do that.

The AutoConnect, while much more difficult to implement, will automatically go to the configure page as soon as you connect to the AP. 

Is there anyway to do that with WiFiManager?

## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32: 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

EgHubs, 2021-04-01:
Will, for me with WiFiManager library when I connect to the ESP AP it automatically opens the configuration page if I'm using my laptop (windows and mac), and with my phone, if I try to open any link with any browser it will automatically redirect you to the configuration page.

tablatronix, 2021-04-01:
You can start webmanager by itself now with cutting edge.

closed 2021-04-01 by tablatronix