tzapu/WiFiManager #1231: ESP32 compilation fails with ESP8266WiFi.h: No such file or directory

NetForces, 2021-03-22:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development (or any branch

Esp8266/Esp32: 

Hardware:			ESP32

Core Version:      	3.0.0

### Description

Compilation with platform.io always fails with 

```
.pio/libdeps/nodemcu-32s/WiFiManager/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory
```

### Settings in IDE

Module:  			nodemcu-32s

platform.io:

```
[platformio]
env_default = nodemcu-32s

[env:nodemcu-32s]
platform = espressif32 
board = nodemcu-32s
framework = arduino
lib_deps =
 tzapu/WiFiManager#Development
```

Additional libraries: 

### Sketch

Any of the examples provided.

### Debug Messages

```
courchea@X1-Andre:{8:46}~/Desktop/test ➭ platformio run
Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing nodemcu-32s (platform: espressif32; board: nodemcu-32s; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 (3.0.0) > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.210126 (1.0.4) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Converting main.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing tzapu/WiFiManager#Development
Library Manager: WiFiManager @ 0.16.0 has been installed!
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFiManager> 0.16.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
Building in release mode
Compiling .pio/build/nodemcu-32s/src/main.ino.cpp.o
Generating partitions .pio/build/nodemcu-32s/partitions.bin
In file included from /home/courchea/Desktop/test/src/main.ino:5:0:
.pio/libdeps/nodemcu-32s/WiFiManager/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************

compilation terminated.
Compiling .pio/build/nodemcu-32s/lib0be/WiFi/ETH.cpp.o
Compiling .pio/build/nodemcu-32s/lib0be/WiFi/WiFi.cpp.o
Compiling .pio/build/nodemcu-32s/lib0be/WiFi/WiFiAP.cpp.o
Compiling .pio/build/nodemcu-32s/lib0be/WiFi/WiFiClient.cpp.o
Compiling .pio/build/nodemcu-32s/lib0be/WiFi/WiFiGeneric.cpp.o
Compiling .pio/build/nodemcu-32s/lib0be/WiFi/WiFiMulti.cpp.o
*** [.pio/build/nodemcu-32s/src/main.ino.cpp.o] Error 1
================================================================================================================================ [FAILED] Took 2.15 seconds
```

NetForces, 2021-03-22:
I tried something. I removed the `lib_deps` from my `platformio.ini` and cloned `master` in the lib folder and now it compiles properly...

tablatronix, 2021-03-22:
development is no longer in use

benhamilto, 2021-03-30:
I think the version of the library on PIO may be out of date. I had the same issue with just `tzapu/WiFiManager` in `lib_deps`

Changing it to use the GitHub link directly fixed it.
```
lib_deps =
    https://github.com/tzapu/WiFiManager.git
```

tablatronix, 2021-03-31:
Makes sense esp32 support doesnt exist in release

Defozo, 2023-05-30:
Can we fix this?

closed 2023-05-31 by tablatronix