tzapu/WiFiManager #1232: Not able to set StaticIP in STA Mode

Abhesheksh, 2021-03-24:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32: ESP8266

Hardware:			ESP-01

Core Version:      	2.4.0, staging

### Description

I am trying to set a static IP once the module is connected to my wifi network so that I can set some additional values for the operation. The problem is that once the module is connected to my wifi, it shows a different localIP than the one I set in SetSTAStaticIPConfig. Please check my code below. I am setting Static IP as 192,168,1,99 but it shows 192.168.1.xx some random value. 

How do I fix this?

### Settings in IDE

Module:  			NodeMcu,

Additional libraries:  
#include <ESP8266WiFi.h>
#include "ESP8266Ping.h"
#include <WiFiManager.h>
#include <EEPROM.h>

### Sketch

#include <ESP8266WiFi.h>
#include "ESP8266Ping.h"
#include <WiFiManager.h>
#include <EEPROM.h>

#define LED 2
#define RELE 0
#define DEBUG True

// Update these with values suitable for your network.
WiFiManager wifiManager;
// Set web server port number to 80
WiFiServer server(80);
String header;

void setup() {
  
#ifdef DEBUG
  Serial.begin(115200);
#endif

  wifiManager.resetSettings();
  WiFi.mode(WIFI_STA);

  //WiFi.softAPConfig(IPAddress(192,168,1,19), IPAddress(192,168,1,1), IPAddress(255,255,255,0));
  // set custom ip for portal
  wifiManager.setAPStaticIPConfig(IPAddress(192,168,1,19), IPAddress(192,168,1,1), IPAddress(255,255,255,0));
 
  
  // fetches ssid and pass from eeprom and tries to connect
  // if it does not connect it starts an access point with the specified name
  // here  "AutoConnectAP"
  // and goes into a blocking loop awaiting configuration
  wifiManager.setConfigPortalTimeout(900);
  wifiManager.setConnectTimeout(900);
  wifiManager.setCountry("IN");
  wifiManager.setWiFiAPHidden(false);
  wifiManager.setSTAStaticIPConfig(IPAddress(192,168,1,99), IPAddress(192,168,0,1), IPAddress(255,255,255,0));
  wifiManager.autoConnect("MYESP_AP");

  
  if(wifiManager.autoConnect() == true)
  {
    Serial.println("TRUE - CONNECT");
  }
  else
  {
    Serial.println("FALSE - NO CONNECT");
  }

  
  // if you get here you have connected to the WiFi
  Serial.println("Connected.");
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());  

}

### Debug Messages

```
messages here
```

tablatronix, 2021-03-24:
Need serial logs

tablatronix, 2021-03-24:
```PHP
*wm:[1] AutoConnect 
*wm:[2] WiFiSetCountry to US
*wm:[2] [OK] esp_wifi_set_country:  US
*wm:[2] ESP32 event handler enabled 
*wm:[2] Connecting as wifi client... 
*wm:[2] Custom static IP/GW/Subnet/DNS 
*wm:[2] Custom STA IP/GW/Subnet 
*wm:[1] STA IP set: 10.0.1.99
*wm:[1] Connecting to SAVED AP: Shawn
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] AutoConnect: SUCCESS 
*wm:[1] STA IP Address: 10.0.1.99
```

seems to work for me

tablatronix, 2021-03-24:
hmm ok I see an issue
let me test some more

Abhesheksh, 2021-03-24:
*WM: [1] AutoConnect 
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  MYESP_AP
*WM: [1] Custom AP IP/GW/Subnet: 
*WM: [2] AP has anonymous access! 
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             MYESP_AP
*WM: [1] password:         
*WM: [1] ssid_len:         8
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.1.19
*WM: [2] esp_wifi_set_country: IN 
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.1.19
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2180 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.1.19 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] Scan is cached 26170 ms ago
*WM: [2] Portal Timeout In 899 seconds
*WM: [3] -> 192.168.1.19 
*WM: [2] <- HTTP Wifi 
*WM: [2] Scan is cached 28000 ms ago
*WM: [1] 6 networks found
*WM: [2] AP:  -60 MYWIFI
*WM: [2] AP:  -64 PanW
*WM: [2] AP:  -70 Ma2.4W
*WM: [2] AP:  -71 AirWI
*WM: [2] AP:  -83 Away
*WM: [2] AP:  -88 A918SF
*WM: [3] _staShowStaticFields 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [3] Sent config page 
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] Portal Timeout In 871 seconds
*WM: [2] Portal Timeout In 841 seconds
*WM: [2] Portal Timeout In 811 seconds
*WM: [2] WiFi Scan completed in 2180 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] Portal Timeout In 870 seconds
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.1.19 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] WiFi Scan completed in 2183 ms
*WM: [2] Portal Timeout In 897 seconds
*WM: [2] <- HTTP Wifi 
*WM: [2] Scan is cached 9 ms ago
*WM: [1] 7 networks found
*WM: [2] AP:  -59 MYWIFI
*WM: [2] AP:  -66 PanW
*WM: [2] AP:  -69 Ma2.4W
*WM: [2] AP:  -76 AirWI
*WM: [2] AP:  -78 Away
*WM: [2] AP:  -93 A918SF
*WM: [2] AP:  -95 DJHLE
*WM: [3] _staShowStaticFields 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [3] Sent config page 
*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
*WM: [3] static ip: 192.168.1.99
*WM: [3] static gateway: 192.168.0.1
*WM: [3] static netmask: 255.255.255.0
*WM: [3] Sent wifi save page 
*WM: [2] processing save 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP: 192.168.1.99
*WM: [2] Custom static IP/GW/Subnet/DNS 
*WM: [2] Custom STA IP/GW/Subnet 
*WM: [0] [ERROR] wifi config failed 
*WM: [1] CONNECTED:
*WM: [1] Connecting to NEW AP: MYWIFI
*WM: [3] Using Password: 12345678
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] Connect to new AP [SUCCESS] 
*WM: [1] Got IP Address: 
*WM: [1] 192.168.1.12 
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
*WM: [1] AutoConnect 
*WM: [2] esp_wifi_set_country: IN 
*WM: [1] AutoConnect: ESP Already Connected 
*WM: [3] STA static IP: 192.168.1.99
*WM: [2] Custom static IP/GW/Subnet/DNS 
*WM: [2] Custom STA IP/GW/Subnet 
*WM: [0] [ERROR] wifi config failed 
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.12
TRUE BOOLEAN CONNECT
Connected.
IP address:	192.168.1.12

tablatronix, 2021-03-24:
This is not a valid gateway maybe?
IPAddress(192,168,0,1)

tablatronix, 2021-03-24:
Gateway is invalid

try
IPAddress(192,168,1,1)

closed 2021-03-24 by tablatronix