tzapu/WiFiManager #1234: Parsing Parameter Value to Int

hitbyatruck, 2021-03-29:
I'm having some problems, creating a custom parameter to store an IP address on a variable on WiFiManager.

After following the examples, I get the Custom parameter to be printed by WiFiManager itself on the serial monitor, but once I try to use it as an IP address on the code or do a Serial.println I never get anything displayed.

I've tried getting and int from the custom parameter field to a variable and couldn't ever find a way to convert the char to an int accordingly as well.

I  can't find a way to get an ip address from a custom parameter and save it as and IPaddress type variable.

Any help would be very appreciated.

Thank you.

tablatronix, 2021-03-29:
Are you using the example that was made for this ?
https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnect/AutoConnectWithFSParametersAndCustomIP/AutoConnectWithFSParametersAndCustomIP.ino

hitbyatruck, 2021-03-29:
@tablatronix Thanks for your reply.

Actually as the Ip adress I need is for a connection to a ATEM mini pro, I was hoping to be able to just input the ATEM address on a variable of type IPAddress and pass it to the library that handles the connection.

As I was able to see the WiFiManager print out the values inserted on the Custom Parameter box, I assumed I could manipulate the value and pass it directly to the function which handles the connection with the ATEM mini pro.

I don't need the ESP32 to have a custom IP address, since the idea is to have 4 ESP32 Tally lights connected to the same wifi network, which makes no difference if they all get the IP Address from DHCP.

But as the ATEM mini pro connection library needs to have the ATEM mini pro IP address and since, in most cases, people use it with a dynamic IP Address, I need to be able to have the user insert their hardware IP address, inside the same network.

So the procedure would be:

1 - connect to each Tally light and configure the Wifi network being used.
2 - Include the IP Address of the ATEM mini pro, which should be connected by itself to the same Wifi network mentioned above.
3 - Pass the custom parameter with the ATEM mini pro IP address to a variable of type IPAddress in order for the M5Stick library to be able to establish a connection with the ATEM mini pro.

I have everything working correctly, except for the Custom parameter being passed to the variable.

So, just to clear a doubt I had with that example: I'll always need to use SPIFFS, in order to be able to use Custom Parameters or I can pass it directly to a variable?

tablatronix, 2021-03-29:
That is what the example does..

  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);

hitbyatruck, 2021-03-29:
I followed that line of code but kept getting errors.

`
WiFiManagerParameter custom_ATEMip("ATEMIP", "ATEM ip", ATEMip, 40);
WifiManager.addParameter(&custom_ATEMip);

IPAddress switcherIp(192, 168, 1, 91); //this should be the ATEMip variable
`
When trying to divide the 4 numbers by their own variables, which I tried in despair, trying to get the ip address from the custom parameters in 4 different variables to see if I was able to handle each number as an int, I kept getting an error saying something like "invalid conversion from 'const char*' to 'char' [-fpermissive]".

I'm away from my work PC right now, so I tried to replicate part of the code above, which is the part I'm struggling with and can't find a way out.

I can't understand the type of variable I get from ATEMip variable above and how to cast it to the switcherIp variable in order to make it work.

tablatronix, 2021-03-30:
IPAddress(255, 255, 255, 255)

hitbyatruck, 2021-03-30:
Will I need something like:

`ATEMip = custom_ATEMip.getValue();`

 in order to get the ip address on the switchIp variable?

tablatronix, 2021-03-30:
well you need to get it and save it somewhere yeah

hitbyatruck, 2021-03-30:
@tablatronix Thank you so much for your help. I'm gonna try to avoid using SPIFFS and ask for the ATEMip address on each connect, to force the request for the user to input the address on each connect, which may suit my needs better at this time.

tablatronix, 2021-03-31:
Just use the params fs example and save your ip to flash or spiffs. Then read it on boot and convert to ipaddress and pass to your application

post your code to a forum or something and you will get solutions for what you cannot get working

hitbyatruck, 2021-03-31:
I don't need to store the ip address on a permanent basis at this point. It may be suitable in further iterations of the program.

For now I'm only trying to accomplish this as in the code below, which has not been very sucessfull do far

```
//WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wm;

  //reset settings - wipe credentials for testing
  //wm.resetSettings();

  
  WiFiManagerParameter custom_ATEM_ip("ATEMip" , "ATEM ip" , ATEM_ip, 40);
  wm.addParameter(&custom_ATEM_ip);
  

  //IPAddress switcherIp(ATEM_IP);
  IPAddress switcherIp.fromString(ATEM_ip);
  AtemSwitcher.begin(switcherIp);
  AtemSwitcher.serialOutput(0x80);
  AtemSwitcher.connect();

```

tablatronix, 2021-03-31:
IPAddress switcherIp.fromString(ATEM_ip);
should work IF you are getvalue on ATEM_ip and you might have to memcopy it  since the param is a ref.

tablatronix, 2021-03-31:
see examples
  strcpy(mqtt_server, custom_mqtt_server.getValue());