tzapu/WiFiManager #1235: ERROR:The Content-Length header occurred more than once, at most one is allowed.

EgHubs, 2021-03-30:
#### Hardware
WiFimanager Branch/Release:  Master

Esp8266

Hardware:			ESP-12e

Core Version:      	2.7.4

### Description
I don't know if this is supposed to be posted here but when I try to make an HTTP request for the esp from my mobile application that I am developing while it's on AP mode and waiting for configuration i.e, `http://192.168.4.1 `
I get the following error
`The Content-Length header occurred more than once, at most one is allowed.
`
though I can communicate it with google chrome as normal.
keep in mind that when I try to communicate to the esp after its configuration like `http://192.168.1.200 ` it's all good though the same code from my flutter application.
I don't know what happens in there, I was hoping for any help by your side
Iam using the latest library release `0.16`
I am developing a flutter application and making an HTTP request with the HTTP [package](https://pub.dev/packages/http)
 
Module:  			NodeMcu

EgHubs, 2021-03-30:
What made me post it here is someone's comment on StackOverflow, he told me that 

> It sounds like your HTTP server is incorrectly sending the Content-Length header in the HTTP response.

tablatronix, 2021-03-31:
I will take a look can you try with master? Also provide serial log and which page are you requesting?

EgHubs, 2021-03-31:
I am requesting the configuration page with the following link:
`http://192.168.4.1/wifisave?s=myWifi&p=myPass&ip=192.168.1.203&gw=192.168.1.1&sn=255.255.255.0&email=new@test.com&pass=123456789&alarm=0`

and this is the log with the master branch (note: I am blocking the MCU from joining my router for testing):

```
WM: Configuring access point... 
*WM: M2
*WM: 62f105A5FI
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Handle root
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Parameter
*WM: alarm
*WM: 0
*WM: Parameter
*WM: email
*WM: new@test.com
*WM: Parameter
*WM: pass
*WM: 123456789
*WM: static ip
*WM: 192.168.1.203
*WM: static gateway
*WM: 192.168.1.1
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.203
*WM: Status:
*WM: 0
*WM: [ERROR] WiFi.begin res:
*WM: 6
*WM: Connection result: 
*WM: 4
*WM: Failed to connect.
Should save config heree in callback
*WM: Waiting for sta (10 secs) .......
```
it seems that the esp receives the HTTP request but it doesn't respond back or it responds back the Content-Length header incorrectly or something like that.

tablatronix, 2021-03-31:
Add it is only that page?
Does it do the same if not sending params as get arguments?

EgHubs, 2021-03-31:
Yes i get that error only in that page, i.e, when i try to make an http request to the esp  after configuration with a link like `192.168.1.203/` it's all good and okay and no error is catched

EgHubs, 2021-04-04:
Sorry but I am still stuck in this error, any news? :)

tablatronix, 2021-04-04:
Sorry I have not had a chance to look at it.

EgHubs, 2021-04-04:
No problem at all for sure, hope you have soon 😄

EgHubs, 2021-04-13:
Here again, still no progress on this error.
when I tried to get the response from an API tester this is what I get
```
> Host: 192.168.4.1
> User-Agent: insomnia/2021.2.2
> Content-Type: multipart/form-data; boundary=X-INSOMNIA-BOUNDARY
> Accept: */*
> Content-Length: 25

| --X-INSOMNIA-BOUNDARY--

* We are completely uploaded and fine
* Mark bundle as not supporting multiuse

< HTTP/1.1 200 OK
< Content-Type: text/html
< Content-Length: 1210
< Content-Length: 1210 //second one.
< Connection: close

```

see I get Content-Length twice here, the server sends the Content-Length twice which is the error title that appears for me. i tried more than one site and I only get the Content-Length once
for example
```
> GET / HTTP/1.1
> Host: github.com
> User-Agent: insomnia/2021.2.2
> Content-Type: multipart/form-data; boundary=X-INSOMNIA-BOUNDARY
> Accept: */*
> Content-Length: 25

| --X-INSOMNIA-BOUNDARY--

* We are completely uploaded and fine
* Mark bundle as not supporting multiuse

< HTTP/1.1 301 Moved Permanently
< Content-Length: 0
< Location: https://github.com/
< connection: close
```
keep in mind that the server receives the request but it responds back incorrectly.
I really hope that this makes it easier to solve the problem.

tablatronix, 2021-04-13:
crap I will look right now

tablatronix, 2021-04-13:
I can only assume that the webserver lib is sending this automatically.. I do not see anything in WM that would send this more than once, I will look into it further.

tablatronix, 2021-04-13:
```C++
// handlewifisave
  server->sendHeader(FPSTR(HTTP_HEAD_CL), String(page.length()));
  server->sendHeader(FPSTR(HTTP_HEAD_CORS), FPSTR(HTTP_HEAD_CORS_ALLOW_ALL));
  server->send(200, FPSTR(HTTP_HEAD_CT), page);
```

I wonder if its the CORS on that page?
Maybe remove those and see what happens

tablatronix, 2021-04-13:
Yup its definitely sending it twice

tablatronix, 2021-04-13:
It seems content-length was added in 2015 maybe it was broken , no idea why this has never been noticed, I can remove them and test on older esp versions and see if it still works.

tablatronix, 2021-04-13:
It also seems to support httpclient requests now like the async lib, so that can also be changed, as add header already does dup checking

tablatronix, 2021-04-13:
Try it now, I tried old versions and its fine, so I am just gonna leave these out now

EgHubs, 2021-04-13:
Woooh !!, nice now it's working perfectly :), I am so happy that I did help to contribute to something important like that in this great library, I also have no idea why this error appeared to me, maybe some updates on the Flutter side refused this error.
I would suggest a quick fix update with a new version like 0.16.1 or something.
again thank you for helping me and thank you again for being such an active person in this library community and helping people.

closed 2021-04-13 by EgHubs

tablatronix, 2021-04-13:
I would assume httpdclients should just ignore dup headers if they are identical, but who knows, technically it is against spec to send multiples

EgHubs, 2021-04-13:
> I would assume httpdclients should just ignore dup headers if they are identical, but who knows, technically it is against spec to send multiples

Wil, based on the [question](https://stackoverflow.com/q/66771165/13444299) I submitted on StackOverFlow someone submitted an [issue](https://github.com/dart-lang/sdk/issues/45679) requesting to ignore the error if the values are identical. like you are recommending.