tzapu/WiFiManager #1236: How to use push button interruupts with wifi manager

vxplore, 2021-04-03:
I am successfully using wifi manager in esp 01 board.
But now i want to add a push button for extra functionaliry like reset, manual operation etc.
But could not figure out how to use both(wifimanager and push button).

My current code is following

#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <EEPROM.h>
WiFiServer server(80);
String header;
int outputState = LOW;
const int outputPin = 2;

void setSavedState(int value)
{
  outputState = value;
  EEPROM.begin(4);
  EEPROM.write(0, value?'1':'0');
  EEPROM.write(1, 's');
  EEPROM.write(2, 'e');
  EEPROM.write(3, 't');
  EEPROM.commit();
  EEPROM.end();
}

int getSavedState()
{
  EEPROM.begin(4);
  if(
    EEPROM.read(1)=='s'&&
    EEPROM.read(2)=='e'&&
    EEPROM.read(3)=='t'
    )
    {
      return EEPROM.read(0)=='1'?HIGH:LOW;
    }
    else
    {
      return LOW;
    }
  
  EEPROM.end();
}

WiFiManager wifiManager;

void setup() {
  outputState = getSavedState();
  Serial.begin(115200);
  pinMode(outputPin, OUTPUT);
  digitalWrite(outputPin, outputState);
  wifiManager.autoConnect("VXSS000000001");
  Serial.println("Connected.");  
  server.begin();
}
String getStateText()
{
  if(outputState==LOW)
  {
    return "off";
  }
  else
  {
    return "on";
  }
}
bool erase_requested = false;
void loop(){
  WiFiClient client = server.available();
  if (client) 
  {
    Serial.println("New Client.");
    String currentLine = "";
    while (client.connected()) 
    {
      if (client.available()) 
      {
        char c = client.read();
        Serial.write(c);
        header += c;
        if (c == '\n') 
        {
          if (currentLine.length() == 0) 
          {
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println("Connection: close");
            client.println();
            if (header.indexOf("GET /load/on") >= 0) 
            {
              Serial.println("Load On");
              setSavedState(HIGH);
              digitalWrite(outputPin, outputState);
            } 
            else if (header.indexOf("GET /load/off") >= 0) 
            {
              Serial.println("Load Off");
              setSavedState(LOW);
              digitalWrite(outputPin, outputState);
            }
            else if (header.indexOf("GET /reset") >= 0) 
            {
              Serial.println("Erasing");
              erase_requested = true;
              break;
            }
            client.println("<!DOCTYPE html><html>");
            client.println("<head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">");
            client.println("<link rel=\"icon\" href=\"data:,\">");
            client.println("<style>html { font-family: Helvetica; display: inline-block; margin: 0px auto; text-align: center;}");
            client.println(".button { background-color: #195B6A; border: none; color: white; padding: 16px 40px;");
            client.println("text-decoration: none; font-size: 30px; margin: 2px; cursor: pointer;}");
            client.println("</style></head>");
            client.println("<body><h1>VXSS000000001</h1>");
            client.println("<p>Load - State " + getStateText() + "</p>");     
            if (getStateText()=="off") 
            {
              client.println("<p><a href=\"/load/on\"><button class=\"button\">ON</button></a></p>");
            } 
            else 
            {
              client.println("<p><a href=\"/load/off\"><button class=\"button button2\">OFF</button></a></p>");
            }

            client.println("<p><a href=\"/reset\"><button class=\"button\">RESET</button></a></p>");
            client.println("</body></html>");
            client.println();
            break;
          } 
          else 
          {
            currentLine = "";
          }
        } 
        else if (c != '\r') 
        {
          currentLine += c;
        }
      }
    }
    header = "";
    client.stop();
    Serial.println("Client disconnected.");
    Serial.println("");
  }
  if(erase_requested)
  {
    erase_requested = false;
    wifiManager.resetSettings();
    Serial.println("Erased");
    delay(100);
    wifiManager.disconnect();
    Serial.println("Disconnected");
    delay(100);
    ESP.restart();
    delay(1000);
  }
}

So what will be the solution

EgHubs, 2021-04-04:
Will, a push-button requires an input pin to read its state, not an output pin, 
if outputPin is the pin you are going to request a reset with then you should assign it as a digital input pin then use digitalRead to read its value.
I would recommend that you use this [example](https://github.com/tzapu/WiFiManager/blob/master/examples/OnDemand/OnDemandConfigPortal/OnDemandConfigPortal.ino) and justify it for your needs.

EgHubs, 2021-04-04:
Here is the code for you, pull TRIGGER_PIN which is pin 0 here to high with a pull-up 10k resistor, also connect the same pin to the ground through a push-button.
if you want to do a reset then click on the button for a second.
```
/**
 * OnDemandConfigPortal.ino
 * example of running the configPortal AP manually, independantly from the captiveportal
 * trigger pin will start a configPortal AP for 120 seconds then turn it off.
 *
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// select which pin will trigger the configuration portal when set to LOW
#define TRIGGER_PIN 0

int timeout = 120; // seconds to run for

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n Starting");
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
}

void loop() {
  // is configuration portal requested?
  if (digitalRead(TRIGGER_PIN) == LOW) {
      delay(500);
  if (digitalRead(TRIGGER_PIN) == LOW) {
  {
    wifiManager.resetSettings();
      ESP.restart();
  }
  } else {
    WiFiManager wm;

    //reset settings - for testing
    //wifiManager.resetSettings();

    // set configportal timeout
    wm.setConfigPortalTimeout(timeout);

    if (!wm.startConfigPortal("OnDemandAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.restart();
      delay(5000);
    }

    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");

  }

  // put your main code here, to run repeatedly:
}
```

tablatronix, 2021-04-04:
Since wm is blocking you will want to use the new non blocking it will be easier

audiofill42, 2021-05-31:
Hello,

I am having same situation with the Wifi Manager.

I am trying to create a project based on ESP8266 & Blynk to use a button and relay to command something.
I would like to have implemented the function of On demand using the reet button if possible or the project button (D0).

Did you solded your problem?
And if Yes, how? Can you share to me the result/solution?

Thanks in advance.