tzapu/WiFiManager #1237: cannot create an AP after autoConnect failed

yuan910715, 2021-04-07:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Release

Esp8266/Esp32:  

Hardware:			ESP8266

Core Version:      	Release v2.0.3-apha Development

### Description

I want to create an offline mode access point to provide some web pages after the autoConnect timeout, but after the autoConnect timeout, I can't search the offline mode access point, whether it's AP or AP_ STA mode, below it's a minimum version program, please help me , thanks

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch

```cpp
#BEGIN
#include <WiFiManager.h>

void setup() {
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFiManager wm;      
    wm.setDebugOutput(true);
    wm.setTimeout(30);
    bool res;
    Serial.println("auto_test.");
    res = wm.autoConnect("auto_test");
    if(!res) {        
       Serial.println("offline mode");     
       WiFi.mode(WIFI_AP_STA);
       IPAddress apIP(192, 168, 4, 1);
       WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
       WiFi.softAP("Offline");
    } 
    else {
       Serial.println("connected.");
    }
}

void loop() {
  // put your main code here, to run repeatedly:

}
#END
```

### Debug Messages

```
14:18:48.453 -> *WM: [2] setSTAConfig static ip not set, skipping 
14:18:48.453 -> *WM: [1] Connecting to SAVED AP: yuan_mi11
14:18:48.453 -> *WM: [3] Using Password: 123456
14:18:48.453 -> *WM: [3] WiFi station enable 
14:18:48.453 -> *WM: [3] enableSTA PERSISTENT ON 
14:18:48.453 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
14:18:57.199 -> *WM: [2] Connection result: WL_NO_SSID_AVAIL
14:18:57.199 -> *WM: [3] lastconxresult: WL_NO_SSID_AVAIL
14:18:57.199 -> *WM: [1] AutoConnect: FAILED 
14:18:57.199 -> *WM: [2] Starting Config Portal 
14:18:57.199 -> *WM: [3] WIFI station disconnect 
14:18:57.199 -> *WM: [3] WiFi station enable 
14:18:57.199 -> *WM: [2] Disabling STA 
14:18:57.199 -> *WM: [2] Enabling AP 
14:18:57.199 -> *WM: [1] StartAP with SSID:  auto_test
14:18:57.677 -> *WM: [2] AP has anonymous access! 
14:18:57.677 -> *WM: [1] SoftAP Configuration 
14:18:57.724 -> *WM: [1] -------------------- 
14:18:57.724 -> *WM: [1] ssid:             auto_test
14:18:57.724 -> *WM: [1] password:         
14:18:57.724 -> *WM: [1] ssid_len:         9
14:18:57.724 -> *WM: [1] channel:          1
14:18:57.724 -> *WM: [1] authmode:        
14:18:57.724 -> *WM: [1] ssid_hidden:     
14:18:57.724 -> *WM: [1] max_connection:   4
14:18:57.724 -> *WM: [1] country:          CN
14:18:57.724 -> *WM: [1] beacon_interval:  100(ms)
14:18:57.724 -> *WM: [1] -------------------- 
14:18:58.200 -> *WM: [1] AP IP address: 192.168.4.1
14:18:58.247 -> *WM: [3] setupConfigPortal 
14:18:58.247 -> *WM: [1] Starting Web Portal 
14:18:58.247 -> *WM: [3] dns server started with ip:  192.168.4.1
14:18:58.247 -> *WM: [2] HTTP server started 
14:19:00.392 -> *WM: [2] WiFi Scan completed in 2185 ms
14:19:00.392 -> *WM: [2] Config Portal Running, blocking, waiting for clients... 
14:19:13.818 -> *WM: [3] -> connectivitycheck.platform.hicloud.com 
14:19:13.818 -> *WM: [2] <- Request redirected to captive portal 
14:19:18.317 -> *WM: [2] Portal Timeout In 8 seconds
14:19:27.171 -> *WM: [1] config portal has timed out 
14:19:27.171 -> *WM: [3] configportal abort 
14:19:27.171 -> *WM: [2] disconnect configportal 
14:19:28.175 -> *WM: [2] restoring usermode STA
14:19:28.175 -> *WM: [2] WiFi Reconnect, was idle 
14:19:28.216 -> *WM: [2] wifi status: WL_DISCONNECTED
14:19:28.216 -> *WM: [2] wifi mode: STA
14:19:28.216 -> *WM: [1] config portal exiting 
14:19:28.216 -> offline mode
14:19:28.263 -> *WM: [3] unloading 

```

tablatronix, 2021-04-07:
You will want to use non blocking mode in master branch.

See the examples to keep running the ap or webserver all the time

yuan910715, 2021-04-08:
I mean, if autoConnect timeout , I want to create an access point and display some my own pages (not WiFi manager portal) ,  I change WiFi.mode(WIFI_AP_STA); and create softAP after timeout ,but I can't search the AP .
I've looked at the non blocking example you mentioned, and I found it's to keep the wifimanager portal page after connect failed.
Do you have any suggestions?  thanks.

tablatronix, 2021-04-08:
you have to start it WiFi.softAP() look at esp examples

yuan910715, 2021-04-09:
yes, I started AP use WiFi.softAP() , please see my code:

#include <WiFiManager.h>

void setup() {
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFiManager wm;      
    wm.setDebugOutput(true);
    wm.setTimeout(30);
    bool res;
    Serial.println("auto_test.");
    res = wm.autoConnect("auto_test");  
    if(!res) {        
       Serial.println("offline mode");     
       WiFi.mode(WIFI_AP_STA);
       IPAddress apIP(192, 168, 4, 1);
       WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
       **WiFi.softAP("Offline");**
    } 
    else {
       Serial.println("connected.");
    }
}

void loop() {
  // put your main code here, to run repeatedly:

}

If I didn't use wm.autoConnect ,I can see Offline AP successfully,  and use autoConnect , after timeout ,no AP , I really don't know why.

tablatronix, 2021-04-09:
oh I didnt see offline mode condition

tablatronix, 2021-04-09:
I would guess        WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0)); is a problem

yuan910715, 2021-04-09:
Hi 
I have deleted these two lines , but it still not work.
`    res = wm.autoConnect("auto_test");
    if(!res) {        
       Serial.println("offline mode");     
       WiFi.mode(WIFI_AP_STA);
    //  IPAddress apIP(192, 168, 10, 1);
      // WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
       WiFi.softAP("Offline");
    } 
`
and it works after I didn't use wm.autoConnect:

`res = false; //wm.autoConnect("auto_test");
    if(!res) {        
       Serial.println("offline mode");     
       WiFi.mode(WIFI_AP_STA);
    //  IPAddress apIP(192, 168, 10, 1);
      // WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
       WiFi.softAP("Offline");
    } `
So I believe that it is some code of wifiManger that affects the create AP after autoConnect timeout, and I also check wifiManager source code and try ,but still not solved . 
Please help me, thanks a lot.

tablatronix, 2021-04-09:
I will try to test this tomorrow

add a wm.disconnect(); before starting the ap see if that helps.
esp likes to get stuck when changing modes, might add a delay there also

yuan910715, 2021-05-10:
hello tablatronix
have you tested it?

tablatronix, 2021-05-10:
I have not, sorry forgot about this

JanithGan, 2021-06-20:
Yes, I also have the same problem. WiFi manager prevents creating an AP after configuring SSID Password with the portal. 
Need to restart the esp to resolve this. 
Any updates?

tablatronix, 2021-06-20:
Sorry I keep forgetting this, So does esp give you a AP create failed error?

JanithGan, 2021-06-28:
Nope, this is the debug output, It just says `softap config unchanged` even though I changed the Mode and SoftAP name. 

And the esp SoftAP shows as a `Hidden Network`.

### Debug Output:
```
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: Home WIFI
*WM: [3] Using Password: ********

WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
scandowifi evt: 8
ne
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with Home WIFI, channel 10
dhcp client start...
wifi evt: 0
ip:192.168.8.106,mask:255.255.255.0,gw:192.168.8.1
wifi evt: 3
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.8.106
wifi evt: 7
wifi evt: 7
wifi evt: 7

Access Point Deployment : [AP] softap config unchanged
Success!
AP IP : 192.168.4.1
AP SSID : 
```
Note that there is no SSID name even though I gave it in the below code.

### My Code:

```
wifiManager.autoConnect(config_ssid);
WiFi.mode(WIFI_AP_STA);
Serial.println("\nConfiguring Soft AP...");
Serial.print("Access Point Deployment : ");
Serial.println(WiFi.softAP("ESP AP", "12345678") ? "Success!" : "Failed!");
```

I have to reset the esp each time (Only if the wifi manager portal is accessed to change SSID passwords) to overcome this. 

Any suggestions?

tablatronix, 2021-06-28:
I do not have this issue, what happens if autoconnet fails and you have a portaltimeout set?

tablatronix, 2021-06-28:
Ok I think I have it now, let me play with it a bit

JanithGan, 2021-06-28:
Same issue for timeout bro.

### Debug Output

```
*WM: [2] Portal Timeout In 0 seconds
*WM: [1] config portal has timed out 
*WM: [3] configportal abort 
:ur 1
*WM: [2] disconnect configportal 
bcn 0
del if1
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: [2] restoring usermode STA
bcn 0
del if1
mode : sta(e8:db:84:9b:89:d3)
*WM: [2] WiFi Reconnect, was idle 
*WM: [2] wifi status: WL_IDLE_STATUS
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
wifi evt: 8
mode : sta(e8:db:84:9b:89:d3) + softAP(ea:db:84:9b:89:d3)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100

Configuring Soft AP...
Access Point Deployment : [AP] softap config unchanged
Success!
AP IP : 192.168.4.1
AP SSID : 
```

tablatronix, 2021-06-28:
try to add a delay after changing the mode

    WiFi.mode(WIFI_AP_STA);
    delay(1000);

tablatronix, 2021-06-28:
ok well that didn't work, hmm off to next idea

JanithGan, 2021-06-28:
Nope adding a delay does not help :-)

tablatronix, 2021-06-28:
There is some kind of corruption going on of the soft ap config...

Turn persistent off before doing your stuff, for some reason persistent flash is not initializing properly or has bad flash and needs a full erase.

```C++
     WiFi.persistent(false);
    WiFi.mode(WIFI_AP);
    Serial.println("\nConfiguring Soft AP...");
    Serial.print("Access Point Deployment : ");
    bool res = WiFi.softAP("ESP_AP", "12345678");    
    Serial.println(res  ? "Success!" : "Failed!");
    wm.debugSoftAPConfig();
```

JanithGan, 2021-06-29:
Thank you for your answer I will try it and let you know

tablatronix, 2021-11-03:
I have not researched this any further, this issue is about starting ap not reconnecting however

KyoshoTom, 2021-11-06:
Thank you. Would anyone know how to make this work using 2x Max7219 modules (8 in 1)? I wanted to add an extra for the text scrolling feature. I looked through the code, but I do not see any way to change this :( Am I missing something?
[image]