tzapu/WiFiManager #1238: OTA fail

mourazo, 2021-04-08:
#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP-32 nodeMCU

Core Version:      	2.4.0, staging

### Description

I'm trying to update the firmware with OTA library function, I call the config portal with the BOOT button, but when I'm trying to upload the sketch, I get the next message in the serial monitor: 
.Bad Size Given

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch

```cpp
#include "WiFi.h"
#include <HTTPClient.h> 
#include <stdio.h>
#include  <WiFiManager.h>

WiFiClient client;

const int PushButton=0; 

#define RELAY_ON 0    
#define RELAY_OFF 1

String sn = "1";
String snWifiName = "SequoRelay-";
char snHostName[20];

TaskHandle_t relaysT;
TaskHandle_t buttonT;

WiFiManager wifiManager;

const int ledInt = 1;

unsigned long timeout = 10000; // 10sec

void setup()
{
  
    Serial.begin(115200);

    snConverter();
    
    wifiManager.setHostname(snHostName);

    
 
    bool res;
   
    wifiManager.setClass("invert"); // dark theme
    wifiManager.setDebugOutput(false);
    // res = wm.autoConnect(); // auto generated AP name from chipid
    Serial.println(" Intentando conectar a la red WiFi anterior");
    res = wifiManager.autoConnect(snHostName,"password"); // password protected ap

     if(!res) {
        Serial.println(" No se pudo conectar");
        //ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println(" Conectado!");
        Serial.println(WiFi.localIP());
    }
    
    pinMode(PushButton, INPUT);
    digitalWrite(21, RELAY_OFF);
    digitalWrite(19, RELAY_OFF);
    pinMode(21, OUTPUT);
    pinMode(19, OUTPUT);
    

      xTaskCreatePinnedToCore(
            relaysTask, /* Function to implement the task */
            "relaysT", /* Name of the task */
            10000,  /* Stack size in words */
            NULL,  /* Task input parameter */
            1,  /* Priority of the task */
            &relaysT,  /* Task handle. */
            0); /* Core where the task should run */
      
      
      
      
      xTaskCreatePinnedToCore(
            buttonTask, /* Function to implement the task */
            "buttonT", /* Name of the task */
            10000,  /* Stack size in words */
            NULL,  /* Task input parameter */
            1,  /* Priority of the task */
            &buttonT,  /* Task handle. */
            1); /* Core where the task should run */
      

}

void relaysTask( void * pvParameters ){

  for(;;){
   activeRelays();
   delay(60000);
  }
}

void buttonTask( void * pvParameters ){

  for(;;){
         int Push_button_state = digitalRead(PushButton);
          if ( Push_button_state == LOW )
              {      
              Serial.println(" Entrando en el modo configuracion");
                bool res;     
          wifiManager.setHostname(snHostName);
          wifiManager.setClass("invert"); // dark theme
          wifiManager.setDebugOutput(false);
          wifiManager.setConfigPortalTimeout(60);
          // res = wm.autoConnect(); // auto generated AP name from chipid
          res = wifiManager.startConfigPortal(snHostName,"password"); 
      
           if(!res) {
              Serial.println(" Fallo al conectar");
              ESP.restart();
          } 
          else {
              //if you get here you have connected to the WiFi    
              Serial.println(" Conectado!");
              Serial.println(WiFi.localIP());
              ESP.restart();
             }
          }
      }
    delay(1000);   
}

void loop()
{

}

static void snConverter(){
  
  snWifiName += sn;
  strcpy(snHostName, snWifiName.c_str());
  
}

void activeRelays(){

  WiFi.begin();
  
  unsigned long startingTime = millis();
  
  while (WiFi.status() != WL_CONNECTED && (millis() - startingTime) < timeout)
  {
    delay(250);
  }

   if(WiFi.status() != WL_CONNECTED) {

    Serial.println(" No se puede conectar con la wifi...");
   
  }
  
  if(WiFi.status()== WL_CONNECTED){   
  
   // transfer
   HTTPClient http; 
   Serial.println(WiFi.localIP());

   http.begin("https://miapi.com/");
   http.addHeader("Content-Type", "application/json");
 
   int httpResponseCode = http.GET();
   String response = http.getString();
  
   Serial.println(httpResponseCode);
    Serial.println(response);
   
   http.end(); 
    
  
   if ( httpResponseCode == 200 ){
    
  
    if(response == "on"){
      digitalWrite(21, RELAY_ON);
      digitalWrite(19, RELAY_ON);
    }else{
      digitalWrite(21, RELAY_OFF);
      digitalWrite(19, RELAY_OFF);
    }
   }

      WiFi.disconnect(true);
   
  }
  

}
#END
```

### Debug Messages

```
messages here
```

tablatronix, 2021-04-08:
Are you using master branch with recent pull ?

tablatronix, 2021-04-08:
Worked for me, whats logs say? Maybe it is actually too big ?

mourazo, 2021-04-09:
I re-export my code and now it works, thanks.

closed 2021-04-09 by mourazo