tzapu/WiFiManager #1240: OTA fails with 1MB ESP8266

BernhardSchlegel, 2021-04-12:
### Basic Infos

OTA update fails (out of memory?) with `WRITE ERROR 4`.

#### Hardware
WiFimanager Branch/Release:  Master (pulled as of today)

Esp8266: 

Hardware:			Sonoff TH 16 (which is why I cant go for more Flash)

Core Version:      	2.4.0, staging

`esptool.py --port /dev/ttyUSB0 flash_id` yields (probably good practice for the issue template?)

```
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 60:01:94:07:01:2e
Uploading stub...
Running stub...
Stub running...
Manufacturer: e0
Device: 4014
Detected flash size: 1MB
```

### Description

OTA update fails

```
sleep disable
...........................................................................................................................................
*wm:[0] [ERROR] OTA Update WRITE ERROR 4
.*wm:[0] [ERROR] OTA Update WRITE ERROR 4
```

### Settings in IDE

platformio.ini

```
[env:sonoff_th]
platform = espressif8266
board = sonoff_th
framework = arduino
lib_deps = 
	paulstoffregen/OneWire@^2.3.5
	bblanchon/ArduinoJson@^6.17.2
	WiFiManager=https://github.com/tzapu/WiFiManager.git
monitor_speed = 115200

[common]
upload_port = /dev/ttyUSB0
```

### Sketch

To big to print, but it breaks down to:

Program size output 

```
   text    data     bss     dec     hex filename
 468804    5572   26256  500632   7a398 .pio/build/sonoff_th/firmware.elf
```

Build ouput 

```
RAM:   [====      ]  38.8% (used 31824 bytes from 81920 bytes)
Flash: [======    ]  62.3% (used 474372 bytes from 761840 bytes)
```

Project inspect

```
RAM 39% (31.3kB)
Flash 48% (494.8kB)
```

Top 5 files

139.7 KB | unknown
-- | --
30.5 KB | …WiFiManager/WiFiManager.cpp
8.0 KB | …ssl/bearssl/src/ec/ec_p256_m15.c
7.1 KB | …Json/JsonDeserializer.hpp
7.0 KB | …src/WiFiClientSecureBearSSL.cpp

### Problem

I don't know why all programm size numbers are different (at least I dont get them aligned). What I do know: The 139.7kB seem to be inevitable, as well as the 139.7kB from the WiFiManager - thus reducing bin size is not really an option. Is there a way I can tweak the partition scheme using `board_build.partitions`?  It would be absolutely awesome to get OTA to work!

tablatronix, 2021-04-12:
I have been trying to reduce the size of wm back down , its just a little too big for ota on 1M ( you need double the size for ota )

62.3% is 12% too much
I will work on it some more, but you can remove the help strings and turn debug off that will help.

BernhardSchlegel, 2021-04-13:
Hi tablatronix, thanks so much for the reply ! Your efforts are really appreciated (I couldn't find a buy me a coffee link...). I was aware of the 50% rule, that's why I tried to generate the binary and library size metrics. I was just hoping, that the latter (48% (494.8kB)) was the correct one.. 

Adding `#define WM_NODEBUG` on top of my main.cpp resulted into the same size (only the build output, since that seems to be the relevant)

```
RAM:   [====      ]  38.8% (used 31824 bytes from 81920 bytes)
Flash: [======    ]  62.3% (used 474372 bytes from 761840 bytes)
```

Questions:

1. Did I do that right?
2. I'm not sure what you mean by "removing help strings" - all I found in `strings_en.h` seemed necessary.

tablatronix, 2021-04-13:
hmm there is another issue discussing the same issue #1154 ?

BernhardSchlegel, 2021-04-14:
I came across this issue before I opened this. I'm maybe lacking oversight, but I was not able to find to find concrete instructions that will get me near to saving 12% flash anywhere.

tablatronix, 2021-11-03:
Yeah not sure, I hope to revisit this at some point, but just have not had time to see how to slim it down