tzapu/WiFiManager #1241: WiFiManager.cpp:1182:35: error: 'HTTPAuthMethod' has not been declared

fani1016, 2021-04-13:
Hello everyone,

I tried an example of WiFiManager Library on my NodeMCU, but i got following errors:

In file included from C:\USER\arduino-1.8.13\libraries\WiFiManager-master\WiFiManager.cpp:13:0:
C:\USER\arduino-1.8.13\libraries\WiFiManager-master\WiFiManager.h:34:2: warning: #warning "ARDUINO_ESP8266_RELEASE_2_3_0, some WM features disabled" [-Wcpp]
 #warning "ARDUINO_ESP8266_RELEASE_2_3_0, some WM features disabled" 
  ^
C:\USER\arduino-1.8.13\libraries\WiFiManager-master\WiFiManager.cpp: In member function 'void WiFiManager::handleRequest()':

C:\USER\arduino-1.8.13\libraries\WiFiManager-master\WiFiManager.cpp:1182:35: error: 'HTTPAuthMethod' has not been declared
     server->requestAuthentication(HTTPAuthMethod::BASIC_AUTH); // DIGEST_AUTH
                                   ^
C:\USER\arduino-1.8.13\libraries\WiFiManager-master\WiFiManager.cpp: In member function 'String WiFiManager::getInfoData(String)':

C:\USER\arduino-1.8.13\libraries\WiFiManager-master\WiFiManager.cpp:2028:44: error: 'class ESP8266WiFiClass' has no member named 'softAPSSID'
     p.replace(FPSTR(T_1),htmlEntities(WiFi.softAPSSID()));

Kindly help me solving this problem.

Thanks.

tablatronix, 2021-04-13:
ok let me look, you are clearly using a very old version of the esp lib

fani1016, 2021-04-13:
Yeah, i updated and everything works fine now.
Thanks for your reply.

closed 2021-04-13 by fani1016

tablatronix, 2021-04-13:
I added a fix that might prevent this

peeta-rudy, 2021-11-17:
Hi I use the newest version of wifimanager 2.0.5-beta and I got the same error. 
Please help me solving this problem.
Thanks

Alives, 2024-08-09:
It looks like code was added to enable authentication then was bypassed.  What is the status with authentication?  I'd like to protect the `/params` and `/update` pages.

tablatronix, 2024-08-10:
I think the only issue was that it doesnt work in captive portal browsers, So instead of trying to restrict this, we would benefit more from html auth

commit 4d37843a (2021-04-13): fix #1241 maybe