tzapu/WiFiManager #1243: Question: how to avoid connecting to Wifi after saving on config portal?

debackel, 2021-04-21:
Hello,
This library is great, many thanks to the development community !!!

I am using the config portal to set up the SSID/password.
But the library automatically tries to connect after saving the wifi parameters.

How can I avoid it (I would like to manage the connection in a specific way in my state machine)?

Many thanks for the help.

Laurent

tablatronix, 2021-04-21:
hmm we use the native esp.begin to save and connect, nothing is implemented to save directly to flash. It would require a bit of programming

debackel, 2021-04-22:
Many thanks for your response.
If I understand correctly, you are using the esp.begin to do both saving the settings in flash memory and to connect to the AP. Is it correct?
 So, if we don't connect, the settings modified by the config portal are not saved to flash memory.
Do you know if there another easy way to save the settings in flash memory without connecting?

tablatronix, 2021-04-22:
ahhh, yeah I forgot about that

tablatronix, 2021-04-22:
ok give that a try, I have no tested it, and I am not sure how to handle the save returns, it still returns status, so you will have to also handle that or use breakafterconfig etc.

tablatronix, 2021-04-22:
This should only affect saving new creds, saving the same is ignored allways anyway..

debackel, 2021-04-22:
Thank you so much !!!!
I'll give it a try as soon as possible.

On your concern about ignoring the same credentials, why is this an issue? why would you want to save the same thing as what is stored in flash already? I probably didn't get the point.

Questions: the way the patch is done, won't you have a "WL_CONNECTED" returned (even without being connected)? Isn't this an issue? Also, the debug output will display "Connect to new AP ...", isn't it?
Shouldn't we output differentiated debug output and return code?

tablatronix, 2021-04-22:
I use the same wificonnect code for all , I would have to refactor this more and do not plan on doing that atm, as I am about cut a beta

tablatronix, 2021-04-22:
ok going to try returning idle and added alt debug

debackel, 2021-04-23:
I did a test this morning (not deep testing, but quick validation), this works great on my D1 Mini.
Thank you so much !
This way, I can manage myself the connection cycle.

debackel, 2021-04-24:
I made deeper testing today. The patch is functional. However, the debug output at different levels is not adequate anymore, as the methods were originally designed for connecting (not for only saving credentials). Also, in processConfigPortal, we expect connectWifi  to  return  WL_CONNECTED from the wifi.begin, while now, "idle" can be returned as well.
As a consequence, processConfigPortal goes to the error branch, and the config portal is not shutdown properly. I will debug further and hopefully propose a patch to have all this treated properly.

tablatronix, 2021-04-24:
Are you using non blocking or blocking?

Did you try using shouldbreakafterconfig ?

debackel, 2021-04-24:
I am using non blocking mode. No, I don't use shouldbreakafterconfig. Can you tell me what this actually does?

tablatronix, 2021-04-24:
It looks like wifi begin returns false also

debackel, 2021-04-24:
> It looks like wifi begin returns false also

I am not sure thie is wifi.begin. wifi.begin returns a value. But in wifimanager, the return code of wifi.begin is tested like this: if(!ret) DEBUG_WM(DEBUG_ERROR,F("[ERROR] wifi begin failed"));
Ar ret = 1 in case of connection, this is OK. But if you don't connect, it returns another value (idle, I guess), which doesn't pass the boolean type of testing.

tablatronix, 2021-04-24:
Yeah I just noticed connectWifi returns conres, I missed that

tablatronix, 2021-04-24:
I tried fixing it but now I am crashing my esp... ugh

debackel, 2021-04-24:
Do you want me to write a patch? You would review it, of course.

tablatronix, 2021-04-24:
Somehow this is exposing some bug somewhere else.. or in esp lib
NO clue, I will have to decode this first

```php
*wm:[1] SAVED with no connect to new AP 
*wm:[2] disconnect configportal 
*wm:[2] restoring usermode STA
*wm:[2] wifi status: WL_CONNECTED
*wm:[2] wifi mode: STA

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4020c35b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000050 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc40 end: 3fffffc0 offset: 0190
3ffffdd0:  00000000 4bc6a7f0 0000cac0 3ffe8831  
```

tablatronix, 2021-04-24:
all I changed was

```C++
        if (connectWifi(_ssid, _pass, _connectonsave) == WL_CONNECTED || (!_connectonsave)) {
```

tablatronix, 2021-04-24:
Ok I added some fixes, if this doesn't work, I will have to reimplement the state machine as this has been kludged too much

debackel, 2021-04-25:
Many thanks. 
Re tested today, seems OK, I think we can close it. This is great stuff, thank you once again  !!!

closed 2021-04-25 by debackel