tzapu/WiFiManager #1246: I am getting errors while compiling basic example.

bkrajendra, 2021-05-01:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp32: 

Hardware:			NodeMCUEsp32

Core Version:      	using latest master from git arduino-esp32

### Description

Problem description"

I am getting errors while compiling basic example.

```
D:\Arduino\libraries\WiFiManager\WiFiManager.cpp: In member function 'void WiFiManager::WiFiEvent(arduino_event_id_t, system_event_info_t)':
D:\Arduino\libraries\WiFiManager\WiFiManager.cpp:3505:17: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
     if(event == SYSTEM_EVENT_STA_DISCONNECTED){
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\WiFiManager\WiFiManager.cpp:3522:20: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
   else if(event == SYSTEM_EVENT_SCAN_DONE){
                    ^~~~~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\WiFiManager\WiFiManager.cpp: In member function 'void WiFiManager::WiFi_autoReconnect()':
D:\Arduino\libraries\WiFiManager\WiFiManager.cpp:3539:79: error: no matching function for call to 'WiFiClass::onEvent(std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type)'
       wm_event_id = WiFi.onEvent(std::bind(&WiFiManager::WiFiEvent,this,_1,_2));
                                                                               ^
In file included from D:\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiSTA.h:28,
                 from D:\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFi.h:32,
                 from D:\Arduino\libraries\WiFiManager\WiFiManager.h:65,
                 from D:\Arduino\libraries\WiFiManager\WiFiManager.cpp:13:
D:\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiGeneric.h:145:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
D:\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiGeneric.h:145:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type' {aka 'std::_Bind<void (WiFiManager::*(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>))(arduino_event_id_t, system_event_info_t)>'} to 'WiFiEventCb' {aka 'void (*)(arduino_event_id_t)'}
D:\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiGeneric.h:146:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventFuncCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventFuncCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
D:\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiGeneric.h:146:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type' {aka 'std::_Bind<void (WiFiManager::*(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>))(arduino_event_id_t, system_event_info_t)>'} to 'WiFiEventFuncCb' {aka 'std::function<void(arduino_event_id_t, arduino_event_info_t)>'}
D:\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiGeneric.h:147:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventSysCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
D:\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiGeneric.h:147:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type' {aka 'std::_Bind<void (WiFiManager::*(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>))(arduino_event_id_t, system_event_info_t)>'} to 'WiFiEventSysCb' {aka 'void (*)(arduino_event_t*)'}
Multiple libraries were found for "WiFi.h"

```
### Settings in IDE

Module:  			NodeMcuESP32

### Sketch

```cpp
#BEGIN
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    WiFi.mode(WIFI_STA); 
    Serial.begin(115200);
    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wm;
    bool res;

    res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

    if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        Serial.println("connected...yeey :)");
    }
}

void loop() {
    
}
#END
```

The same example compiles ok with esp8266.

tablatronix, 2021-05-01:
What esp32 lib version?

tablatronix, 2021-05-02:
The arguments were changed for events in new idf, The S2 branch should have a fix already

bkrajendra, 2021-05-02:
Yes. Got it working using S2 branch. But still not showing of AP. Need to test more. will get back once I get enough testing.

rlhelinski, 2021-09-10:
For me, the fix was to revert to version 1.0.6 of the esp32 board package rather than trying to use 2.0.0.

tablatronix, 2021-09-10:
Can you test the S2 branch, I am about to merge it when I know its not breaking anything, should work for all sdk versions

rlhelinski, 2021-09-10:
I can try. Is there a way to get a git clone (with a particular branch checked out) in the Arduino IDE?

tablatronix, 2021-09-10:
Yes of course, but it can be a pain, dont worry about it then.

CYR15000, 2021-10-30:
Hi Gentleman,
i have the same problem 
Os : MacOs 12.0.1
Arduino IDE : 1.8.16
ESP Library 2.0.0 (test but niet)
ESP Library : DFRobot  (test but niet)
ESP : Firebeetle ESP32-E
Error :
iFiManager.cpp: In member function 'void WiFiManager::WiFiEvent(arduino_event_id_t, system_event_info_t)':
/Users/cgironde/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:3505:17: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
     if(event == SYSTEM_EVENT_STA_DISCONNECTED){
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

King Regards

ccadic, 2021-11-09:
I have the issue when trying to compile mainlessmesh on an ESP32S2 dev boartd . The wifi.h complains

In file included from C:\Users\ccadic\Documents\Arduino\libraries\Painless_Mesh\src/painlessMesh.h:22,
                 from C:\Users\ccadic\Documents\Arduino\libraries\Painless_Mesh\examples\basic\basic.ino:9:
C:\Users\ccadic\Documents\Arduino\libraries\Painless_Mesh\src/arduino/wifi.hpp: In member function 'void painlessmesh::wifi::Mesh::eventHandleInit()':
C:\Users\ccadic\Documents\Arduino\libraries\Painless_Mesh\src/arduino/wifi.hpp:251:22: error: 'SYSTEM_EVENT_SCAN_DONE' is not a member of 'arduino_event_id_t'
         WiFiEvent_t::SYSTEM_EVENT_SCAN_DONE);
                      ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\ccadic\Documents\Arduino\libraries\Painless_Mesh\src/arduino/wifi.hpp:260:22: error: 'SYSTEM_EVENT_STA_START' is not a member of 'arduino_event_id_t'
         WiFiEvent_t::SYSTEM_EVENT_STA_START);
                      ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\ccadic\Documents\Arduino\libraries\Painless_Mesh\src/arduino/wifi.hpp:271:22: error: 'SYSTEM_EVENT_STA_DISCONNECTED' is not a member of 'arduino_event_id_t'
         WiFiEvent_t::SYSTEM_EVENT_STA_DISCONNECTED);
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\ccadic\Documents\Arduino\libraries\Painless_Mesh\src/arduino/wifi.hpp:281:22: error: 'SYSTEM_EVENT_STA_GOT_IP' is not a member of 'arduino_event_id_t'
         WiFiEvent_t::SYSTEM_EVENT_STA_GOT_IP);
                      ^~~~~~~~~~~~~~~~~~~~~~~

tablatronix, 2021-11-09:
fixed in git, ill push a beta right now

ccadic, 2021-11-09:
yes. latest  Painless_Mesh (yesterday's version) latest arduino IDE (yersterday's stable version). Looks like the issue is in wifi. 
shall I load the wifimanager  from this git ?

tablatronix, 2021-11-09:
I just pushed v2.0.5-beta

zekageri, 2022-03-08:
Same problem with etnernet:

```C
void ethEvent(WiFiEvent_t event){
    if( event == SYSTEM_EVENT_ETH_START ){
        ETH.setHostname(hsh_fileSystem.config.hostname);
    }else if( event == SYSTEM_EVENT_ETH_GOT_IP ){
        Serial.printf("\nConnected to Ethernet\n");
        Serial.printf("NS - IP:\t%s\n",          ETH.localIP().toString().c_str());
        Serial.printf("NS - MAC:\t%s\n",         ETH.macAddress().c_str());
        Serial.printf("NS - Host:\t%s\n",        ETH.getHostname());
    }else if( event == SYSTEM_EVENT_ETH_DISCONNECTED ){
        Serial.println("\nNS - Ethernet disconnected");
    }
}
```

Compiler warnings about `comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t'` and crash when got an event interrupt.

```C
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40091ff8  PS      : 0x00060d33  A0      : 0x80093e34  A1      : 0x3ffdc690  
A2      : 0xdf010000  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060d23
A6      : 0x00060d20  A7      : 0x0000cdcd  A8      : 0x0000cdcd  A9      : 0xffffffff
A10     : 0x3ffe116c  A11     : 0x00000004  A12     : 0x00000004  A13     : 0x3ffb9270
A14     : 0x9f810000  A15     : 0x003fffff  SAR     : 0x00000015  EXCCAUSE: 0x0000001d
EXCVADDR: 0xdf010000  LBEG    : 0x40102b08  LEND    : 0x40102b4c  LCOUNT  : 0x0000003c

Backtrace:0x40091ff5:0x3ffdc6900x40093e31:0x3ffdc6d0 0x400940ad:0x3ffdc6f0 0x4008423a:0x3ffdc710 0x400d387a:0x3ffdc740 0x400d5f29:0x3ffdc760 0x400d6434:0x3ffdc800 0x400d6f33:0x3ffdc830 
  #0  0x40091ff5:0x3ffdc6900 in compare_and_set_native at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:288

ELF file SHA256: 0000000000000000

Rebooting...
```

If i replace the system events ( that i used before without problem ) like  `SYSTEM_EVENT_ETH_START `, `SYSTEM_EVENT_ETH_GOT_IP ` and `SYSTEM_EVENT_ETH_DISCONNECTED` with the new `ARDUINO_EVENT_ETH_START `, `ARDUINO_EVENT_ETH_GOT_IP ` and `ARDUINO_EVENT_ETH_DISCONNECTED ` i also got a crash but different:

```C
assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == 
SPINLOCK_FREE)

Backtrace:0x40083e1d:0x3ffdc5500x4008ee6d:0x3ffdc570 0x400944a5:0x3ffdc590 0x4009209f:0x3ffdc6c0 0x40093e31:0x3ffdc700 0x400940ad:0x3ffdc720 0x4008423a:0x3ffdc740 0x400d3872:0x3ffdc770 0x400d5f21:0x3ffdc790 0x400d642c:0x3ffdc830 0x400d6f2b:0x3ffdc860
  #0  0x40083e1d:0x3ffdc5500 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402

ELF file SHA256: 0000000000000000

Rebooting...
```

tablatronix, 2022-03-08:
There is a check in there to use the correct events, perhaps it is not working on your enviroment? 
What version of IDf are you using, are you using arduino ?

Chicco7, 2022-03-31:
Hi, about the problem of "onEvent" in the 2.0x framework I found this:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino
Use "ARDUINO_EVENT_WIFI_STA_DISCONNECTED" for the WiFi disconnection event. (Follow the link for the other event).
For me it work fine

zekageri, 2022-03-31:
It was my bad. Iam using arduino esp32 but i used some old version and `the spinlock_acquire` crash was about some other part of my program. The new events working fine.

zekageri, 2022-04-13:
So. I updated the package and the the ETH events but my esp do not get a gotip event sometimes. It looks like when i perform a software restart, only the connection event fires, but not the gotip event. The esp is connected and got an ip however

tablatronix, 2022-04-13:
that sounds like the esp32 double hitter bug