tzapu/WiFiManager #1247: Using WiFiManager with SPIFFS

Schwenkner, 2021-05-06:
I've had a look into the samples using SPIFFS for saving additional parameters.
But I cannot getting it properly work for me. May be I'm doing something wrong.

Since I want to be able to access the config page also when the application is running, I'm doing
`
  if (wifiManager.autoConnect("TestAP"))
  {
    wifiManager.startWebPortal();
  }
`
It works fine, I get an AP, when nothing is configured and afterwards I get a connection. And I also can access the config page afterwards.

But when I include SPIFFS.begin(); into my code (doing this at the very beginning of setup every time an AP is created and tit does not connect to previously configured STA.
Sometimes, when I connect, then to the AP shows me that I'm connected. And in that case I can access the server with the AP as well as with STA. But almost only an AP is established.

Removing SPIFFS.begin() from my code, all works correct after I have configured ma STA again.
Without SPIFFS.begin() tried to read/write a config file, all fails.

Any ideas, what I may doing wrong?

Schwenkner, 2021-05-06:
I have played a bit more and installed the "AutoConnectWithFSParameters.ino" sample.
It behaves the same. Everytime I start the program new, an AP is established and no STA.

Further in the example logs I see:
`mounting FS...`
but no
`mounted file system`
which should be the next.

And when trying to save the parameters, I get:
```
saving config
failed to open config file for writing
{
  "mqtt_server": "192.168.178.80",
  "mqtt_port": "1883",
  "api_token": ""
}

```

Schwenkner, 2021-05-06:
Can it be that the WiFiManager uses SPIFFS for the netwaork credentials whe I include SPIFFS and otherwise EEPROM?
I've wrote a small test program for SPIFFS and it does not mount the file system.
May be then, that is the reason for my issues.

What could be the reason for the SPIFFS error?

Schwenkner, 2021-05-06:
I've read something more about SPIFFS and Wemos D1 Mini.
I have the Pro version and it seems to have problems with SPIFFS.

After I switched to Mini R2 (not pro) I coud get SPIFFS working and also the WiFiManager.
Do not still understand all, but a bit more.

tablatronix, 2021-05-06:
You should be using littlefs, spiffs is deprecated

closed 2021-11-03 by tablatronix

ji3g4m6zo6, 2022-04-10:
This work for me.
```
// Format SPIFFS if failed
SPIFFS.begin(true);
```

xelfe, 2022-04-18:
> This work for me.
> 
> ```
> // Format SPIFFS if failed
> SPIFFS.begin(true);
> ```

It's work for me too but the static ip is not save. At reboot my node take the default IP (10, 0, 1, 99) instead of the one I entered into the portal.