tzapu/WiFiManager #1254: Question: Flag for successful connection

dronecz, 2021-05-26:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  feature_asyncwebserver

Hardware: Custom ESP32 board

Core Version:      	1.0.6

### Description

 I´m making setup wizard for my device and I have screen where I call function which start WiFiManager and  would like to ext this screen when there was successful connection to the network. 

Is there any function or callback which I can use for this already?

tablatronix, 2021-06-14:
Not yet, wm currently shutsdown the ap to attempt the connect (for stability), so maintaining a webservice is a problem.

dronecz, 2021-06-14:
Thanks for info.

I solved it like this: 

```
void loop() {
if (state == 103) { // state where I turn WM ON via button press
    wifi_config_t conf;
    esp_wifi_get_config(WIFI_IF_STA, &conf);
    apName = String(reinterpret_cast<const char*>(conf.sta.ssid)); // reading SSID from flash which should be empty
    if (apName != NULL) { // if SSID is not empty 
      Serial.println("Registered new SSID: " + apName);
      startTime = millis(); 
      if (WiFi.localIP()[0] > 0) { // if IP address is not 0
        setupWiFiScreenDone(); // call for scrren which shows full IP address
      } else {
        //if we did not get IP address, we have to clear SSID
        WiFi.disconnect(true);
        Serial.println("No IP address, restart WiFi setup.");
        setupWiFiScreenFail(); //information screen, which after timout calls another screen with state 103
      }
    }
  }
}
```

This works, but I have to enter SSID manualy as I have [this ](https://github.com/tzapu/WiFiManager/issues/1121#issuecomment-840089847) issue.