tzapu/WiFiManager #1255: Redirecting to startpage

Blacksmurf-GF, 2021-05-27:
Hello, first of all, thank you very much for the great work.

I have the following problem: I added two customfields. Everything works great. But after pressing the save button I only get to see "saved" (saving the settings works fine) But after saving I want to go back to the start page to enter the WIFI settings. Is that possible? If so, how can I do this? Many Thanks

sstopkin, 2022-02-06:
I have the same question. Also - is it possible to show all custom parameters on main portal page (not in setup)? Thank you.

tablatronix, 2022-02-07:
Thats the default, or don't have param in your menu. Not sure what the question is

sstopkin, 2022-02-07:
@tablatronix the original question from @SmurfettesBrother is - how to redirect from param page to main page after pushing "save" button? Currently user getting a popup with text "saved" and nothing else happened. Is is possible to navigate the user to main page (ideally after 3-5 seconds of delay)?

Thank you.

jayvdw13, 2022-10-25:
any progress on this?

tablatronix, 2022-10-26:
There is a test option
`    boolean       _showBack               = false; // show back button`