tzapu/WiFiManager #1260: Need to Disable Interrupt before Saving Credentials!

vginside, 2021-06-14:
Dear Sir,
  I have a problem! My node MCU keeps getting reset after i save my new wifi credentials from config page. I found the problem is happening because my Global  interrupts are ON (Interrupts are occurring every 10ms and i need it ON even when it is in AP mode). When in AP mode and after i enter new wifi credentials after clicking save, NodeMCU wont save credentials and reset the chip! Normally when updating EEPROM if Global Interrupts are ON same problem happens! If i turn off global Interrupts this problem dont happen. Can you please fix this problem? Solution is easy actually I mean just before saving credentials in EEPROM you need to disable Interrupts(using "noInterrupts();") and after saved you need to re-enable Global Interrupts(using "interrupts();")! If you can really fix this problem! 
  Also I'm have no idea how to edit library, but if you can just guide me how to edit the code to disable global Interrupt and Enable global Interrupts just before you write in EEPROM WIFI credentials i can also do it on my own! 

Deeply awaiting your reply!

 Best Regards!

tablatronix, 2021-06-14:
WM does not write to EEPROM, the ESP library does. Sounds like a ESP issue.

vginside, 2021-06-14:
Dear Sir,
  Thanks for you quick reply... But is there some way to disable interrupt before calling save then re-enable it after save successful? Deeply awaiting your reply.

 Best Regards!

tablatronix, 2021-06-14:
I think the ESP library should handle this already, are you using a recent version?
There are callbacks available

```
    std::function<void(WiFiManager*)> _apcallback;
    std::function<void()> _webservercallback;
    std::function<void()> _savewificallback;
    std::function<void()> _presavecallback;
    std::function<void()> _saveparamscallback;
    std::function<void()> _resetcallback;
```

vginside, 2021-06-14:
Dear Sir,
  Thanks for you reply... I'm using alpha 2.0.3 latest one.. I'm new to arduino so i dont know how to edit library! Can you guide me or write me code to disable and re-enable the same? Deeply awaiting your reply.

Best Regards!

tablatronix, 2021-06-14:
I mean ESP, is your ESP library up to date?

vginside, 2021-06-14:
Dear Sir,
  Yes it is latest! 3.0.0

Best Regards!

tablatronix, 2021-06-14:
```cPP
  wm.setPreSaveConfigCallback(presaveWifiCallback);
  wm.setSaveConfigCallback(saveWifiCallback);

void presaveWifiCallback(){
  Serial.println("[CALLBACK] saveCallback fired");
  // disable interrupts
}

void saveWifiCallback(){
  Serial.println("[CALLBACK] saveCallback fired");
 // restore interrupts
}
```

vginside, 2021-06-15:
Dear Sir,
  Sorry due to time zone difference i'm replying late... I tried with above code but still the same problem.
Function  "presaveWifiCallback" callback is called and that too it is called twice but "saveWifiCallback" is never called! 
Also i tried using just callback function, with never even enabling the interrupt to test if callback are called but still "saveWifiCallback" is not called. 
  I'm getting serial log when keeping interrupt enable as follows:
"
*WM: [1] AutoConnect 
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  ESP_WIFI
*WM: [2] AP has anonymous access! 
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             ESP_WIFI
*WM: [1] password:         
*WM: [1] ssid_len:         8
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2184 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] Scan is cached 21014 ms ago
*WM: [3] -> mobilesecuritycore-detection.norton.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> mobilesecuritycore-detection.norton.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> mobilesecuritycore-detection.norton.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> mobilesecuritycore-detection.norton.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> mobilesecuritycore-detection.norton.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] Scan is cached 21563 ms ago
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] Scan is cached 21590 ms ago
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] Scan is cached 21614 ms ago
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] Scan is cached 21640 ms ago
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] Scan is cached 21672 ms ago
*WM: [3] -> mobilesecuritycore-detection.norton.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] Scan is cached 21716 ms ago
*WM: [2] Portal Timeout In 119 seconds
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] WiFi Scan completed in 2183 ms
*WM: [2] <- HTTP Wifi 
*WM: [2] Scan is cached 3133 ms ago
*WM: [1] 11 networks found
*WM: [2] AP:  -34 Gohil's-Home
*WM: [2] AP:  -44 Gohil_Jio24G
*WM: [2] AP:  -59 DADI 2
*WM: [2] AP:  -78 shree radhe 2Ghz
*WM: [2] AP:  -79 Express Wi-Fi by Facebook
*WM: [2] AP:  -81 Jigar_4G
*WM: [2] AP:  -83 Vinits Den 4g
*WM: [2] AP:  -83 hiteshp_4g
*WM: [2] AP:  -86 Ritesh508
*WM: [2] AP:  -91 TP-LINK_8044_RPT
*WM: [2] AP:  -93 Shreeji 2.4G
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [3] Sent config page 
*WM: [2] Portal Timeout In 116 seconds
*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
[CALLBACK] presaveCallback fired
[CALLBACK] presaveCallback fired
*WM: [3] Sent wifi save page 
*WM: [2] processing save 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] CONNECTED:
*WM: [1] Connecting to NEW AP: Gohil's-Home
*WM: [3] Using Password: (My password)
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:137 __yield

>>>stack>>>

ctx: cont
sp: 3ffffad0 end: 3fffffc0 offset: 0000
"
  Deeply awaiting your reply.

vginside, 2021-06-15:
Dear Sir,
  Just to illustrate my problem, Attached is a cut down version of my code as given below.
All you need is an NODEMCU and a clock generator every 10millisec (which i have on external interrupt pin!)  You will notice ESP reset issue: (ofc i dont use wm.resetSettings, but problem is same as saving wifi credentials).
  
"#include <WiFiManager.h>        //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

#define cExternalInterruptIO  4    //GPIO4     NodeMCU-Pin13 Pin16  

void ICACHE_RAM_ATTR ISR_cExternalInterruptIOChanged();      
bool boolWifiConnectSuccess = false;          

void setup() 
{
  // put your setup code here, to run once:
  Serial.begin(115200);                               // Serial Com Port Speed
  Serial.println();                                   
  Serial.println("Program Start/Reset");
  pinMode(cExternalInterruptIO, INPUT_PULLUP); 
  pinMode(LED_BUILTIN, OUTPUT); 
  attachInterrupt(digitalPinToInterrupt(cExternalInterruptIO), ISR_cExternalInterruptIOChanged, CHANGE); // Ext Interrupt attached on toggle; occurs every 10mSec approx
  WiFiManager wifiManager;
  wifiManager.resetSettings();
  wifiManager.setConfigPortalTimeout(120);    //Use this function to timeout AP portal page; Parameter time in seconds
  boolWifiConnectSuccess = wifiManager.autoConnect("ESP_WIFI");
  if(!(boolWifiConnectSuccess))         // check if wifi is connected within configportaltimeout if not restart
  {
    Serial.println("Failed to connect");
    boolWifiConnectSuccess = true;              // If connection Timeout you still mark flag as true for main loop LED Blink
    //ESP.restart();
  } 
  else                              // Wifi is connected before config portal timeout
  {
    //if you get here you have connected to the WiFi
    Serial.println("Connected");
  }
}

void loop() {
  // put your main code here, to run repeatedly:
}

void presaveWifiCallback()
{
  Serial.println("[CALLBACK] presaveCallback fired");
  //noInterrupts();                   // disable interrupts
}

void saveWifiCallback()
{
  Serial.println("[CALLBACK] saveCallback fired");
  //interrupts();                     // restore interrupts
}

void ISR_cExternalInterruptIOChanged()
{
  Serial.println("External Interrupted!");
}

"

  Hope this helps you identify the problem if it with ESP or library! Deeply awaiting your reply.

Best Regards!

tablatronix, 2021-06-15:
Where are you setting your callbacks?   wm.setPreSaveConfigCallback(presaveWifiCallback);

Also wm probably needs to be global

vginside, 2021-06-15:
Dear Sir,
  I have updated the program as you suggested, but still it wont work update code is as given below:
 "
#include <WiFiManager.h>        //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

#define cExternalInterruptIO  4    //GPIO4     NodeMCU-Pin13 Pin16  

void ICACHE_RAM_ATTR ISR_cExternalInterruptIOChanged();      
bool boolWifiConnectSuccess = false;          
WiFiManager wifiManager;
void presaveWifiCallback();
void saveWifiCallback();

void setup() 
{
  // put your setup code here, to run once:
  Serial.begin(115200);                               // Serial Com Port Speed
  Serial.println();                                   // Prinmt a blank line
  Serial.println("Program Start/Reset");
  pinMode(cExternalInterruptIO, INPUT_PULLUP); 
  pinMode(LED_BUILTIN, OUTPUT); 
  attachInterrupt(digitalPinToInterrupt(cExternalInterruptIO), ISR_cExternalInterruptIOChanged, CHANGE); // Ext Interrupt attached on toggle; occurs every 10mSec approx
  wifiManager.setPreSaveConfigCallback(presaveWifiCallback);
  wifiManager.setSaveConfigCallback(saveWifiCallback);
  wifiManager.resetSettings();
  wifiManager.setConfigPortalTimeout(120);    //Use this function to timeout AP portal page; Parameter time in seconds
  boolWifiConnectSuccess = wifiManager.autoConnect("ESP_WIFI");
  if(!(boolWifiConnectSuccess))         // check if wifi is connected within configportaltimeout if not restart
  {
    Serial.println("Failed to connect");
    boolWifiConnectSuccess = true;              // If connection Timeout you still mark flag as true for main loop LED Blink
    //ESP.restart();
  } 
  else                              // Wifi is connected before config portal timeout
  {
    //if you get here you have connected to the WiFi
    Serial.println("Connected");
  }
}

void loop() {
  // put your main code here, to run repeatedly:
}

void presaveWifiCallback()
{
  Serial.println("[CALLBACK] presaveCallback fired");
  noInterrupts();                   // disable interrupts
}

void saveWifiCallback()
{
  Serial.println("[CALLBACK] saveCallback fired");
  interrupts();                     // restore interrupts
}

void ISR_cExternalInterruptIOChanged()
{
  //External Interrupt Code
  Serial.println("External Interrupted!");
}
"

Serial Output is as follows:
"Program Start/Reset
*WM: [1] resetSettings 
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x4020dac8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb80 end: 3fffffc0 offset: 0190
3ffffd10:  00000000 0000000a 3ffffd83 00000008  
"

  It will keep resetting the ESP infinite times! with same message over serial port over and over. It wont even start in AP mode unless i remove the external interrupt! As soon as external interrupt is unplugged the program would work normally. Deeply awaiting your reply!

Best Regards!

tablatronix, 2021-06-16:
esp8266?

tablatronix, 2021-06-16:
No clue, 

```
0x40209af4: HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
0x40209f61: Print::write(char const*) at ??:?
0x40209af4: HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
0x4020c596: raise_exception at core_esp8266_postmortem.cpp:?
0x401002aa: millis at ??:?
0x4020c5f3: __panic_func at ??:?
0x4020c079: yield at ??:?
0x4020ed17: ESP8266WiFiSTAClass::waitForConnectResult(unsigned long) at ??:?
0x40203e4b: WiFiManager::connectWifi(String, String, bool) at ??:?
0x40100061: _UserExceptionVector at ??:?
0x4020af18: String::copy(char const*, unsigned int) at ??:?
0x4020b104: String::operator=(String const&) at ??:?
0x4020b134: String::String(String const&) at ??:?
0x4020918a: WiFiManager::processConfigPortal() at ??:?
0x401001bc: ets_post at ??:?
0x402094e6: WiFiManager::startConfigPortal(char const*, char const*) at ??:?
0x40202e0a: void WiFiManager::DEBUG_WM<__FlashStringHelper const*>(__FlashStringHelper const*) at ??:?
0x40210e64: std::_Function_base::_Base_manager<void (*)()>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
0x40209654: WiFiManager::autoConnect(char const*, char const*) at ??:?
0x40209b00: HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
0x40100110: ISR_cExternalInterruptIOChanged() at ??:?
0x40210e50: std::_Function_handler<void (), void (*)()>::_M_invoke(std::_Any_data const&) at ??:?
0x40202e0a: void WiFiManager::DEBUG_WM<__FlashStringHelper const*>(__FlashStringHelper const*) at ??:?
0x40210e64: std::_Function_base::_Base_manager<void (*)()>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
0x40210e50: std::_Function_handler<void (), void (*)()>::_M_invoke(std::_Any_data const&) at ??:?
0x40210e64: std::_Function_base::_Base_manager<void (*)()>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
0x40201118: setup at ??:?
0x4020103c: saveWifiCallback() at ??:?
0x40210e64: std::_Function_base::_Base_manager<void (*)()>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
0x40210e50: std::_Function_handler<void (), void (*)()>::_M_invoke(std::_Any_data const&) at ??:?
0x4020c144: loop_wrapper() at core_esp8266_main.cpp:?
0x40100ea1: cont_wrapper at ??:?
```

Pretty sure you cannot disable interrupts when using wifi, but it shouldn't crash, remove the serial print from your int handler and use a flag to handle this instead

closed 2021-11-03 by tablatronix