tzapu/WiFiManager #1265: Not an issue - a question on connecting to a choice of multiple SSID's

dontpanic21, 2021-06-23:
Hi,

I'm trying to use wifimanager in a situation where the connecting device is roaming. There are multiple access points available which all have different SSID's however the password is the same for all of them.

What would be the best strategy for autoconnection to any/nearest available AP? Can I use autoconnect() to do this somehow, or do I need to save all the SSID's in SPIFFS  and use startConfigPortal() to pick one to connect to?

Thanks,
Will.

tablatronix, 2021-06-23:
You can check out esp multi ap mode, but it doesnt let you pick, it auto cycles through them. wm can only handle the one flash saved config.

Your best bet is to just code your own connection loop

dontpanic21, 2021-06-23:
Ok thanks for the advice,
Will.

closed 2021-06-23 by dontpanic21