tzapu/WiFiManager #1271: Crash when calling autoConnect() from class method

vcolella, 2021-07-19:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266: 

Hardware:			ESP-12e

Core Version:      	2.5.2

### Description

I'm trying to call ` autoConnect()` inside a class method. It compiles just fine but for some reason that I don't understand it causes a crash. From what I understand last call in the stack was at `ESP8266WiFiSTAClass::waitForConnectResult()`, so I'm not even sure if the issue has to do with WiFiManager itself.

Am I doing something wrong ?

I tried to find a solution based on #381 but couldn't really get anywhere.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)

Additional libraries: None

### Sketch

```cpp
#include <WiFiManager.h>

class Foo
{
public:
  Foo();
private:
  void fooMember();
};

Foo::Foo()
{
  fooMember();
}

void Foo::fooMember()
{
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFiManager wm;
  wm.setDebugOutput(true);
  wm.autoConnect("myNetwork");
}

Foo myObject;

void setup() {
  //
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Messages

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
*WM: [1] getCoreVersion():          2_5_2
*WM: [1] system_get_sdk_version():  2.2.1(cfd48f3)
*WM: [1] system_get_boot_version(): 5
*WM: [1] getFreeHeap():             50624
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: TEST 2.4GHz
*WM: [3] Using Password: 123456
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 

Soft WDT reset
```

### Decoded Stack
```
0x402080f0: ESP8266WiFiSTAClass::waitForConnectResult() at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 440
0x402080fe: ESP8266WiFiSTAClass::waitForConnectResult() at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 442
0x40204274: WiFiManager::waitForConnectResult(unsigned short) at C:\Users\Victor Colella\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 912
0x40207e80: ESP8266WiFiSTAClass::status() at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x4020438d: WiFiManager::waitForConnectResult() at C:\Users\Victor Colella\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 902
0x40204152: WiFiManager::wifiConnectDefault() at C:\Users\Victor Colella\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 842
0x40201434: String::operator!=(char const*) const at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/WString.h line 191
0x40205aa2: WiFiManager::connectWifi(String, String) at C:\Users\Victor Colella\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 784
0x4020d0b8: String::String(char const*) at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 36
0x40207e80: ESP8266WiFiSTAClass::status() at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x4020cf20: String::~String() at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 125
0x40206ec4: WiFiManager::autoConnect(char const*, char const*) at C:\Users\Victor Colella\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 297
0x40205e16: WiFiManager::debugPlatformInfo() at C:\Users\Victor Colella\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 2646
0x4020c7b0: HardwareSerial::begin(unsigned long, SerialConfig, SerialMode, unsigned char) at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\HardwareSerial.cpp line 51
0x4020105c: Foo::fooMember() at C:\Users\VICTOR~1\AppData\Local\Temp\arduino_modified_sketch_878937/sketch_jul19a.ino line 22
0x40220100: etharp_query at core/ipv4/etharp.c line 934
0x40220100: etharp_query at core/ipv4/etharp.c line 934
0x40220301: etharp_output_LWIP2 at core/ipv4/etharp.c line 861
0x4020d01a: String::reserve(unsigned int) at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 148
0x4020d078: String::copy(char const*, unsigned int) at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 211
0x4020d0b8: String::String(char const*) at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 36
0x4020d0b8: String::String(char const*) at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 36
0x40207a24: ESP8266WiFiGenericClass::_eventCallback(void*) at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 218
0x4020ce41: UpdaterClass::UpdaterClass() at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\Updater.cpp line 45
0x4020107c: _GLOBAL__sub_I__ZN3FooC2Ev() at C:\Users\VICTOR~1\AppData\Local\Temp\arduino_modified_sketch_878937/sketch_jul19a.ino line 33
0x4020e151: init_done() at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 149
0x4020d180: String::move(String&) at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 238
0x4020d180: String::move(String&) at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 238
0x4020d180: String::move(String&) at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 238
0x4020a18c: ESP8266WebServer::responseCodeToString(int) at C:\Users\Victor Colella\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp line 663

```

vcolella, 2021-07-26:
I was able to reproduce the problem including only `ESP8266WiFi`, so I guess I'm closing the issue here.

closed 2021-07-26 by vcolella

tablatronix, 2021-07-29:
Almost looks like a wdt or a stuck loop.. strange