tzapu/WiFiManager #1273: Converting Optional String parameter  into an IPAddress class variable

bgrigoriu, 2021-07-23:
Hi Everyone,

Had two simple newbie questions.
1. Does anyone have an example on how to read IP address as custom parameter ( and result in a valid IPAddress class variable). 
I saw that the optionalIPFromString() exists but it is not clear how it works and what checks are done. 

2. Can you do a minimal control of the value of a Custom parameter introduced during startup ? (For example, testing that a serial number fits some minimal criteria and not having to connect to WIFI network then test outside the parameter and restart the process if not appropriate?

bgrigoriu, 2021-07-24:
Just found the ParamsChildClass example. 
Answers part of the question.

tablatronix, 2021-07-24:
I think This was just asked in another issue ipaddress casting

bgrigoriu, 2021-07-24:
Hi, Can you give me the issue# ?
The whole work you all did is just amazing !
Thanks for helping .
Just started learning a few months ago and all the info provided is well received.

bgrigoriu, 2021-07-24:
Found some info in the OnDemandConfigPortal. The html checks for format but not for range for the ip. 
Need to learn some Html then.

tablatronix, 2021-07-24:
Range? Like you want to verify ip?

bgrigoriu, 2021-07-24:
yes. 
I am trying to build a device for "lay people" and they will need to input an Ip adress and a serial number of a commercial device. So it has to be "....proof". The html code test for format but not for range and you could eventually  input 999.999.999.999 
No one will ever think to input this but who knows?

bgrigoriu, 2021-07-25:
Just found a solution: 
the pattern  for defining Ip addresses should not be "pattern='\\d{1,3}\\.\\d{1,3}\\.\\d{1,3}\\.\\d{1,3}'")  which allows numbers over 255 but eventually 
((^25[0-5])|(^2[0-4][0-9])|(^1[0-9][0-9])|(^[1-9][0-9])|(^[1-9]))\\.(25[0-5]|2[0-4][0-9]|[01]?[0-9][0-9]?)\\.(25[0-5]|2[0-4][0-9]|[01]?[0-9][0-9]?)\\.(25[0-5]|2[0-4][0-9]|[01]?[0-9][0-9]?)$
OR
((^25[0-5])|(^2[0-4][0-9])|(^1\\d{1,2})|(^[1-9]))\\.(25[0-5]|2[0-4][0-9]|[01]?[0-9][0-9]?)\\.(25[0-5]|2[0-4][0-9]|[01]?[0-9][0-9]?)\\.(25[0-5]|2[0-4][0-9]|[01]?[0-9][0-9]?)$
or alike
Hope helps someone else

tablatronix, 2021-07-25:
you can use custom regex masks in some browsers for html, or add a js check.

bgrigoriu, 2021-07-25:
Ok
Could you please suggest some examples/tutorial/training material/courses ?