tzapu/WiFiManager #1274: If parameter get empty (e.g. like "" insteed of "0") when saving it breaks future-wise!

kbaggen, 2021-07-29:
Hi,
on ESP8266 and 2.0.3 ALPHA, is seems that if a parameter for instance Temp offset set at "0" breaks if one by accident save it as "", and then it is not possible to later save it again with either "0" or "1" nor "1000".

Hence, it seems either Wifimanger or Filesystem make a fuss if the paramater get completly empty and saved. Not sure how else to describe it and what eventually to do.

Only ways to correct is to erase everything and start over!

Anyone seen this? And I am pretty sure I had same issue in former builds of wifiM.

So question is if a fix in wifimanger to always set a empty field tilll "0" could be added?

(BTW....thanks for the amazing WifiManger)

kbaggen, 2021-07-29:
I am not sure  this is connected till empty "", but something else, so I close it agian! Sorry!

closed 2021-07-29 by kbaggen

tablatronix, 2021-07-29:
Sounds like a char null termination issue