tzapu/WiFiManager #1275: resetSettings();  not working

nhlives, 2021-07-31:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP-12E

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu 1,0 (esp-12E module)
Additional libraries: 

### Sketch

```cpp
#BEGIN
       WiFiManager wifiManager;
        wifiManager.resetSettings();
        delay(4000);
        ESP.reset();
#END
```

### Debug Messages

The resetSettings(); doesn't bring up the AP.  I need to cause the AP to start after a successful connect. This did work on platform 2.7.4,  I'm on 3.0.2
```

tablatronix, 2021-08-01:
What does serial say?  Esp32 or 8266?

tarontop, 2021-08-06:
```
    wifiManager.resetSettings();
    ESP.eraseConfig(); 
    delay(2000);
    ESP.reset(); 
```

This works for me

hungtcs, 2021-08-15:
> ```
>     wifiManager.resetSettings();
>     ESP.eraseConfig(); 
>     delay(2000);
>     ESP.reset(); 
> ```
> 
> This works for me

`ESP.eraseConfig(); ` working for me.

radacina, 2021-12-23:
Per ESP SDK doc, the advantage over eraseConfig would be IMHO that the "damage" would be limited to the struct size and address:

```
void user_set_station_config(void) 
{ 
char ssid[32] = "";  
char password[64] = "";  
struct station_config stationConf;  
stationConf.bssid_set = 0;      //need not check MAC address of AP 
os_memcpy(&stationConf.ssid, ssid, 32);  
os_memcpy(&stationConf.password, password, 64);  
wifi_station_set_config(&stationConf);  
}
```

tablatronix, 2021-12-24:
ESP8266 or ESP32?????

radacina, 2021-12-24:
> ESP8266, I updated the comment

tablatronix, 2021-12-24:
I am assuming ( lack of logs ) that erase is failing.

I will do some testing with newer libs

wm.resetsettings does a simple disconnect() which hits the sdk with an empty config struct, so could be bugs there..
wm.erase does a ESP.eraseconfig() with bool return 

Ill see if i can add a more reliable method, and add status checking

tablatronix, 2021-12-24:
Also some of these will fail if sta mode is faulty or not active, which could also be a bug in sdk sometimes setmode fails

AndyT86, 2022-03-26:
After reading I think I know, I am expecting resetSettings() be permanent but it is not, thus buggy. Commenting out that code is having entire suite reading back in EEPROM'd SSID/password instead of actually erasing it; while removing comment it is like a Safe Mode to block the EEPROM, the actual erasing is somewhere else; given the intent and name this is buggy to use even though it may be actual intent of the suite in the view of WifiManager team.

tablatronix, 2022-03-26:
Huh?

AndyT86, 2022-03-26:
It's not permanent erasure of configuration data, that's what I mean, all it does to my ESP32 is like someone pressed F8 in Windows bootup. That is not expected, commenting out and it reads back in the data written to somewhere to connect to what was set instead of starting anew.

AndyT86, 2022-03-26:
I figured it "bypass" the Wi-Fi creds no erasure really happened, one could tell since it pre-dots the passcode field and rebooting the ESP reads in the cred prior to these, the chip is ESP32-WROOM-32 on a Wemos D1 R32 board; the one shaped like Uno, since that is one that happens to be compatible here, I have not tried the no frill DevKit board I also have. The core lib is Espressif https://github.com/espressif/arduino-esp32

tablatronix, 2022-03-27:
Not sure what compatible means, all esp are compatible regardless of what dev board/project.

I don't know what pre dots means, the info page clearly shows wifi information as the serial output.. It should be fairly clear even by the status at the bottom of the home page...

tablatronix, 2022-03-27:
works for me just fine...

```php
*wm:[1] Free heap:        277632
*wm:[1] ESP SDK version:  v3.3.5-1-g85c43024c
[WIFI] WIFI INFO DEBUG
Mode: STA
Channel: 1
SSID (5): leela
Passphrase (8): Turdtr0n
BSSID set: 0
[WIFI] SAVED: YES
[WIFI] SSID: leela
[WIFI] PASS: Turdtr0n
*wm:[1] resetSettings 
*wm:[1] SETTINGS ERASED 
[WIFI] WIFI INFO DEBUG
Mode: STA
Channel: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
[WIFI] SAVED: NO
[WIFI] SSID: 
[WIFI] PASS: 
*wm:[2] Added Parameter:
*wm:[2] Added Parameter: server
```

tablatronix, 2022-03-27:
Cannot reproduce, unless you can show me what its doing. So you are using git staging esp32? IDF? what?

AndyT86, 2022-03-27:
I meant USB chip compatibility, some of them are compatible under Windows while the rest need drivers to do the ISP or FTDI; that's neither here nor there. Forget about "pre-dot" I just meant I can observe visually old data is retained. But as you said, serial log is saying exactly that too. A moment while I gather stuff.

tablatronix, 2022-03-27:
and tested staging OK

```oho
INFORMATION] TEST
*wm:[1] Free heap:        259012
*wm:[1] ESP SDK version:  v4.4-beta1-308-gf3e0c8bc41
[WIFI] WIFI INFO DEBUG
[WIFI] SAVED: YES
[WIFI] SSID: leela
[WIFI] PASS: Turdtr0n
*wm:[1] resetSettings 
[  1211][V][WiFiGeneric.cpp:275] _arduino_event_cb(): STA Stopped
[  1212][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 3 - STA_STOP
*wm:[1] SETTINGS ERASED 
[WIFI] WIFI INFO DEBUG
[WIFI] SAVED: NO
[WIFI] SSID: �&�?xV��
[WIFI] PASS: 
*wm:[2] Added Parameter:
```

tablatronix, 2022-03-27:
Ok I see the problem here.
esp32 does nothing if wifi is not enabled, let me see how else I can do this, direct erase etc.
Also getting stuff like ssid fails in the same way even if its set

AndyT86, 2022-03-27:
I think I figure it out, it works before calling autoConnect(), otherwise the serial log says it is working but it would not be after, might as well since no one need to care whether reset is pre or post turning on.

tablatronix, 2022-03-27:
resetsettings starts sta actually so nm what i said, but there might be a race condition, i will try adding a delay in there to make sure and also add a check to see if it actually suceeds

tablatronix, 2022-03-27:
hmm i can test that too, I use ondemand button and it usually also works so might be same issue

AndyT86, 2022-03-27:
Oh while I got your attention:

collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\actsa\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
Error compiling for board WEMOS D1 MINI ESP32.

Some #define and #ifndef stuff I suspect.

tablatronix, 2022-03-27:
No idea, sounds like an arduino issue,

AndyT86, 2022-03-27:
I think I have huge IDE issue, don't know where it is wrong it was fine last night.

diegotam855, 2022-05-08:
Hi @tablatronix, I have a similar issue, so as you said the code is working fine with this line in the setup void:

WiFi.mode(WIFI_STA);

But it doesn't work when I move it into a IF statment:

if (digitalRead(TRIGGER_PIN) == LOW){      
        wifiManager.resetSettings();
        delay(500);
        ESP.restart();
        delay(1000);
    }

I'm trying to get the memory erase when I press a button on sturtup.
Any suggestion?

AndyT86, 2022-05-08:
If I may interject, I believe I can be of some help instead of @tablatronix on this, Shawn once said there maybe a bug and I am getting around this at startup differently:
[image]
This I am using ESP32, I have both DevKit and Wemos Mini version but shouldn't matter.

diegotam855, 2022-05-08:
> If I may interject, I believe I can be of some help instead of @tablatronix on this, Shawn once said there maybe a bug and I am getting around this at startup differently: [image] This I am using ESP32, I have both DevKit and Wemos Mini version but shouldn't matter.

Thank you for answering!
But it seams that wifimanager doesn't have `.erase()` function, what's that?

diegotam855, 2022-05-08:
Here is what I'm doing now

```
bool forceConfig = false;

    if (digitalRead(TRIGGER_PIN) == LOW){    
      forceConfig = true;
    }    

    if (forceConfig){
      wm.setConfigPortalTimeout(timeout);
      if (!wm.startConfigPortal(ESP_AP)){
        Serial.println("failed to connect and hit timeout");
        delay(1000);
        ESP.restart();
        delay(1000);
      }
    }
    else{
      if (!wm.autoConnect("ESP_AP")){
          Serial.println("Failed to connect");
          delay(1000);
          ESP.restart();
          delay(1000);
      }
      else {
          Serial.println("Connected!");
      }
```

AndyT86, 2022-05-08:
No it's not like that, check mine; it just flows into no branching is needed besides that if button ... little detour.

tablatronix, 2022-05-09:
No idea it always erases for me.. what esp version?

tom-ch1, 2022-11-14:
do we actually know what the various wm functions are supposed to do? I did not find a documentation of resetSettings() nor erase() :-(

tablatronix, 2022-11-14:
lol, seriously

https://github.com/tzapu/WiFiManager/wiki/API-reference

Brian-Kristiansen, 2023-03-31:
Hi -

Sorry to revive this old issue... 

I ran into the same problem as he original poster, today - trying to use resetSettings() to clear WiFi  credentials on a ESP8266 (Wemos D1 mini). I use the library tzapu/WiFiManager@^0.16.0. 

When using resetSettings() right before autoConnect() it correctly starts the 192.168.4.1 website etc.. but when using resetSettings() to only clear the credentials, and then immediately restart the board without calling resetSettings(), the previous WiFi credentials are loaded and used by autoConnect(). It looks like the resetSettings() didn't save/clear. Even calling WiFi.disconnect(true) didn't clear the credentials. 

I suspected, maybe it tried to clear the EEPROM information, but was now allowed/ignored. So I tried:
```
  WiFi.mode(WIFI_STA);
  WiFiManager wm;

  // Clearing WiFi credentials
  **WiFi.persistent(true);**
  wm.resetSettings();
```

This was a succes - setting WiFi.peristent(true) works for me!! 
Looks like the WiFi library defaults to WiFi.persistent(false) - so the resetSettings() never persists the clearing. -  OR my cheap ESP8266 board does not follow standards .. or ..

Anyway - I hope my tip can help somebody else in the same situation.

@tablatronix - Would it be risky to add a WiFi.persistent(true) in the resetSettings() code?

Brian-Kristiansen, 2023-04-01:
Thanks @tablatronix, I see my problem. 
I use v0.16.0 - as this is what lib_deps in PlatformIO fetches for me automatically. I cannot see any newer versions there.

In my version of v0.16.0 it looks like this:
```
void WiFiManager::resetSettings() {
  DEBUG_WM(F("settings invalidated"));
  DEBUG_WM(F("THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA."));
  WiFi.disconnect(true);
  //delay(200);
} 

```
.. which corresponds to the log messages I get, when running the application.

Now - how to get PlatformIO to use v2.x? Hmmm.. When will v2.x upgrade from being a RC? ;-) 

Thanks.

tablatronix, 2023-04-01:
You have to use the git address in lib deps

tablatronix, 2024-08-17:
Does it not bring up the captive portal, what if you goto the ip manually in browser?

Suresh6060, 2024-08-17:
> Does it not bring up the captive portal, what if you goto the ip manually in browser?

It doesnt connect. I can see the AutoCoonectAp under the list. However if I click on it, It says ApConnectAP has no internet Tap for options . Looks like it behaves like any other WiFi on the network. The page for entering SSID and PASSWORD doesn't show.

tablatronix, 2024-08-17:
Thats the captive portal, goto 192.168.4.1 and see if it works,.

Suresh6060, 2024-08-17:
> Thats the captive portal, goto 192.168.4.1 and see if it works,.

I think I got why it behaves like that. I have several other libraries working on connecting to WiFi in the first instance like WebServer.h, HTTPClient.h etc. 

They are still connected when wm.resetSettings() is executed and wm.startConfigPortal or autoConnect is tried. This messes with the functionality because the other objects are still latched to the wifi. I don't know how this works out internally. 

To confirm this , I used a new sketch with bare minimum code elements to test only WiFiManager. It works nicely every time. 

Probably I need to find out how the other objects could be stopped  before re-configuring WiFiManager and restarted on successful connection with new WiFi credentials.    

Thank you for your time and efforts.

tablatronix, 2024-08-18:
There is a issue in esp with releasing servers and reusing ports no solution afaik. So you cannot stop and restart another web object as the first iss never actually  cleared

Suresh6060, 2024-08-19:
> There is a issue in esp with releasing servers and reusing ports no solution afaik. So you cannot stop and restart another web object as the first iss never actually cleared

I used server.stop() 
server is WebServer server(80); declared earlier. 
There was another issue. 

wm.resetSettings() only cleared the WiFi details stored by  WiFiManager wm; 

What was stored by WiFi.h still existed. 
WiFi.disconnect() only disconnects the network . The information(s) stored earlier still exist. This creates conflicts when wm is re-configured. 

In order to synch both WiFi.disconnect(false, true) was used in the begin. This clears the WiFi cache too and wm sends th config with No AP set. 

When the new AP is received, both wm and WiFi gets synchronised again and it works. 

server.begin() was used to start the server after it was stopped. 

Now the whole code works nicely as expected. The only glitch I find is in order to load the in built pages in the server, the esp2 had to reset again soon after config settings. That is fine as of now.

tablatronix, 2024-08-24:
The credentials are the same as esp, we dont use flash