tzapu/WiFiManager #1276: How to Determine AP Credential Failure using autoConnect()?

tonyc770, 2021-07-31:
I am using the autoConnect() function, but if the wrong WiFi credentials are entered there is no feedback while in AP mode.
Only when the credentials are correct does the autoConnect() return true, or if there is a timeout, returns false.

Is there a way to detect that the credentials failed?

Thanks...

monodop, 2021-11-02:
I am also interested to know if there's a good way to determine this.

tablatronix, 2021-11-03:
you might be able to use 
`    // get last connection result, includes autoconnect and wifisave
    uint8_t       getLastConxResult();`

monodop, 2021-11-04:
is there a callback we can use to check once an attempt to save the configuration has been made?

For success, we have `setSaveConfigCallback`, but I can't see a corresponding callback for failure.

tablatronix, 2021-11-06:
Hmm could be handy let me see what works best

monodop, 2022-01-05:
Any thoughts?