tzapu/WiFiManager #1279: How add a custom webpage to the portal?

AKTanara, 2021-08-05:
First of all thanks for the great job...

Is it possible to add a page to your library for a specific url?
I'm running [SSDP Library](https://github.com/luc-github/ESP32SSDP) which needs to serve a xml file like this to share UPNP info of the device in the network:
http://192.168.1.40/description.xml
I have managed to do this with a workaround:
```
SSDP.setHTTPPort(85);
ESP8266WebServer HTTP(85);
```

But it would be more pro to have this file on the same webserver that serves WiFiManager Portal in port 80 that is set to be active all the time even when successfully connected to a SAP as STA (Like described in #656) and get rid of `ESP8266WebServer HTTP(85);` server that is only meant to provide this XML file...

And I apologize in advance if this question is already answered and I couldn't find it there.
Regards

JJSlabbert, 2021-09-04:
This will be nice. My thinking was also to run an additional webserver on another port.

tablatronix, 2021-09-05:
Yes see the super example