tzapu/WiFiManager #1287: Display Dynamic Sensor Readings on WiFi Configuration page

JJSlabbert, 2021-09-04:
I need to display dynamic sensor readings on the Configure WiFi (or some where else) page. The sensor readings need to be updated +/- every 3 seconds. I was able to get correct sensor readings using the following HTML code in custom parameters, but the sensor readings is static. It seems like they are taken the moment the configuration portal starts. Refreshing the page with the browser or the blue refresh button does not refresh the sensor reading. 

My Setup: Reading Battery voltage on A0
When Configuration portal starts, the battery is connected. My code reeds the voltage correct. If I remove the battery and hit refresh, the reading does not change. Any advice?

```
    //Reading Battery Voltage on A0, create HTML string, convert the HTML into constant Character Aray
    String string_html_batvolt = "<p><i>Battery Voltage: </i><b>"+String(analogRead(A0)/atof(batconvfact))+"</b></p>";
    const char *html_batvolt = string_html_batvolt.c_str();
    
    //Adding the custom parameters
    WiFiManagerParameter custom_batvolt(html_batvolt);
    wm.addParameter(&custom_batvolt); 
```

tablatronix, 2021-09-05:
use non blocking mode and update the param in loop

JJSlabbert, 2021-09-06:
Tnx. I got it working with a custom page, I will use your suggestion also. Tnx for the great project.