tzapu/WiFiManager #1288: ESP32 compile fail with newest 2.0 ESP tool chain.  (Fixed using branch esp32s2)

chlor3, 2021-09-05:
As I was under the impression esp32s2 was behind master I ended up fixing this myself only to discover the fix already in commit a9507c9.  I recommend merging commit a9507c9 into Master as it fixes compile fail with newest 2.0.0 tool-chain build for ESP32.  

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: Esp32

Hardware:			DOIT ESP32 DEVKIT V1

Core Version:      	Release v2.0.3

touchgadget, 2021-09-06:
I think one more change is needed. I am using master + cherry-pick a9507c9 + the following change. I can compile the Basic example for ESP32, ESP32-C3, and ESP32-S2. Espressif Arduino ESP32 board package 2.0.0.

```
diff --git a/WiFiManager.cpp b/WiFiManager.cpp
index 1dd199e..305ed11 100644
--- a/WiFiManager.cpp
+++ b/WiFiManager.cpp
@@ -3529,7 +3529,7 @@ void WiFiManager::WiFiEvent(WiFiEvent_t event,arduino_event_info_t info){
         WiFi.reconnect();
       #endif
   }
-  else if(event == SYSTEM_EVENT_SCAN_DONE){
+  else if(event == ARDUINO_EVENT_SC_SCAN_DONE){
     uint16_t scans = WiFi.scanComplete();
     WiFi_scanComplete(scans);
   }

```

tablatronix, 2021-09-07:
Yeah they changed the events its a pita i have been trying to figure out how to make it backwards compatible

tablatronix, 2021-09-08:
I am working on this, I merged master into branch for the moment, I need to get my pio working again for new idf afaik there is no arduino 2.0 support yet and my soala wont build for some reason

tablatronix, 2021-09-08:
What is your build platform ? Arduino ide or PIO ?

tablatronix, 2021-09-09:
Can you try the s2 branch for both ESPlibs and S2/C3 etc

This should work seamlessly for 1.x and 2.x and IDF

touchgadget, 2021-09-09:
@tablatronix 
Summary: esp32s2 branch works fine using arduino-esp32 board packages 1.0.6 and 2.0.0.

Tested using Arduino IDE 1.8.16 and arduino-esp32 board package 2.0.0. By the way, the 2.0.0 package is now on the default/stable release. I am using this one although I assume it is the same as the 2.0.0 package on the dev release.

Test steps using the Basic example
* First time (no saved credentials) AP scan/select works.
* The saved credentials are used after power cycling the board.

1. ESP32 generic dev, ESP32 Dev Module, All OK.
2. ESP32-C3 DevKitM-1 v1.0, ESP32C3 Dev Module, All OK.
3. ESP32-S2-Saola-1_v1.2, ESP32S2 Dev Module, all OK.

Tests repeated using board package 1.0.6.

1. ESP32 generic dev, ESP32 Dev Module, All OK.
2. ESP32-C3 and ESP32-S2 are not supported by this version.

tablatronix, 2021-09-10:
It is not available for pio yet

closed 2021-11-03 by tablatronix