tzapu/WiFiManager #1291: Using __FlashStringHelper for customHTML

Mollayo, 2021-09-08:
__FlashStringHelper can be used for customHTML; this is useful for reducing the memory usage of WifiManager. However, this would require to make the following modifications to the WiFiManagerParameter class:

- Make the _customHTML parameter as protected instead of private.
- Make the getCustomHTML() method virtual so that this method can be overridden by the child class.

With these modifications, it would then be possible to implement a WiFiManagerParameter child class to implement a CustomHTML that uses __FlashStringHelper.  See below. 

If you agree, I can implement this feature myself and make a pull request.

```javascript
String flashStrBuffer;

class FlashCustomHTMLParameter : public WiFiManagerParameter {
public:
    FlashCustomHTMLParameter(const __FlashStringHelper *str):WiFiManagerParameter(NULL)
    {
      _customHTML=(const char*)str;
    }
    const char* getCustomHTML() const {
      // Convert __FlashStringHelper* to const char* using the String class
      flashStrBuffer=(const __FlashStringHelper*)_customHTML;
      return flashStrBuffer.c_str();
    }
};
```

tablatronix, 2021-09-08:
hmm why did I think this was already supported?

tablatronix, 2021-09-08:
Can you have this PR ready soon? I need to make a release should I wait ?

Mollayo, 2021-09-08:
Ok. I will do it within today.

Mollayo, 2021-09-08:
I just made the pull request.

closed 2021-09-08 by tablatronix