tzapu/WiFiManager #1293: Flash size: FS 128k / eagle.flash.1m128.ld is a minimum requirement

ruedli, 2021-09-16:
### Basic Infos
FS memory requirement not documented, must be at least 128kb.
#### Hardware
WiFimanager Branch/Release:  Master

Esp8266: SonOff S26

Hardware: ESP8286 1Mb

Core Version:      	2.4.0, staging

### Description
After making the sketch work for small examples I struggled making it work for my more demanding sketch, which required an mqtt interface, an API (to a 3D printer) while keeping OTA all within 1Mb.

The "problem" I ran into was that the webpages for the config portal would just not work: they failed to save parameters or wifi credentials and connecting to the cp was a nightmare.

It was all resolved by ensuring enough memory for the FS. 64kB was NOT enough, the portal worked smoothly again after I gave it 128kb. Done from the arduino IDE with the "1Mb / 128k FS OTA:~438KB" option under "tools"->"flash size" or in platformio by adding the line: 

board_build.ldscript = eagle.flash.1m128.ld

and the script itself, which you can download if you search for options to size the flash memory in platformio, see: https://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size

This could be due on the fact that I needed to add 7 parameters beyond the wifi credentials. You could use more or less depending on your environment.

I am documenting this, because I could not find any guidance on selecting the FS and figuring it out cost me some grey hears.
Without this (for me) it would not work stable, which should not be confused with the stability of WiFiManager.

### Settings in IDE

Module:  Sonoff S26

Additional libraries:
pubsubclient
octoprintAPI
ArduinoOTA 

### Sketch
https://github.com/ruedli/OctoPlugout

ruedli, 2021-09-17:
The stability around this topic keeps confusing me. Without changing the use of the portal, it would crash upon saving wifi configuration / or params - one or the other. increasing the  FS size then would let the params be saved, but wifi could not be saved -or- vice versa. Very difficult to understand the relationship and requirements of this: is it really the FS size related or inherent instability caused by hardware? I inspected the crash stack dumps (using the "elf" files) and they seemed heavily involved with processing the "saving" somewhere in the library. It was not reflecting anything of my own code. 7 (long) text params (just under 200 bytes together) + wifi credentials can't be too overwhelming?

ruedli, 2021-09-17:
like now I get: 
[image]

and the browser to the CP says connection aborted

tablatronix, 2021-09-17:
Are you using a FS ? Try turning it off and see if it still crashes

ruedli, 2021-09-17:
@tablatronix A quick test with eagle.flash.1m.ld (1Mb no FS)  revealed "not crashing". Relative sketch size is 5% lower: around 40%.

RAM:   [====      ]  40.8% (used 33432 bytes from 81920 bytes)
Flash: [====      ]  39.1% (used 400880 bytes from 1023984 bytes)

I could save both params and credentials. A few days ago this would not work. What is the logic behind this?

tablatronix, 2021-09-17:
you might have to add heap debugging, all html output is on the heap atm.
Without the code or a small repro sketch, no idea..

whats the decoded exception say exactly ?

ruedli, 2021-09-17:
One parameter is a c-string, I convert it with .c_str() fro the param value and declare it with the proper length of 33 to include the 0 byte. I fill that value fully. Is is recommended to declare a longer length for the WiFiManager parameter? It would not explain why the crash above occurred on wifi save.

ruedli, 2021-09-17:
@tablatronix recompiling now with FS 64k to trigger a crash (when you want one....) I am not against heap debugging, only need to know how to go into that. I am compiling now wit "stack protection on". It was off.

ruedli, 2021-09-17:
@tablatronix Thnx for your help!!!

OK, I uploaded from IDE OTA, while keeping a serial connected, so I see the upload and start of the program. DIFFERENT BEHAVIOUR.

I see the upload: normal start of my program, it is working, then I press the button long (debug output 43-45) to trigger a resetsettings in my sketch and do an autoconnect. WM confirms the resetsettings, 

Then I see a  ets Jan  8 2013,rst cause:1, boot mode:(3,7)

NEXT it restarts, but still KNOWS the wifi credentials.... I have not entered the portal... How about settings erased??????

` ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v00062e30
@cp:B0
ld
:
 Starting
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter:
*WM: [1] AutoConnect
*WM: [2] Connecting as wifi client...
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping
*WM: [1] Connecting to SAVED AP: RRSW
*WM: [3] Using Password: ********
*WM: [3] WiFi station enable
*WM: [3] enableSTA PERSISTENT ON
*WM: [1] connectTimeout not set, ESP waitForConnectResult...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 17
cnt

connected with RRSW, channel 6
dhcp client start...
ip:192.168.3.49,mask:255.255.255.0,gw:192.168.3.254
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS
*WM: [1] STA IP Address: 192.168.3.49
connected...yeey :)
pm open,type:2 0
Button Pressed
Button Held
Erasing Config, restarting
*WM: [1] resetSettings
*WM: [3] WiFi station enable
*WM: [3] enableSTA PERSISTENT ON
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
mode : null
*WM: [1] SETTINGS ERASED 

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v00057090
~ld
:
 Starting
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter:
*WM: [1] AutoConnect
*WM: [2] Connecting as wifi client...
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping
*WM: [1] Connecting to SAVED AP: RRSW
*WM: [3] Using Password: *******
*WM: [3] WiFi station enable
*WM: [3] enableSTA PERSISTENT ON
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 17
cnt

connected with RRSW, channel 6
dhcp client start...
ip:192.168.3.49,mask:255.255.255.0,gw:192.168.3.254
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS
*WM: [1] STA IP Address: 192.168.3.49
connected...yeey :)
pm open,type:2 0
Button Pressed
Button Held
Erasing Config, restarting
*WM: [1] resetSettings
*WM: [3] WiFi station enable
*WM: [3] enableSTA PERSISTENT ON
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
mode : null
*WM: [1] SETTINGS ERASED 

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v00057090
~ld
:
 Starting
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter:
*WM: [1] AutoConnect
*WM: [2] Connecting as wifi client...
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping
*WM: [1] Connecting to SAVED AP: RRSW
*WM: [3] Using Password: ********
*WM: [3] WiFi station enable
*WM: [3] enableSTA PERSISTENT ON
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 17
cnt

connected with RRSW, channel 6
dhcp client start...
ip:192.168.3.49,mask:255.255.255.0,gw:192.168.3.254
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS
*WM: [1] STA IP Address: 192.168.3.49
connected...yeey :)
pm open,type:2 0`

ruedli, 2021-09-17:
The logged text "Erasing Config, restarting" is not in my code, but also not prefixed with *WM, so I do not know where i is coming from. It is confusing, as I am not restarting in my sketch (there is a different debug string when that would occur), but a restart is what seems to happen.

This is my code after the autoconnect. The restart is "from autoconnect itself?" 

`	wm.autoConnect("SetupOctoPlugout"); // no password protected ap
    if (WifiNotAvailable()) {
        Serial.println(F("Failed to connect, restarting"));
        ESP.restart();
		while(true); //Do not continue....
    } 
    else {
        //if you get here you have connected to the WiFi    
        #ifdef debug_
		Serial.println(F("connected...yeey"));
		#endif
    }
	`

tablatronix, 2021-09-17:
well its probably in your loop where you check for button press?

ruedli, 2021-09-17:
I think I uploaded a different sketch from the IDE, the results were not valid so I removed the items to avoid confusion. I have tried with FS64 and FS128, but it is a case of being with the doctor: when you  are there the pain is gone. I'll continue with more tests: could the "stack protection: enabled" have anything to do with it?

tablatronix, 2021-09-17:
I really do not know what that is

ruedli, 2021-09-17:
Thnx for your knwoledge eye on this, and your friendly help:

OK, about 10 builds further, I managed to crash it with FS 64. I now basically need to upload it through serial, as OTA is not an option with the erased wificredentials and the portal not saving.

This is what I see on the serial:

`
*wm:[2] AP:  -90 Polk MagniFi MAX-8702.l003
*wm:[2] AP:  -91 FRITZ!Box guest access
*wm:[2] AP:  -92 DIRECT-2771AC69
*wm:[2] NUM CLIENTS:  1
*wm:[2] <- HTTP WiFi save  

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4022143c epc2=0x00000000 epc3=0x00000000 excvaddr=0x09fe0000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb40 end: 3fffffc0 offset: 0190
3ffffcd0:  40258d90 00000008 3ffef6a0 40212058
3ffffce0:  3ffef708 09fe0000 3ffffd10 40212313
3ffffcf0:  3fff247c 3ffef694 3ffef608 40212345
3ffffd00:  3ffef708 3ffef694 3ffef608 4020d331
3ffffd10:  00000000 3fff2400 00000000 4020bc28
3ffffd20:  4021d474 00000000 000003e8 3fff040c
3ffffd30:  00000000 3ffffd3c ffffffff 00000000
3ffffd40:  3fff0454 09fe0000 00000000 402127b8
3ffffd50:  00000000 00000000 00637067 00000000
3ffffd60:  3ffef708 00000001 3fff0634 4021c9cc
3ffffd70:  00000001 00000000 00000000 401000e9
3ffffd80:  00000000 00000000 00000000 40209295
3ffffd90:  00000000 3fff0454 3fff040c 4020c390
3ffffda0:  00000000 00000000 00000000 69746365
3ffffdb0:  00000000 3ffef708 40100389 00000000
3ffffdc0:  00000000 00000001 00000000 00000000
3ffffdd0:  00000000 00000000 3fff040c 00000000
3ffffde0:  3ffef708 3fff0430 3fff040c 4020e58e
3ffffdf0:  4021d3e0 00000000 00001388 402168cf
3ffffe00:  00000000 00000000 3fff2404 00000000
3ffffe10:  3ffef708 3ffef708 3ffef608 4020e7cd
3ffffe20:  3ffef708 00000001 3ffef608 40208c32
3ffffe30:  00000000 00000000 00000001 40100274
3ffffe40:  6e6f6365 00000000 3ffef608 00000000
3ffffe50:  3ffef708 00000001 3ffef608 00000000
3ffffe60:  3ffef708 00000001 3ffef608 4020eb00
3ffffe70:  00000000 6e6f6365 00000000 00000000
3ffffe80:  73646e6f 00000000 00000000 6e6f6365
3ffffe90:  00000000 3ffef8f8 3ffef608 40206b78
3ffffea0:  3ffe8a5d 00000000 00000000 00000000
3ffffeb0:  3ffef3a1 3ffef608 3ffef608 00000000
3ffffec0:  3ffef3a1 3ffef608 3ffef608 4020ecda
3ffffed0:  40106211 009391ee 3ffefaa4 00000000
3ffffee0:  3ffeef30 3ffefaa4 000001f4 00000001
3ffffef0:  3ffe8a5d 3ffefaa4 000001f4 402132b0
3fffff00:  40213c51 000001f4 00000001 00000001
3fffff10:  3ffef3a1 3ffef608 3ffef8f8 40201baa
3fffff20:  402527d3 00000001 0000000d 401014e8
3fffff30:  3fff041c 3ffef4b8 3ffef4b8 40100424
3fffff40:  3fffdad0 0000cf7c 3ffef3c8 40202048
3fffff50:  4303a800 00000000 3ffef8f8 3fffff88
3fffff60:  007a1200 267e38fd 3ffef800 4020298e
3fffff70:  00000000 feefeffe 4021c45c 402010c4
3fffff80:  00000001 00000000 00000001 40100274
3fffff90:  3fffdad0 00000000 3ffefa28 3ffefa3c
3fffffa0:  3fffdad0 00000000 3ffefa28 402133d0
3fffffb0:  feefeffe feefeffe 3ffe85ec 40100f9d
<<<stack<<<
[... 37 lines cut ...]
0x40209295: WiFiManager::DEBUG_WM (WiFiManager::wm_debuglevel_t, char const*, unsigned int) at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 79
0x4020c390: WiFiManager::DEBUG_WM__FlashStringHelper const*, char*>(WiFiManager::wm_debuglevel_t, __FlashStringHelper const*, char*) at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 79
0x40100389: schedule_function(std::function  const&) at c:\users\ruudr\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits/move.h line 189
0x4020e58e: esp8266webserver::ESP8266WebServerTemplate ::_prepareHeader(String&, int, char const*, unsigned int) at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 79
0x4021d3e0: DhcpServer::begin(ip_info*) at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\LwipDhcpServer.cpp line 1058
0x402168cf: EspClass::flashWriteUnalignedMemory(unsigned int, unsigned char const*, unsigned int) at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\Esp.cpp line 786
0x4020e7cd: esp8266webserver::ESP8266WebServerTemplate ::send(int, String const&, String const&) at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 455
0x40208c32: WiFiManager::getInfoData(String) at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 146
0x40100274: schedule_recurrent_function_us(std::function  const&, unsigned int, std::function  const&) at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/PolledTimeout.h line 129
0x4020eb00: WiFiManager::captivePortal() at C:\Users\ruudr\OneDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 1862
0x40206b78: ArduinoOTAClass::_onRx() at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 79
0x4020ecda: WiFiManager::handleNotFound() at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 334
0x402132b0: esp8266::MDNSImplementation::MDNSResponder::_releaseUDPContext() at c:\users\ruudr\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3. [... cut ...]
0x40100274: schedule_recurrent_function_us(std::function  const&, unsigned int, std::function  const&) at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/PolledTimeout.h line 129
0x402133d0: esp8266::MDNSImplementation::MDNSResponder::_allocUDPContext() at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266mDNS\src\LEAmDNS_Helpers.cpp line 187
0x40100f9d: realloc(void*, size_t) at C:\Users\ruudr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1160 
`

ruedli, 2021-09-17:
As you can see, the parameter callback fired, the serial print being the prove of that. 

I saved them into EEPROM, no issues there. 

I didn't set a countrycode as when I tried with NL (for the Netherlands) this tiny little chip in the Sonoff tried to convince me that my country is not existing. OK, my country small but still...

ruedli, 2021-09-17:
Just in case you like the colored picture better...
[image]

ruedli, 2021-09-17:
Back to no FS: working WiFimanager, but OTA gives me the "No answer to our Authentication". Is OTA really needing a tiny FS? It has these lines that I just copied without considering anything on it:

`
ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_FS
      type = "filesystem";
    }

    // NOTE: if updating FS this would be the place to unmount FS using FS.end()
    Serial.println("Start updating " + type);
  }); 
`

ruedli, 2021-09-17:
After many rebuilds I had problems with either Wifi saving credentials or OTA in 60% of the cases. My work mode where I have the least problems is with the FS to 0. _WiFimanager as far as I can see has no failures_. **OTA fails**, but then after I power off/power on of the device, it works again.... This behaviour might have led me to believe that it was related to the FS setting, as obviously when I removed it from mains to plug it into a serial port for flashing, I also removed power and subsequently "OTA worked again". Correlation and causality... not necessarily the same!

I will now continue in the no FS mode and resolve OTA issues by powering off and see how I go.

It is very important to me to better understand the underlying mechanism and to know what works and what not. I am about to invite people in the forum on 3D printing to try this out. Now that OTA and Wifi configuration work, this is feasible. However if many people would need a serial upload (which is not straightforward for a Sonoff 26) I am reluctant to roll it out.

A few words to better understand and get a feeling how stable a solution is based on mqtt, OTA, an API and wifimanager would be most welcome! If not for anything else I suggest to close this issue. Your help was much appreciated! At least you convinced me to take the FS out of the questions!

ruedli, 2021-09-17:
For an idea what I am working on, see: https://github.com/ruedli/OctoPlugout

ruedli, 2021-09-17:
(and just after plugging it in the wall : I cannot update credentials, need to kick it over serial again :-(  )

It sure feels unstable...

ruedli, 2021-09-17:
I have the feeling, that the latest from master is not good for me. Neither do I need all functionality of this one: just wifi credentials and 7 parameters. Is there another release / label that I can try?

ruedli, 2021-09-17:
For now I store the wifi credentials in EEPROM, disable the wifi page and display these parameters on the parameter page together with all others. By changing the logic around autoconnect, and disconnecting (and reconnecting with my own parameters) around it I get the functionality needed without crashing. This is because parameter saving always works, the crash is with credential saving. Dirty (and slow) but it works.... 

I will try again when I learn about a stable way to get the normal wifi credential management.

This is the code I am using:

`
	WiFi.mode(WIFI_STA);
	if (forcePortal) {
		if (WifiAvailable()) { WiFi.disconnect();}
	} else {
		WiFi.begin(s_wifi_ssid,s_wifi_pass);
		long Now=millis();
		while ((WifiNotAvailable()) and ((millis()-Now)<15000)) delay(10);			// Try if connection is there for 5s
	}
	
	wm.autoConnect("SetupOctoPlugout"); // no password protected ap: if the password / ssid are wrong the portal is active.

	if (WifiNotAvailable()) {
		WiFi.mode(WIFI_STA);
		WiFi.begin(s_wifi_ssid,s_wifi_pass);
		long Now=millis();
		while ((WifiNotAvailable()) and ((millis()-Now)<15000)) delay(10);	
	}		

    if (WifiNotAvailable()) {
        Serial.println(F("Failed to connect, restarting"));
        ESP.restart();
		while(true); //Do not continue....
    } else {
        //if you get here you have connected to the WiFi    
        #ifdef debug_
		Serial.println(F("connected...yeey"));
		#endif
    }
`

tablatronix, 2021-09-18:
I honestly am lost, i have no idea what you are doing at this point, your code needs to be separated into functions, FS has nothing to do with credentials, that stack trace is useless, it looks like mdns was the last thing, nothing about saving, its just all strange. And I am not sure what memory partitions you are using.

I would a start with a working example from the repo and see if the problem still exists, it could be a SDK bug, it could be your params not staying in scope, it could be a hundred things.

You are also saving to EEPROM ? are you sure your address space is correct? You could be writing to invalid memory

ruedli, 2021-09-18:
and... after preventing having to go  into the (always crashing)  credentials screen and rewriting everything to use params only, today the param screen crashes upon saving..... I am not sure what to do other then manually coding this functionality. If a more stable version is available, I look forward to hear.

ruedli, 2021-09-18:
@tablatronix - again thnx for your feedback

On the EEPROM structure: yes after writing the routines for this, I investigated what was in the EEPORM with a separate sketch. It was all OK. Your surprise wrt. msdn got me thinking. My PC connects through wifi with the portal, but is connected with a fixed line to my router amd the internet as well. When I connect to the wifi of the AP, a new msdn news page pops up in my browser, and I close this browser tab and then connect to the config window in other browser. Could all this information coming from the msdn somehow land in WiFiManager and crash things? They have nothing to do with the config portal and it would explain the semi random character of things going wrong. Could incoming messages other then from a config portal be ignored somehow? I'll try more, with the fixed connection to my PC disconnected.

ruedli, 2021-09-21:
@tablatronix  

I integrated the "onDemandnonBlocking" example in my project, as that was closest to what I wanted: AP is invoked when wifi credentials are  invalid, or not provided and a portal for the parameter editing on the actual connected network under a provided hostname. 

**All is well now.** 

The most predominant thing I seem to have missed before was the include of mdns in my main. In my case #include <ESP8266mDNS.h>.  It compiled fine without it, but I think it did not not entirely work without issues.

Thnx to tablatronix for signalling this, even while I confused "MS news portal" with mDNS: the domain name service that channels traffic in AP mode to the config page. The fact that I had two internet connections active (wired and wireless to the AP) was not helping either.

For the record: **FS 128k is no influence**: my ESP just crashed and when I compiled without these FS settings and with it, it did not crash. I wrongly correlated this, down the road this became clear.

** very happy that it works now ** 

My code with wifimanager integrated is still on https://github.com/ruedli/OctoPlugout

closed 2021-09-21 by ruedli

tablatronix, 2021-09-21:
Thats odd that it was showing up like that I have never seen an MDNS in stack or a crash in that code

ruedli, 2021-09-21:
Two things were wrong: I did not have the mdns include in my main sketch,
and I had both WiFi adapter as ethernet cable active, which might have
confused the DNS redirection, as it was definitely initiating and loading a
homepage from Microsoft, and not just the 192.168.4.1. That traffic maybe
came through somehow, asking to be translated by the esp.

tablatronix, 2021-09-22:
ah maybe it was hammering the esp, I have seen that also when your device has an internet connection, all background processes keep hitting the ap