tzapu/WiFiManager #1296: Configuration captive portal chanel

sebosfato, 2021-09-29:
Hello guys i have a quick question im not being able to find an answer... 

When i try to connect my nodemcu to my mac it only make a efective conection and show the captive portal if i change the chanel like this... seting the 6 will make it connect and work ok... i found this issue having the mac to run the diagnostics and it detected that channel one was full or something like that... 

if i use this in setup my index page run 
  WiFi.softAP(ssid,password,6,hiden);
but if i run the config portal it wont connect

however when it goes to the configuration portal (wifi manager) it dont even connect... i was trying to find a way to set the channel of the configuration portal is there a piece of line like this? 

https://github.com/khoih-prog/ESP_WiFiManager
11. Using random AP-mode channel to avoid conflict

// From v1.0.10 only
// Set config portal channel, default = 1. Use 0 => random channel from 1-13
ESP_wifiManager.setConfigPortalChannel(0);
//////

## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void setup() {

}

void loop() {

}
#END
```

### Debug Messages

```
messages here
```

sebosfato, 2021-09-30:
I found the command in the wifimanager cpp 

is it automatic channel? i didnt found much info about... anyways it would not connect if i didnt change the channel... 

should'nt it try a different channel every time it reset if it were automatic?

tablatronix, 2021-09-30:
I am not sure how the esp lib works by default, it might not be actually automatic but first one, not sure thats an ESP question.

Another issue you might be having is your esp country not set, so try that also.

sebosfato, 2021-10-05:
nice answer thanks a lot... im actually using your library... 

Another question arising is how to deal with routers and keep my esp8266 to be reach by a mobile on other router. Im trying to find how to set subnet so everyone see each others...