tzapu/WiFiManager #1299: setAPStaticIPConfig(); doesn't work

JelleReith, 2021-10-04:
### Basic Infos

setAPStaticIPConfig(); doesn't work reliable. The problem is that WiFi.softAPConfig has to be called after WiFi.softAP. 

Fix: 
  if (_apPassword != "") {
    if(channel>0){
      ret = WiFi.softAP(_apName.c_str(), _apPassword.c_str(),channel,_apHidden);
    }  
    else{
      ret = WiFi.softAP(_apName.c_str(), _apPassword.c_str(),1,_apHidden);//password option
    }
  } else {
    #ifdef WM_DEBUG_LEVEL
    DEBUG_WM(DEBUG_VERBOSE,F("AP has anonymous access!"));    
    #endif
    if(channel>0){
      ret = WiFi.softAP(_apName.c_str(),"",channel,_apHidden);
    }  
    else{
      ret = WiFi.softAP(_apName.c_str(),"",1,_apHidden);
    }  
  }
  // setup optional soft AP static ip config
  if (_ap_static_ip) {
    #ifdef WM_DEBUG_LEVEL
    DEBUG_WM(F("Custom AP IP/GW/Subnet:"));
    #endif
    if(!WiFi.softAPConfig(_ap_static_ip, _ap_static_gw, _ap_static_sn)){
      #ifdef WM_DEBUG_LEVEL
      DEBUG_WM(DEBUG_ERROR,F("[ERROR] softAPConfig failed!"));
      #endif
    }
  }

#### Hardware
WiFimanager Branch/Release:  Master

Esp32: 

Hardware:			ESP-12e

Core Version:      	2.4.0, staging

tablatronix, 2021-10-04:
This would be in direct conflict with this

https://github.com/tzapu/WiFiManager/issues/721

sooo hmmm

JelleReith, 2021-10-04:
> This would be in direct conflict with this
> 
> #721
> 
> sooo hmmm

tablatronix, 2021-10-04:
Sorry that was enable ap, not start ap, my bad

let me check , I am pretty sure the ip needs to be setup first last time I looked at the code,

tablatronix, 2021-10-04:
Yeah I was right,  we moved it before softap on purpose and tested it, so that seems odd that both work? Which makes no sense.

tablatronix, 2021-10-04:
You will need to enable wifi debugging and see if you have `[APConfig]` messages

JelleReith, 2021-10-04:
I also got it to work some times when it was before softAp but it fails about 50% of the time. When it's after softAp it behaves much better and never fails. I'm honestly not sure why it fails ...

tablatronix, 2021-10-04:
it looks like there is either an error being throw ( bad wifi config )

I am looking at the code and other than an error for bad config I do not see what could be running different, other than something getting toggled.

you can see there is alot of debuggin in there and return errors also

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp

JelleReith, 2021-10-04:
The esp isn't giving me any errors and is running the code fine, but sometimes it just uses the default ip of 192.168.4.1 and sometimes it's using the custom ip.

tablatronix, 2021-10-04:
Do you have ESP debugging on ?

JelleReith, 2021-10-05:
Yes debugging is on. I did a few more test but it really seems that the config statement must be after the startAP. Maybe a clue is in the logic of my program: First it does a wm.autoConnect(), if it works it gets the current time from a ntpServer, after that it disconnects from the wifi and starts a configPortal. If the autoConnect fails it just starts the configPortal by default. The AP that's started by wm.autoConnect() sometimes has the custom IP adres, but sometimes it doesn't. I did a few tests with the example "AutoConnectWithFSParametersAndCustomIP". I added "  wm.setAPStaticIPConfig(IPAddress(172, 217, 28, 1), IPAddress(172, 217, 28, 1), IPAddress(255, 255, 255, 0));" before " if (!wm.autoConnect("AutoConnectAP", "password")) {".

After pressing the esp32 reset button a few times this is the serial monitor output:
12:06:41.475 -> mounting FS...
12:06:41.545 -> mounted file system
12:06:41.651 -> *wm:[2] Added Parameter: server
12:06:41.651 -> *wm:[2] Added Parameter: port
12:06:41.651 -> *wm:[2] Added Parameter: api
12:06:41.651 -> *wm:[1] AutoConnect 
12:06:41.651 -> *wm:[1] No Credentials are Saved, skipping connect 
12:06:41.651 -> *wm:[2] Starting Config Portal 
12:06:41.651 -> *wm:[2] AccessPoint set password is VALID 
12:06:41.651 -> *wm:[2] Disabling STA 
12:06:41.685 -> *wm:[2] Enabling AP 
12:06:41.685 -> *wm:[1] StartAP with SSID:  AutoConnectAP
12:06:41.685 -> *wm:[1] Custom AP IP/GW/Subnet: 
12:06:42.173 -> *wm:[1] AP IP address: 192.168.4.1
12:06:42.173 -> *wm:[1] Starting Web Portal 
12:06:42.173 -> *wm:[2] HTTP server started 
12:06:42.207 -> *wm:[2] Config Portal Running, blocking, waiting for clients... 
12:06:42.788 -> ets Jun  8 2016 00:22:57
12:06:42.788 -> 
12:06:42.788 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:06:42.788 -> configsip: 0, SPIWP:0xee
12:06:42.788 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:06:42.788 -> mode:DIO, clock div:1
12:06:42.788 -> load:0x3fff0018,len:4
12:06:42.788 -> load:0x3fff001c,len:1216
12:06:42.821 -> ho 0 tail 12 room 4
12:06:42.821 -> load:0x40078000,len:10944
12:06:42.821 -> load:0x40080400,len:6388
12:06:42.821 -> entry 0x400806b4
12:06:43.229 -> 
12:06:43.229 -> mounting FS...
12:06:43.299 -> mounted file system
12:06:43.404 -> *wm:[2] Added Parameter: server
12:06:43.404 -> *wm:[2] Added Parameter: port
12:06:43.404 -> *wm:[2] Added Parameter: api
12:06:43.404 -> *wm:[1] AutoConnect 
12:06:43.404 -> *wm:[1] No Credentials are Saved, skipping connect 
12:06:43.404 -> *wm:[2] Starting Config Portal 
12:06:43.404 -> *wm:[2] AccessPoint set password is VALID 
12:06:43.404 -> *wm:[2] Disabling STA 
12:06:43.438 -> *wm:[2] Enabling AP 
12:06:43.438 -> *wm:[1] StartAP with SSID:  AutoConnectAP
12:06:43.438 -> *wm:[1] Custom AP IP/GW/Subnet: 
12:06:43.926 -> *wm:[1] AP IP address: 192.168.4.1
12:06:43.926 -> *wm:[1] Starting Web Portal 
12:06:43.926 -> *wm:[2] HTTP server started 
12:06:43.926 -> *wm:[2] Config Portal Running, blocking, waiting for clients... 
12:06:44.901 -> ets Jun  8 2016 00:22:57
12:06:44.901 -> 
12:06:44.901 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:06:44.901 -> configsip: 0, SPIWP:0xee
12:06:44.901 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:06:44.901 -> mode:DIO, clock div:1
12:06:44.901 -> load:0x3fff0018,len:4
12:06:44.901 -> load:0x3fff001c,len:1216
12:06:44.901 -> ho 0 tail 12 room 4
12:06:44.901 -> load:0x40078000,len:10944
12:06:44.937 -> load:0x40080400,len:6388
12:06:44.937 -> entry 0x400806b4
12:06:45.312 -> 
12:06:45.312 -> mounting FS...
12:06:45.412 -> mounted file system
12:06:45.516 -> *wm:[2] Added Parameter: server
12:06:45.516 -> *wm:[2] Added Parameter: port
12:06:45.516 -> *wm:[2] Added Parameter: api
12:06:45.516 -> *wm:[1] AutoConnect 
12:06:45.516 -> *wm:[1] No Credentials are Saved, skipping connect 
[... 28 lines cut ...]
12:06:47.365 -> *wm:[1] AutoConnect 
12:06:47.365 -> *wm:[1] No Credentials are Saved, skipping connect 
12:06:47.399 -> *wm:[2] Starting Config Portal 
12:06:47.399 -> *wm:[2] AccessPoint set password is VALID 
12:06:47.399 -> *wm:[2] Disabling STA 
12:06:47.399 -> *wm:[2] Enabling AP 
12:06:47.399 -> *wm:[1] StartAP with SSID:  AutoConnectAP
12:06:47.399 -> *wm:[1] Custom AP IP/GW/Subnet: 
12:06:47.920 -> *wm:[1] AP IP address: 172.217.28.1
12:06:47.920 -> *wm:[1] Starting Web Portal 
12:06:47.920 -> *wm:[2] HTTP server started 
12:06:47.920 -> *wm:[2] Config Portal Running, blocking, waiting for clients... 
12:06:49.310 -> ets Jun  8 2016 00:22:57
12:06:49.310 -> 
12:06:49.310 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:06:49.348 -> configsip: 0, SPIWP:0xee
12:06:49.348 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:06:49.348 -> mode:DIO, clock div:1
12:06:49.348 -> load:0x3fff0018,len:4
12:06:49.348 -> load:0x3fff001c,len:1216
12:06:49.348 -> ho 0 tail 12 room 4
12:06:49.348 -> load:0x40078000,len:10944
12:06:49.348 -> load:0x40080400,len:6388
12:06:49.348 -> entry 0x400806b4
12:06:49.754 ->

tablatronix, 2021-10-05:
ok I was looking at esp8266, my bad, let me check this on esp32

tablatronix, 2021-10-05:
Why are you setting the gateway to the ip ?

wm.setAPStaticIPConfig(IPAddress(172, 217, 28, 1), IPAddress(172, 217, 28, 1), IPAddress(255, 255, 255, 0))

tablatronix, 2021-10-05:
https://github.com/espressif/arduino-esp32/issues/985

tablatronix, 2021-10-05:
Ok so

it appears
'softAPConfig' sets config after, because ap has to be up as it only sets runtime config
'wifi_softap_config' should be capable of setting it before

esp8266 combines these I think and restarts dhcps as needed.

JelleReith, 2021-10-05:
> Why are you setting the gateway to the ip ? Is this normal to do?
> 
> wm.setAPStaticIPConfig(IPAddress(172, 217, 28, 1), IPAddress(172, 217, 28, 1), IPAddress(255, 255, 255, 0))

Hmm not sure if it's considered normal but when I do this the captive portal also gives a pop-up on Samsung devices. Would you set it in a different way? I added a delay after softAP and its reliable now. Would there be a way to check if the AP has started maybe? Anyway thanks for your help!

tablatronix, 2021-10-05:
I linked that issue above, have not entirely read it yet