tzapu/WiFiManager #1300: Hostname does not work

VladimirVecera, 2021-10-05:
I'm trying to set a hostname, but it doesn't work. I still have esp32-987C40 on my wifi despite CTRL + F5.
What am I doing wrong?

`wm.setHostname("WifiName");`

ESP32 core 2.0.0
Version wifimanager: 2.0.4

tablatronix, 2021-10-05:
Which hostname are you looking at?

VladimirVecera, 2021-10-05:
I want to have my own hostname, ie "WifiName", but the hostname cannot be set, wifimanager ignores the settings and if I look in the router, the name remains the same, ie esp32-987C40

tablatronix, 2021-10-06:
Perhaps you are not calling autoconnect, either way mine works fine..

[image]
[image]
[image]

closed 2021-11-03 by tablatronix