tzapu/WiFiManager #1306: Change "CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO"

whogarden, 2021-10-29:
### Basic Infos

#### Hardware
ESP-WROOM-32 4MB (Dev board)

### Description

Arduino IDE 1.8.13
Board selected : ESP32 Dev Module
Partition Scheme: "Default 4MB with spiffs(1.2MB APP/1.5MB SPIFFS)"
Sketch : ..\libraries\WiFiManager\examples\Basic\Basic.ino
Library from https://github.com/tzapu/WiFiManager version=2.0.4-beta
Compilation error with Arduino core for the ESP32 release v2.0.1RC1
Error :
..\libraries\WiFiManager/strings_en.h:374:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope
const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

with Arduino core for the ESP32 release v2.0.1RC1
The define 'CONFIG_ESP32_PHY_MAX_TX_POWER' is renamed 
Please see issue #5821

### Sketch

```
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
    // it is a good practice to make sure your code sets wifi mode how you want it.

    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wm;

    // reset settings - wipe stored credentials for testing
    // these are stored by the esp library
    //wm.resetSettings();

    // Automatically connect using saved credentials,
    // if connection fails, it starts an access point with the specified name ( "AutoConnectAP"),
    // if empty will auto generate SSID, if password is blank it will be anonymous AP (wm.autoConnect())
    // then goes into a blocking loop awaiting configuration and will return success result

    bool res;
    // res = wm.autoConnect(); // auto generated AP name from chipid
    // res = wm.autoConnect("AutoConnectAP"); // anonymous ap
    res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

    if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }

}

void loop() {
    // put your main code here, to run repeatedly:   
}
```

### Debug Messages:

In file included from E:\Documents\me\Labo\Arduino\Croquis\libraries\WiFiManager-master/WiFiManager.h:114,
                 from C:\Users\me\AppData\Local\Temp\arduino_modified_sketch_866864\Basic.ino:1:
E:\Documents\me\Labo\Arduino\Croquis\libraries\WiFiManager-master/strings_en.h:374:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope
 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:\Documents\me\Labo\Arduino\Croquis\libraries\WiFiManager-master/strings_en.h:374:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'
 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                              CONFIG_ESP_PHY_MAX_TX_POWER
E:\Documents\me\Labo\Arduino\Croquis\libraries\WiFiManager-master/strings_en.h:375:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope
 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:\Documents\me\Labo\Arduino\Croquis\libraries\WiFiManager-master/strings_en.h:375:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'
 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                              CONFIG_ESP_PHY_MAX_TX_POWER
E:\Documents\me\Labo\Arduino\Croquis\libraries\WiFiManager-master/strings_en.h:376:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope
 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:\Documents\me\Labo\Arduino\Croquis\libraries\WiFiManager-master/strings_en.h:376:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'
 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                              CONFIG_ESP_PHY_MAX_TX_POWER

exit status 1
Erreur de compilation pour la carte ESP32 Dev Module
```

JasXSL, 2021-10-30:
To clarify the above: 
This is the original issue https://github.com/espressif/arduino-esp32/issues/5821
`CONFIG_ESP32_PHY_MAX_TX_POWER` was renamed to `CONFIG_ESP_PHY_MAX_WIFI_TX_POWER` in the latest ESP core.

tablatronix, 2021-10-30:
Yeah just found this also

touchgadget, 2021-10-30:
There is a backwards compatibility #define for CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER in sdkconfig.h. It has the same value as CONFIG_ESP32_PHY_MAX_TX_POWER and CONFIG_ESP_PHY_MAX_WIFI_TX_POWER. The advantage is this works for arduino-esp32 1.0.6, 2.0.0, and 2.0.1-RC1. There is no backwards compat #define for CONFIG_ESP32_PHY_MAX_TX_POWER.

```
diff --git a/strings_en.h b/strings_en.h
index f28c18d..543792d 100644
--- a/strings_en.h
+++ b/strings_en.h
@@ -371,9 +371,9 @@ const char* const WIFI_MODES[] PROGMEM = { "NULL", "STA", "AP", "STA+AP" };
 //     int8_t                max_tx_power;   /**< This field is used for getting WiFi maximum transmitting power, call esp_wifi_set_max_tx_power to set the maximum transmitting power. */
 //     wifi_country_policy_t policy;  /**< country policy */
 // } wifi_country_t;
-const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
-const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
-const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
+const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
+const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
+const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
 #elif defined(ESP8266) && !defined(WM_NOCOUNTRY)
 // typedef struct {
 //     char cc[3];               /**< country code string */
@@ -483,4 +483,4 @@ typedef union {
 } arduino_event_info_t;
 
 */
-#endif
```

tablatronix, 2021-10-30:
Thanks for looking into it, I had to just do something to fix the event names that also all changed using a ver check

closed 2021-10-30 by tablatronix

whogarden, 2021-11-04:
Please, can you increase the version number in library.properties.
Thank you.

tablatronix, 2021-11-04:
Sure I will cut a new beta in a bit

commit 72b53316 (2021-10-30): possible fix #1306