tzapu/WiFiManager #1309: setcountry crashing esp32 S2

tablatronix, 2021-11-09:


tablatronix, 2022-02-03:
Need more testing, no idea, it doesn't crash when I call outside of wm..

uzi18, 2022-02-17:
@tablatronix  same here at platformio

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:184
load:0x40078000,len:12380
ho 0 tail 12 room 4
load:0x40080400,len:2912
entry 0x400805c4
E (322) psram: PSRAM ID read error: 0xffffffff
E (322) spiram: SPI RAM enabled but initialization failed. Bailing out.

Starting
*wm:[1] Free heap:        221356
*wm:[1] ESP SDK version:  v4.4-dev-2990-g583026f04
TEST_CP ENABLED
*wm:[2] Starting Config Portal
*wm:[2] Disabling STA
*wm:[2] Enabling AP
*wm:[1] StartAP with SSID:  ESP32 WiFi
*wm:[2] AP has anonymous access!
*wm:[1] AP IP address: 192.168.4.1
*wm:[2] setting softAP Hostname: ESP32 WiFi
*wm:[2] WiFiSetCountry to US
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.

Core  0 register dump:
PC      : 0x4014039d  PS      : 0x00060b30  A0      : 0x8013df5f  A1      : 0x3ffafca0
A2      : 0x00000000  A3      : 0x3f402414  A4      : 0x3ffc5c68  A5      : 0x3ffbf748  
A6      : 0x00000000  A7      : 0x0000cdcd  A8      : 0x8014039b  A9      : 0x3ffafc80
A10     : 0x00000014  A11     : 0x3f402414  A12     : 0x00000003  A13     : 0x3f41de69  
A14     : 0x00000001  A15     : 0x0000000b  SAR     : 0x0000001c  EXCCAUSE: 0x00000003
EXCVADDR: 0x3f402419  LBEG    : 0x40144f44  LEND    : 0x40144f5c  LCOUNT  : 0x00000000  

Backtrace:0x4014039a:0x3ffafca00x4013df5c:0x3ffafcd0 0x40172b5d:0x3ffafcf0
 #0  0x4014039a:0x3ffafca00 in wifi_set_country at ??:?
```

CircuitSetup, 2022-05-26:
I'm getting the same behavior, but I unfortunately cannot use the exception decoder because that does not work with Arduino 2.0.x either. I've tried in PIO and Arduino IDE.

tablatronix, 2022-05-27:
Yeah I have not been able to debug this, I do know that it only does it if setting it through wm, so I suspect a memory safety issue with the structs I use.